openwrt/packages #7348: mariadb is crashing after first connection attempt

silvah2, 2018-11-08:
OpenWrt SNAPSHOT, r8431-df1191bcea, linksys wrt1200ac, mariadb 10.2.17

After migration from older snapshot (about maybe 3 months old) first connection attempt makes mariadb crash. (mysqld is going down)

All the configs are fine. Something is broken.

micmac1, 2018-11-09:
Hello @silvah2 

I've seen reports before that there are problems on mvebu. Are you also seeing "illegal instruction" in the syslog when this happens?

I won't be able to debug this myself as I don't have this hardware.

Kind regards,
Sebastian

micmac1, 2018-11-09:
I've had some feedback that this may not be related to a kernel version bump. I see the following changes not too long ago which might be of interest:

1) commit [kernel: only optimized for size if small_flash](https://github.com/openwrt/openwrt/commit/cf7154db07c0b8746be290c9e11dcab3c91d237e)
2) commit [binutils: Use 2.31.1 by default](https://github.com/openwrt/openwrt/commit/fface27db82e7926c86a6b15746446f633d533f3)

I think 2) might fit the timeline better. Could you try fresh git checkout, change binutils version in menuconfig to 2.30 and try if that is the culprit?

Kind regards,
Seb

silvah2, 2018-11-09:
where is this in menuconfig?

micmac1, 2018-11-09:
1. Global build settings  --->
      Kernel build options  --->
        Compiler optimization level (Optimize for performance)  --->

2. Advanced configuration options (for developers)  --->
      Toolchain Options  --->
        Binutils Version (Binutils 2.30)  --->

silvah2, 2018-11-09:
ok, compiling

silvah2, 2018-11-09:
still not working:
first connection attempt:
ERROR 2013 (HY000): Lost connection to MySQL server at 'handshake: reading inital communication packet', system error: 11

and then
root@LEDE:~# ps|grep mysql
 4014 root      1052 S    grep mysql

micmac1, 2018-11-11:
Hello,

Could you try out the pull request #7379? That issue caused a segfault for me on mips. I'm not sure if your issue is the same (different arch, no debug provided). But timeline-wise it might be.

Regards,
Seb

champtar, 2018-11-11:
Just merged #7379

silvah2, 2018-11-12:
works now, thx

closed 2018-11-12 by silvah2