openwrt/packages #7360: net/openvswitch: Build broken if IPV6 is disabled

ynezz, 2018-11-09:
Maintainer: @yousong 
Environment: x86-64, openwrt: f975ab8f4e3d5b8a8e81870c70d427f9d84b203b, packages: 4453a8d30e7eac6dde033404852d0d77746783f7

Description:

I'm building OpenWrt with `CONFIG_IPV6` disabled which leads to the following failure:

```
touch /openwrt/build_dir/target-x86_64_musl/linux-x86_64/openvswitch-2.10.1/.prepared_3e3688d8489a3b194a3716085068b98b_6664517399ebbbc92a37c5bb081b5c53_check
. /openwrt/include/shell.sh; gzip -dc /openwrt/dl/openvswitch-2.10.1.tar.gz | tar -C /openwrt/build_dir/target-x86_64_musl/linux-x86_64/openvswitch-2.10.1/.. -xf - 
[ ! -d ./src/ ] || cp -fpR ./src/. /openwrt/build_dir/target-x86_64_musl/linux-x86_64/openvswitch-2.10.1

Applying ./patches/0001-musl-compatibility.patch using plaintext: 
patching file configure.ac
Hunk #1 succeeded at 124 (offset 2 lines).
patching file lib/netdev-linux.c
patching file lib/ovs-atomic.h

Applying ./patches/0100-netdev-linux-Use-unsigned-int-for-ifi_flags.patch using plaintext: 
patching file lib/netdev-linux.c
Hunk #1 succeeded at 3117 (offset 2 lines).

Applying ./patches/0101-netdev-linux-Let-interface-flag-survive-internal-por.patch using plaintext: 
patching file lib/netdev-linux.c
Hunk #1 succeeded at 3120 (offset 2 lines).

Applying ./patches/0102-python-separate-host-target-python-for-cross-compile.patch using plaintext: 
patching file Makefile.am
patching file m4/openvswitch.m4

Applying ./patches/0103-ovs-ctl-fix-setting-hostname.patch using plaintext: 
patching file utilities/ovs-ctl.in

Applying ./patches/0104-ovs-lib-fix-install_dir.patch using plaintext: 
patching file utilities/ovs-lib.in

Applying ./patches/0105-build-disable-building-tests.patch using plaintext: 
patching file Makefile.am

Applying ./patches/0106-ovs-save-compatible-with-busybox-ip-command.patch using plaintext: 
patching file utilities/ovs-save

Applying ./patches/0107-datapath-use-KARCH-when-building-linux-datapath-modu.patch using plaintext: 
patching file datapath/linux/Makefile.main.in
touch /openwrt/build_dir/target-x86_64_musl/linux-x86_64/openvswitch-2.10.1/.prepared_3e3688d8489a3b194a3716085068b98b_6664517399ebbbc92a37c5bb081b5c53
rm -f /openwrt/build_dir/target-x86_64_musl/linux-x86_64/openvswitch-2.10.1/.configured_*
rm -f /openwrt/staging_dir/target-x86_64_musl/stamp/.openvswitch_installed
(cd /openwrt/build_dir/target-x86_64_musl/linux-x86_64/openvswitch-2.10.1; rm -f aclocal.m4; if [ -f ./configure.ac ] || [ -f ./configure.in ]; then [ -d ./autom4te.cache ] && rm -rf autom4te.cache; [ -e ./config.rpath ] || ln -s /openwrt/scripts/config.rpath ./config.rpath; touch NEWS AUTHORS COPYING ABOUT-NLS ChangeLog; AUTOM4TE=/openwrt/staging_dir/host/bin/autom4te AUTOCONF=/openwrt/staging_dir/host/bin/autoconf AUTOMAKE=/openwrt/staging_dir/host/bin/automake ACLOCAL=/openwrt/staging_dir/host/bin/aclocal AUTOHEADER=/openwrt/staging_dir/host/bin/autoheader LIBTOOLIZE=/openwrt/staging_dir/host/bin/libtoolize LIBTOOL=/openwrt/staging_dir/host/bin/libtool M4=/openwrt/staging_dir/host/bin/m4 AUTOPOINT=true /openwrt/staging_dir/host/bin/autoreconf -v -f -i -s -B /openwrt/staging_dir/host/share/aclocal -I /openwrt/staging_dir/target-x86_64_musl/host/share/aclocal -I /openwrt/staging_dir/hostpkg/share/aclocal -I /openwrt/staging_dir/target-x86_64_musl/usr/share/aclocal -I m4 -I . . || true; fi; );
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: /openwrt/staging_dir/host/bin/aclocal -I /openwrt/staging_dir/target-x86_64_musl/host/share/aclocal -I /openwrt/staging_dir/hostpkg/share/aclocal -I /openwrt/staging_dir/target-x86_64_musl/usr/share/aclocal -I m4 -I . --force -I m4
autoreconf: configure.ac: tracing
autoreconf: running: /openwrt/staging_dir/host/bin/libtoolize --force
OpenWrt-libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, `build-aux'.
OpenWrt-libtoolize: linking file `build-aux/ltmain.sh'
OpenWrt-libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
OpenWrt-libtoolize: linking file `m4/libtool.m4'
OpenWrt-libtoolize: linking file `m4/ltoptions.m4'
OpenWrt-libtoolize: linking file `m4/ltsugar.m4'
OpenWrt-libtoolize: linking file `m4/ltversion.m4'
OpenWrt-libtoolize: linking file `m4/lt~obsolete.m4'
autoreconf: running: /openwrt/staging_dir/host/bin/autoconf --include=/openwrt/staging_dir/target-x86_64_musl/host/share/aclocal --include=/openwrt/staging_dir/hostpkg/share/aclocal --include=/openwrt/staging_dir/target-x86_64_musl/usr/share/aclocal --include=m4 --include=. --prepend-include=/openwrt/staging_dir/host/share/aclocal --force
autoreconf: running: /openwrt/staging_dir/host/bin/autoheader --include=/openwrt/staging_dir/target-x86_64_musl/host/share/aclocal --include=/openwrt/staging_dir/hostpkg/share/aclocal --include=/openwrt/staging_dir/target-x86_64_musl/usr/share/aclocal --include=m4 --include=. --prepend-include=/openwrt/staging_dir/host/share/aclocal --force
autoreconf: running: /openwrt/staging_dir/host/bin/automake --add-missing --force-missing
autoreconf: Leaving directory `.'
(cd /openwrt/build_dir/target-x86_64_musl/linux-x86_64/openvswitch-2.10.1/./; if [ -x ./configure ]; then find /openwrt/build_dir/target-x86_64_musl/linux-x86_64/openvswitch-2.10.1/ -name config.guess | xargs -r chmod u+w; find /openwrt/build_dir/target-x86_64_musl/linux-x86_64/openvswitch-2.10.1/ -name config.guess | xargs -r -n1 cp --remove-destination /openwrt/scripts/config.guess; find /openwrt/build_dir/target-x86_64_musl/linux-x86_64/openvswitch-2.10.1/ -name config.sub | xargs -r chmod u+w; find /openwrt/build_dir/target-x86_64_musl/linux-x86_64/openvswitch-2.10.1/ -name config.sub | xargs -r -n1 cp --remove-destination /openwrt/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/openwrt/build_dir/target-x86_64_musl/linux-x86_64/openvswitch-2.10.1:openvswitch-2.10.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto -std=gnu99" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/openwrt/build_dir/target-x86_64_musl/linux-x86_64/openvswitch-2.10.1:openvswitch-2.10.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto -std=gnu99 " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/openwrt/build_dir/target-x86_64_musl/linux-x86_64/openvswitch-2.10.1:openvswitch-2.10.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto -std=gnu99 " CPPFLAGS="-I/openwrt/staging_dir/target-x86_64_musl/usr/include -I/openwrt/staging_dir/target-x86_64_musl/include -I/openwrt/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/usr/include -I/openwrt/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/include/fortify -I/openwrt/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/include " LDFLAGS="-L/openwrt/staging_dir/target-x86_64_musl/usr/lib -L/openwrt/staging_dir/target-x86_64_musl/lib -L/openwrt/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/usr/lib -L/openwrt/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/lib -znow -zrelro "  ovs_cv_flake8=no ovs_cv_python3=no ovs_cv_sphinx=no ovs_cv_python=python2.7 ovs_cv_python_host=/openwrt/staging_dir/hostpkg/bin/python2.7 KARCH=x86   ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls --disable-ipv6 _python_sysroot="/openwrt/staging_dir/target-x86_64_musl" _python_prefix="/usr" _python_exec_prefix="/usr" --enable-ndebug --enable-shared --disable-libcapng  --with-linux=/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-4.14.79 ; fi; )
[... cut ...]
/openwrt/build_dir/target-x86_64_musl/linux-x86_64/openvswitch-2.10.1/datapath/linux/ip6_gre.c:1676:30: error: storage size of 'ip6gre_protocol' isn't known
 static struct inet6_protocol ip6gre_protocol __read_mostly = {
                              ^~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[10]: *** [/openwrt/build_dir/target-x86_64_musl/linux-x86_64/openvswitch-2.10.1/datapath/linux/ip6_gre.o] Error 1
make[9]: *** [_module_/openwrt/build_dir/target-x86_64_musl/linux-x86_64/openvswitch-2.10.1/datapath/linux] Error 2
make[9]: Leaving directory `/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-4.14.79'
```

yousong, 2018-11-09:
Looks like you have enabled `CONFIG_DEVEL` to build in-tree kernel module.  Please consider providing a patch or solution if possible.  It may take a while for me to set aside time to track this down for the moment.  Thank you.

ynezz, 2018-11-09:
@yousong 

Thanks for your quick response!

> Looks like you have enabled CONFIG_DEVEL

```
# CONFIG_DEVEL is not set
```

Any other idea? Packages from `openwrt-18.06` at 5552214d822c7e5de5bda0633016a0be8c24ae7f builds fine with the same config.

yousong, 2018-11-09:
Please share the output of `./script/diffconfig.sh` (with sensitive parts clipped out).

The log showed that configure command was supplied with `--with-linux=` option which should only happen when intree kernel module was selected, which depends on `CONFIG_DEVEL`.  If that's not the case, it's a bug.

yousong, 2018-11-09:
This is the relevant snippet in the code https://github.com/openwrt/packages/blob/master/net/openvswitch/Makefile#L299-L304

As for the in-tree datapath code build failure when IPv6 was enabled, it's fact in the upstream code.  I may consider making the relevant config options dependent on `CONFIG_IPV6`.

ynezz, 2018-11-09:
> Please share the output of ./script/diffconfig.sh
```
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_Generic=y
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_BUSYBOX_CONFIG_CHPASSWD=y
CONFIG_BUSYBOX_CONFIG_FEATURE_IPC_SYSLOG=y
CONFIG_BUSYBOX_CONFIG_FEATURE_IPC_SYSLOG_BUFFER_SIZE=4
CONFIG_BUSYBOX_CONFIG_FEATURE_REMOTE_LOG=y
CONFIG_BUSYBOX_CONFIG_FEATURE_SYSLOGD_READ_BUFFER_SIZE=256
CONFIG_BUSYBOX_CONFIG_SYSLOGD=y
# CONFIG_BUSYBOX_DEFAULT_FEATURE_IPV6 is not set
CONFIG_GRUB_TIMEOUT="0"
# CONFIG_IPV6 is not set
# CONFIG_KERNEL_IPV6 is not set
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_EC=y
CONFIG_OPENSSL_WITH_NPN=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENVPN_openssl_ENABLE_DEF_AUTH=y
CONFIG_OPENVPN_openssl_ENABLE_FRAGMENT=y
CONFIG_OPENVPN_openssl_ENABLE_HTTP=y
CONFIG_OPENVPN_openssl_ENABLE_LZ4=y
CONFIG_OPENVPN_openssl_ENABLE_LZO=y
CONFIG_OPENVPN_openssl_ENABLE_MULTIHOME=y
CONFIG_OPENVPN_openssl_ENABLE_PF=y
CONFIG_OPENVPN_openssl_ENABLE_PORT_SHARE=y
CONFIG_OPENVPN_openssl_ENABLE_SERVER=y
CONFIG_OPENVPN_openssl_ENABLE_SMALL=y
CONFIG_OPENVPN_openssl_ENABLE_SOCKS=y
CONFIG_PACKAGE_ATH_DEBUG=y
CONFIG_PACKAGE_ca-bundle=y
CONFIG_PACKAGE_ddns-scripts=y
CONFIG_PACKAGE_e100-firmware=y
# CONFIG_PACKAGE_firewall is not set
CONFIG_PACKAGE_haserl=y
CONFIG_PACKAGE_hostapd-common=y
CONFIG_PACKAGE_hostapd-mini=y
CONFIG_PACKAGE_ipset=y
CONFIG_PACKAGE_iptables-mod-conntrack-extra=y
CONFIG_PACKAGE_iptables-mod-ipopt=y
CONFIG_PACKAGE_iptables-mod-nat-extra=y
CONFIG_PACKAGE_kmod-3c59x=y
CONFIG_PACKAGE_kmod-8139cp=y
CONFIG_PACKAGE_kmod-8139too=y
CONFIG_PACKAGE_kmod-e100=y
CONFIG_PACKAGE_kmod-forcedeth=y
CONFIG_PACKAGE_kmod-gpio-button-hotplug=y
CONFIG_PACKAGE_kmod-gpio-nct5104d=y
CONFIG_PACKAGE_kmod-ipt-conntrack-extra=y
CONFIG_PACKAGE_kmod-ipt-ipopt=y
CONFIG_PACKAGE_kmod-ipt-ipset=y
CONFIG_PACKAGE_kmod-ipt-nat-extra=y
CONFIG_PACKAGE_kmod-ipt-raw=y
CONFIG_PACKAGE_kmod-leds-apu2=y
CONFIG_PACKAGE_kmod-leds-gpio=y
CONFIG_PACKAGE_kmod-libphy=y
CONFIG_PACKAGE_kmod-natsemi=y
CONFIG_PACKAGE_kmod-ne2k-pci=y
CONFIG_PACKAGE_kmod-nf-ipt6=m
CONFIG_PACKAGE_kmod-nf-nathelper=y
CONFIG_PACKAGE_kmod-nfnetlink=y
CONFIG_PACKAGE_kmod-nls-base=y
CONFIG_PACKAGE_kmod-pcnet32=y
CONFIG_PACKAGE_kmod-ppp=m
CONFIG_PACKAGE_kmod-pppoe=m
CONFIG_PACKAGE_kmod-pppox=m
CONFIG_PACKAGE_kmod-sched=y
CONFIG_PACKAGE_kmod-sched-core=y
CONFIG_PACKAGE_kmod-sis900=y
CONFIG_PACKAGE_kmod-sky2=y
CONFIG_PACKAGE_kmod-slhc=m
CONFIG_PACKAGE_kmod-sp5100_tco=y
CONFIG_PACKAGE_kmod-tg3=y
CONFIG_PACKAGE_kmod-tulip=y
CONFIG_PACKAGE_kmod-tun=y
CONFIG_PACKAGE_kmod-usb-core=y
CONFIG_PACKAGE_kmod-usb-ehci=y
CONFIG_PACKAGE_kmod-usb-ohci=y
CONFIG_PACKAGE_kmod-usb-serial=y
CONFIG_PACKAGE_kmod-usb2=y
CONFIG_PACKAGE_kmod-usb3=y
CONFIG_PACKAGE_kmod-via-rhine=y
CONFIG_PACKAGE_kmod-via-velocity=y
CONFIG_PACKAGE_libfreeradius2=y
CONFIG_PACKAGE_libip6tc=m
CONFIG_PACKAGE_libipset=y
CONFIG_PACKAGE_libltdl=y
CONFIG_PACKAGE_liblzo=y
CONFIG_PACKAGE_libmnl=y
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_libpcap=y
CONFIG_PACKAGE_libustream-openssl=y
CONFIG_PACKAGE_openvpn-openssl=y
# CONFIG_PACKAGE_ppp is not set
CONFIG_PACKAGE_tc=y
CONFIG_TARGET_IMAGES_PAD=y
CONFIG_VDI_IMAGES=y
CONFIG_VMDK_IMAGES=y
```

yousong, 2018-11-10:
Hi, @ynezz , it's a bug in the package Makefile.  I just pushed a fix to the master branch.  Feel free to report back if the issue still persists.

Thank you for the feedback ;)

closed 2018-11-10 by yousong

ynezz, 2018-11-10:
Hi @yousong, thanks for such a quick fix! It's working fine now, thanks!