openwrt/packages #7363: /utils/open2300 build failed, project is no longer maintained, repository has removed

ahmadrasyidsalim, 2018-11-09:
Maintainer: @szogi 
Environment: brcm6xx, generic, OpenWrt-18.06

Description: I try to build all of the feeds but some repository has removed!

```
make download V=s
make[1]: Entering directory '/home/ahmadrasyidsalim/OpenWrt'
make[2]: Entering directory '/home/ahmadrasyidsalim/OpenWrt/tools/gmp'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/ahmadrasyidsalim/OpenWrt/tools/gmp'
time: tools/gmp/download#0.15#0.08#0.49
make[2]: Entering directory '/home/ahmadrasyidsalim/OpenWrt/tools/mpfr'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/ahmadrasyidsalim/OpenWrt/tools/mpfr'
time: tools/mpfr/download#0.15#0.08#0.43
make[2]: Entering directory '/home/ahmadrasyidsalim/OpenWrt/tools/mpc'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/ahmadrasyidsalim/OpenWrt/tools/mpc'
time: tools/mpc/download#0.14#0.08#0.42
make[2]: Entering directory '/home/ahmadrasyidsalim/OpenWrt/tools/libelf'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/ahmadrasyidsalim/OpenWrt/tools/libelf'
time: tools/libelf/download#0.13#0.10#0.43
make[2]: Entering directory '/home/ahmadrasyidsalim/OpenWrt/tools/expat'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/ahmadrasyidsalim/OpenWrt/tools/expat'
time: tools/expat/download#0.15#0.08#0.40
make[2]: Entering directory '/home/ahmadrasyidsalim/OpenWrt/tools/m4'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/ahmadrasyidsalim/OpenWrt/tools/m4'
time: tools/m4/download#0.12#0.11#0.43
make[2]: Entering directory '/home/ahmadrasyidsalim/OpenWrt/tools/libtool'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/ahmadrasyidsalim/OpenWrt/tools/libtool'
time: tools/libtool/download#0.11#0.12#0.68
make[2]: Entering directory '/home/ahmadrasyidsalim/OpenWrt/tools/autoconf'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/ahmadrasyidsalim/OpenWrt/tools/autoconf'
time: tools/autoconf/download#0.14#0.08#0.44
make[2]: Entering directory '/home/ahmadrasyidsalim/OpenWrt/tools/automake'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/ahmadrasyidsalim/OpenWrt/tools/automake'
time: tools/automake/download#0.12#0.11#0.40
make[2]: Entering directory '/home/ahmadrasyidsalim/OpenWrt/tools/flex'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/ahmadrasyidsalim/OpenWrt/tools/flex'
time: tools/flex/download#0.14#0.09#0.46
make[2]: Entering directory '/home/ahmadrasyidsalim/OpenWrt/tools/bison'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/ahmadrasyidsalim/OpenWrt/tools/bison'
time: tools/bison/download#0.15#0.08#0.40
make[2]: Entering directory '/home/ahmadrasyidsalim/OpenWrt/tools/pkg-config'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/ahmadrasyidsalim/OpenWrt/tools/pkg-config'
time: tools/pkg-config/download#0.13#0.10#0.46
make[2]: Entering directory '/home/ahmadrasyidsalim/OpenWrt/tools/mklibs'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/ahmadrasyidsalim/OpenWrt/tools/mklibs'
time: tools/mklibs/download#0.13#0.10#0.51
make[2]: Entering directory '/home/ahmadrasyidsalim/OpenWrt/tools/zlib'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/ahmadrasyidsalim/OpenWrt/tools/zlib'
time: tools/zlib/download#0.14#0.09#0.42
make[2]: Entering directory '/home/ahmadrasyidsalim/OpenWrt/tools/sstrip'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/ahmadrasyidsalim/OpenWrt/tools/sstrip'
time: tools/sstrip/download#0.12#0.07#0.39
make[2]: Entering directory '/home/ahmadrasyidsalim/OpenWrt/tools/make-ext4fs'
make[2]: Nothing to be done for 'download'.
[... 2662 lines cut ...]
time: package/feeds/packages/ngircd/ssl/download#0.42#0.28#1.03
make[2]: Entering directory '/home/ahmadrasyidsalim/OpenWrt/feeds/packages/net/ngircd'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/ahmadrasyidsalim/OpenWrt/feeds/packages/net/ngircd'
time: package/feeds/packages/ngircd/nossl/download#0.45#0.29#1.16
make[2]: Entering directory '/home/ahmadrasyidsalim/OpenWrt/feeds/packages/net/nlbwmon'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/ahmadrasyidsalim/OpenWrt/feeds/packages/net/nlbwmon'
time: package/feeds/packages/nlbwmon/download#0.94#1.08#3.24
make[2]: Entering directory '/home/ahmadrasyidsalim/OpenWrt/feeds/packages/net/nmap'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/ahmadrasyidsalim/OpenWrt/feeds/packages/net/nmap'
time: package/feeds/packages/nmap/nossl/download#0.48#0.26#1.27
make[2]: Entering directory '/home/ahmadrasyidsalim/OpenWrt/feeds/packages/net/nmap'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/ahmadrasyidsalim/OpenWrt/feeds/packages/net/nmap'
time: package/feeds/packages/nmap/ssl/download#0.47#0.27#1.18
make[2]: Entering directory '/home/ahmadrasyidsalim/OpenWrt/feeds/packages/net/nmap'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/ahmadrasyidsalim/OpenWrt/feeds/packages/net/nmap'
time: package/feeds/packages/nmap/nossl/download#0.49#0.25#1.22
make[2]: Entering directory '/home/ahmadrasyidsalim/OpenWrt/feeds/packages/net/nmap'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/ahmadrasyidsalim/OpenWrt/feeds/packages/net/nmap'
time: package/feed

szogi, 2018-11-09:
Thanks for your ticket.
From the developers' site: Project is no longer maintained
http://www.lavrsen.dk/foswiki/bin/view/Open2300/WebHome

I'll ask him about the svn repo.

neheb, 2019-08-04:
This issue is fixed in current master.

closed 2019-08-04 by neheb