openwrt/packages #7367: Packages that needs fix without deprecated APIs for OpenSSL

frymezim, 2018-11-09:
@neheb 
Their are 3 packages that I know causing breakage on compilation without deprecated APIs. Can you provide a patch for this?
lighttpd
wget
libssh2

Also is there a easier way to apply the pull request on my own buildroot because I was able to do this on openwrt git, editing the .git/config and use git merge origin/pr/####.

neheb, 2018-11-09:
Is this on OpenSSL 1.1.x? I've only been testing 1.0.2. Wget patch to fix that has been submitted upstream.

As far as pull requests go, I used to use git-pr for that. Just git-pr ### and git checkout original followed by git cherry-pick.

Nowadays I add the branch directly using git remote and cherry pick that.

frymezim, 2018-11-10:
Yes it's OpenSSL 1.1.x been trying that patch and those three have problems with compilation.

neheb, 2018-11-10:
I tried fixing wget. Didn't work. Some autoconf problem.

frymezim, 2018-11-10:
Ok cool, so I want to test those patches can you give me an example how to merge your pull request on my  build environment from the /home/username/openwrt/ folder thanks!

frymezim, 2018-11-11:
Another issue I found that needed OpenSSL deprecated API.

```
make[3]: Entering directory '/home/dkmori/openwrt/feeds/packages/lang/python/python-pip-conf'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/dkmori/openwrt/feeds/packages/lang/python/python-pip-conf'
time: package/feeds/packages/python-pip-conf/compile#0.07#0.02#0.09
make[3]: Entering directory '/home/dkmori/openwrt/feeds/packages/lang/python/python'
make[3]: Leaving directory '/home/dkmori/openwrt/feeds/packages/lang/python/python'
time: package/feeds/packages/python/host-compile#0.15#0.15#0.32
make[3]: Entering directory '/home/dkmori/openwrt/feeds/packages/lang/python/python'
mkdir -p /home/dkmori/openwrt/bin/targets/kirkwood/generic/packages /home/dkmori/openwrt/build_dir/target-arm_xscale_musl_eabi/Python-2.7.15/ipkg-arm_xscale/python-openssl/CONTROL /home/dkmori/openwrt/staging_dir/target-arm_xscale_musl_eabi/pkginfo
find /home/dkmori/openwrt/build_dir/target-arm_xscale_musl_eabi/Python-2.7.15/ipkg-install -name "*\.exe" | xargs rm -f
/usr/bin/env bash ../python-package-install.sh "2" "/home/dkmori/openwrt/build_dir/target-arm_xscale_musl_eabi/Python-2.7.15/ipkg-install" "/home/dkmori/openwrt/build_dir/target-arm_xscale_musl_eabi/Python-2.7.15/ipkg-arm_xscale/python-openssl" "/home/dkmori/openwrt/staging_dir/hostpkg/bin/python2.7" "" "$V_PyPackage_python_openssl_filespec"
copying: '/usr/lib/python2.7/lib-dynload/_hashlib.so'
File not found '/home/dkmori/openwrt/build_dir/target-arm_xscale_musl_eabi/Python-2.7.15/ipkg-install/usr/lib/python2.7/lib-dynload/_ssl.so'
process filespec error-ed
make[3]: *** [Makefile:304: /home/dkmori/openwrt/bin/packages/arm_xscale/packages/python-openssl_2.7.15-1_arm_xscale.ipk] Error 1
make[3]: Leaving directory '/home/dkmori/openwrt/feeds/packages/lang/python/python'
time: package/feeds/packages/python/compile#0.17#0.10#0.24
make[2]: *** [package/Makefile:109: package/feeds/packages/python/compile] Error 2
make[2]: Leaving directory '/home/dkmori/openwrt'
make[1]: *** [package/Makefile:103: /home/dkmori/openwrt/staging_dir/target-arm_xscale_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/dkmori/openwrt'
make: *** [/home/dkmori/openwrt/include/toplevel.mk:218: world] Error 2
```

neheb, 2018-11-12:
Ha. Ha. Ha.

Python is a mess. I have a patch in my local tree that fixes Python itself. 2 and 3 actually. python-cryptography requires such massive patch that it's not even worth it. I don't expect python-openssl to be much better.

It's probably better to add a dependency on OPENSSL_WITH_DEPRECATED.

commodo, 2018-11-14:
@frymezim 
can you provide a `.config` file that reproduces this ? and some details
i can't seem to reproduce it on my end

neheb, 2018-11-14:
@commodo compile without deprecated APIs on OpenSSL.

neheb, 2018-11-14:
@frymezim https://github.com/neheb/packages/commits/pydep

commodo, 2018-11-15:
Python still seems to build for me.
Tried it on a Marvel Kirkwood, since that looks like it was in the report.

SSL section looks like this:
```
#
# SSL
#
# CONFIG_PACKAGE_libgnutls is not set
# CONFIG_PACKAGE_libmbedtls is not set
CONFIG_PACKAGE_libopenssl=y
CONFIG_OPENSSL_WITH_EC=y
# CONFIG_OPENSSL_WITH_EC2M is not set
# CONFIG_OPENSSL_WITH_SSL3 is not set
# CONFIG_OPENSSL_WITH_DEPRECATED is not set
CONFIG_OPENSSL_NO_DEPRECATED=y
# CONFIG_OPENSSL_WITH_DTLS is not set
# CONFIG_OPENSSL_WITH_COMPRESSION is not set
CONFIG_OPENSSL_WITH_NPN=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SRP=y
# CONFIG_OPENSSL_HARDWARE_SUPPORT is not set
# CONFIG_OPENSSL_OPTIMIZE_SPEED is not set
# CONFIG_OPENSSL_ENGINE_CRYPTO is not set
# CONFIG_PACKAGE_libwolfssl is not set
```

neheb, 2018-11-15:
Oh right. He's talking about the OpenSSL patch for 1.1 on the main repo.

neheb, 2018-11-16:
ACK. The original code is totally wrong.

frymezim, 2018-11-18:
Python Patches are working fine.
Compiled & Tested NSA325 / Kirkwood

@neheb 
Another issue I found is for OpenSSH FYI this need I'm using OpenSSL 1.1.1
```
make[5]: Leaving directory '/home/dkmori/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/openssh-without-pam/openssh-7.9p1/openbsd-compat'
arm-openwrt-linux-muslgnueabi-gcc -O3 -pipe -mcpu=cortex-a15 -mfpu=neon-vfpv4 -mfloat-abi=hard -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/home/dkmori/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/openssh-without-pam/openssh-7.9p1=openssh-7.9p1 -Wformat -Werror=format-security -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -pipe -Wall -Wpointer-arith -Wuninitialized -Wsign-compare -Wformat-security -Wsizeof-pointer-memaccess -Wno-pointer-sign -Wno-unused-result -fno-strict-aliasing -ftrapv -fno-builtin-memset -fstack-protector-strong -fPIE   -I. -I. -I/home/dkmori/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include -I/home/dkmori/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/include -I/home/dkmori/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.2.0_musl_eabi/usr/include -I/home/dkmori/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.2.0_musl_eabi/include/fortify -I/home/dkmori/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.2.0_musl_eabi/include  -D_XOPEN_SOURCE=600 -D_BSD_SOURCE -D_DEFAULT_SOURCE -DSSHDIR=\"/etc/ssh\" -D_PATH_SSH_PROGRAM=\"/usr/bin/ssh\" -D_PATH_SSH_ASKPASS_DEFAULT=\"/usr/lib/ssh-askpass\" -D_PATH_SFTP_SERVER=\"/usr/lib/sftp-server\" -D_PATH_SSH_KEY_SIGN=\"/usr/lib/ssh-keysign\" -D_PATH_SSH_PKCS11_HELPER=\"/usr/lib/ssh-pkcs11-helper\" -D_PATH_SSH_PIDDIR=\"/var/run\" -D_PATH_PRIVSEP_CHROOT_DIR=\"/var/empty\" -DHAVE_CONFIG_H -c ssh.c -o ssh.o
ssh.c: In function 'main':
ssh.c:819:8: warning: implicit declaration of function 'SSLeay_version'; did you mean 'RSA_get_version'? [-Wimplicit-function-declaration]
        SSLeay_version(SSLEAY_VERSION)
        ^~~~~~~~~~~~~~
        RSA_get_version
ssh.c:819:23: error: 'SSLEAY_VERSION' undeclared (first use in this function); did you mean 'SSH_VERSION'?
        SSLeay_version(SSLEAY_VERSION)
                       ^~~~~~~~~~~~~~
                       SSH_VERSION
ssh.c:819:23: note: each undeclared identifier is reported only once for each function it appears in
ssh.c:1040:2: warning: implicit declaration of function 'OpenSSL_add_all_algorithms'; did you mean 'ssh_OpenSSL_add_all_algorithms'? [-Wimplicit-function-declaration]
  OpenSSL_add_all_algorithms();
  ^~~~~~~~~~~~~~~~~~~~~~~~~~
  ssh_OpenSSL_add_all_algorithms
ssh.c:1041:2: warning: implicit declaration of function 'ERR_load_crypto_strings'; did you mean 'ERR_load_DH_strings'? [-Wimplicit-function-declaration]
  ERR_load_crypto_strings();
  ^~~~~~~~~~~~~~~~~~~~~~~
  ERR_load_DH_strings
make[4]: *** [Makefile:160: ssh.o] Error 1
make[4]: Leaving directory '/home/dkmori/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/openssh-without-pam/openssh-7.9p1'
make[3]: *** [Makefile:276: /home/dkmori/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/openssh-without-pam/openssh-7.9p1/.built] Error 2
make[3]: Leaving directory '/home/dkmori/openwrt/feeds/packages/net/openssh'
time: package/feeds/packages/openssh/without-pam/compile#37.22#12.42#45.56
make[2]: *** [package/Makefile:109: package/feeds/packages/openssh/compile] Error 2
make[2]: Leaving directory '/home/dkmori/openwrt'
make[1]: *** [package/Makefile:103: /home/dkmori/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/dkmori/openwrt'
make: *** [/home/dkmori/openwrt/include/toplevel.mk:218: world] Error 2
```

neheb, 2018-11-18:
Most parts of the OpenSSH stuff have been fixed upstream. Just needs to be backported. Or wait for a new version to be released.

tripolar, 2018-11-18:
@frymezim this code is part of 7.9_p1

frymezim, 2018-11-20:
@neheb 
I got an error in wget

```
make[3]: Entering directory '/home/test/openwrt/feeds/packages/net/wget'
rm -f /home/test/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wget-ssl/wget-1.19.5/.configured_*
rm -f /home/test/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.wget.ssl_installed
(cd /home/test/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wget-ssl/wget-1.19.5/./; if [ -x ./configure ]; then find /home/test/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wget-ssl/wget-1.19.5/ -name config.guess | xargs -r chmod u+w; find /home/test/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wget-ssl/wget-1.19.5/ -name config.guess | xargs -r -n1 cp --remove-destination /home/test/openwrt/scripts/config.guess; find /home/test/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wget-ssl/wget-1.19.5/ -name config.sub | xargs -r chmod u+w; find /home/test/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wget-ssl/wget-1.19.5/ -name config.sub | xargs -r -n1 cp --remove-destination /home/test/openwrt/scripts/config.sub; AR="arm-openwrt-linux-muslgnueabi-gcc-ar" AS="arm-openwrt-linux-muslgnueabi-gcc -c -O3 -pipe -mcpu=cortex-a15 -mfpu=neon-vfpv4 -mfloat-abi=hard -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/home/test/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wget-ssl/wget-1.19.5=wget-1.19.5 -Wformat -Werror=format-security -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=arm-openwrt-linux-muslgnueabi-ld NM="arm-openwrt-linux-muslgnueabi-gcc-nm" CC="arm-openwrt-linux-muslgnueabi-gcc" GCC="arm-openwrt-linux-muslgnueabi-gcc" CXX="arm-openwrt-linux-muslgnueabi-g++" RANLIB="arm-openwrt-linux-muslgnueabi-gcc-ranlib" STRIP=arm-openwrt-linux-muslgnueabi-strip OBJCOPY=arm-openwrt-linux-muslgnueabi-objcopy OBJDUMP=arm-openwrt-linux-muslgnueabi-objdump SIZE=arm-openwrt-linux-muslgnueabi-size CFLAGS="-O3 -pipe -mcpu=cortex-a15 -mfpu=neon-vfpv4 -mfloat-abi=hard -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/home/test/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wget-ssl/wget-1.19.5=wget-1.19.5 -Wformat -Werror=format-security -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-O3 -pipe -mcpu=cortex-a15 -mfpu=neon-vfpv4 -mfloat-abi=hard -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/home/test/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wget-ssl/wget-1.19.5=wget-1.19.5 -Wformat -Werror=format-security -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/test/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include -I/home/test/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/include -I/home/test/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.2.0_musl_eabi/usr/include -I/home/test/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.2.0_musl_eabi/include/fortify -I/home/test/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.2.0_musl_eabi/include " LDFLAGS="-L/home/test/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib -L/home/test/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/lib -L/home/test/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.2.0_musl_eabi/usr/lib -L/home/test/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.2.0_musl_eabi/lib -fpic -specs=/home/test/openwrt/include/hardened-ld-pie.specs -znow -zrelro "  ac_cv_header_uuid_uuid_h=no  ./configure --target=arm-openwrt-linux --host=arm-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  --disable-rpath --disable-iri --with-included-libunistring --without-libuuid --without-libpsl --with-ssl=openssl --with-libssl-prefix="/home/test/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr" ; fi; )
configure: loading site script /home/test/openwrt/include/site/arm
configure: configuring for GNU Wget 1.19.5
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for arm-openwrt-linux-strip... arm-openwrt-linux-muslgnueabi-strip
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... arm-openwrt-linux-gnu
checking whether make supports nested variables... (cached) yes
checking for arm-openwrt-linux-gcc... arm-openwrt-linux-muslgnueabi-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether arm-openwrt-linux-muslgnueabi-gcc accepts -g... yes
[... cut ...]
admin/           devel/           .github/         .keys/           LICENSE          net/             .travis_do.sh    
.circleci/       fonts/           ipv6/            lang/            mail/            README.md        .travis.yml      
CONTRIBUTING.md  .git/            kernel/          libs/            multimedia/      sound/           utils/           
test@test-virtual-machine:~/openwrt$ nano feeds/packages/net/wget/patches/010-openssl-Do-not-use-engines-when-OpenSSL-does-not-sup.patch 
```

yousong, 2019-01-05:
@frymezim , you will need to find out why wget configure cannot find `libssl`, probably by checking config.log in the build dir.

yousong, 2019-01-05:
I'd like to also note that the 3 packages in the original report now has support for building without openssl deprecated apis.  I think @neheb does most of the work here, thank you.

closed 2019-01-05 by yousong