openwrt/packages #7416: forked-daapd: build failing for ipq806x?

dwb, 2018-11-13:
Maintainer: @ejurgensen
Environment: OpenWRT 18.06.1 r7258-5eb055306f on `ipq806x` `arm_cortex-a15_neon-vfpv4`
Description:
`forked-daapd` v26.4 is currently unavailable for the above environment, the latest version being v25.0. Is the build failing for this target? I tried to run it individually, using your CI Docker image, the CI build scripts, and changing the SDK base URL for the `ipq806x` one, and indeed it doesn't succeed, though I can't see why. I include the tail end of the transcript here. I'm afraid I don't have the entire initial build output, but on re-running it, this snippet contains the only occurrences of the string `forked-daapd`.

Thanks for any help!

```
find /root/build_dir/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/libwebsockets-full/libwebsockets-3.0.1 -mindepth 1 -maxdepth 1 -not '(' -type f -and -name '.*' -and -size 0 ')' -and -not -name '.pkgdir' | xargs -r rm -rf
make[2]: Leaving directory '/root/openwrt_packages/libs/libwebsockets'
time: package/feeds/packages/libwebsockets/full/compile#18.11#2.90#26.67
make[2]: Entering directory '/root/openwrt_packages/sound/forked-daapd'
touch -r /root/build_dir/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/forked-daapd-26.4/.built /root/build_dir/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/forked-daapd-26.4/.autoremove 2>/dev/null >/dev/null
Makefile:79: recipe for target 'compile' failed
make[2]: [compile] Error 1 (ignored)
find /root/build_dir/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/forked-daapd-26.4 -mindepth 1 -maxdepth 1 -not '(' -type f -and -name '.*' -and -size 0 ')' -and -not -name '.pkgdir' | xargs -r rm -rf
/root/build_dir/staging_dir/host/bin/find: '/root/build_dir/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/forked-daapd-26.4': No such file or directory
make[2]: Leaving directory '/root/openwrt_packages/sound/forked-daapd'
time: package/feeds/packages/forked-daapd/compile#0.05#0.01#0.28
make[1]: Leaving directory '/root/build_dir'
root@a82373003c99:~/build_dir# echo $?
127
```

thess, 2018-11-14:
Yes, you are correct - v26.4 is the current version in the trunk/snapshots branch. v25.0 is the version released in 18.06 packages. In general, we do not automatically backport newer package versions to released branches. We do consider critical errors (preventing use) and security patches as necessary and time-permitting.

If I read your comments correctly, you are interested in building v26.4 using the 18.06 SDK? I that is true, youj would be better served on our forum (https://forum.openwrt.org) and asking for help there. General package support here is sketchy at best.

neheb, 2020-01-12:
This has been fixed.

closed 2020-01-12 by neheb