openwrt/packages #7420: Moving docker images to openwrt Docker Hub repository.

thess, 2018-11-14:
@champtar - If I can successfully get the openwrt account on Docker Hub set as an organization and with teams similar to what we do here on GitHub, would you like to move your CircleCI Docker container to a project repo? I will add the same Owners and Members (or an initial sub-set) as we have here.

Currently there is an `openwrt` account on Docker Hub with no repositories which was created 3yrs ago and shows no activity. I have contacted Docker Hub support to claim this ID for our use so stay tuned.

Any objections?

jow-, 2018-11-17:
Please keep Debian as base for now, its also currently is used as baseline for release builds.

While I have no objects to Alpine per-se, I think its better to concentrate on our target related breakage and not fight with musl/Alpine host related quirks.

thess, 2018-11-17:
@diizzyy - Buildbots are Debian and they work. I see no compelling reason to take on the headaches of another platform.

diizzyy, 2018-11-17:
It was merely a suggestion as more and more projects are moving that way due to overhead as far as CI-goes and there are people on musl hosts trying to get OpenWrt working from time to time so it might be of interest for contributors. 

@thess 
Most of the time that is...

champtar, 2018-11-17:
@thess no objections at all, I just went the easy way at first. I recommend to continue using container digest even if it means asking people to rebase sometimes
@diizzyy from what I remember container download is super fast

thess, 2018-12-05:
After not being able to get a response from Docker Hub support, I have created the **openwrtorg** organization repository for our use. Currently, @champtar, @yousong and myself have admin/checkin rights. 

PR #7588 updates the CircleCI docker image location and other script updates

Anyone in the @members group here AND who has a Docker Hub account can be added to the team which has check-in privs in this new repository (https://hub.docker.com/r/openwrtorg/) - just ask.

thess, 2018-12-09:
CircleCI config seems to be functional with new Docker images and **openwrtorg** Docker Hub repo - closing this

closed 2018-12-09 by thess

diizzyy, 2019-01-02:
Just as reference, compiling master and "various" packages (additional from package repo) works fine without any additional patches but I'm not going to claim that everything works.