openwrt/packages #7433: openvpn (in r8459-3a6bddd)

mistepien, 2018-11-15:
Nov 15 11:16:27 thorgal uhttpd[1440]: /usr/lib/lua/luci/dispatcher.lua:565: Invalid controller file found
Nov 15 11:16:27 thorgal uhttpd[1440]: The file '/usr/lib/lua/luci/controller/openvpn.lua' contains an invalid module line.
Nov 15 11:16:27 thorgal uhttpd[1440]: Please verify whether the module name is set to 'luci.controller.openvpn' - It must correspond to the file path!
Nov 15 11:16:27 thorgal uhttpd[1440]: stack traceback:
Nov 15 11:16:27 thorgal uhttpd[1440]: #011[C]: in function 'assert'
Nov 15 11:16:27 thorgal uhttpd[1440]: #011/usr/lib/lua/luci/dispatcher.lua:565: in function 'createindex'
Nov 15 11:16:27 thorgal uhttpd[1440]: #011/usr/lib/lua/luci/dispatcher.lua:591: in function 'createtree'
Nov 15 11:16:27 thorgal uhttpd[1440]: #011/usr/lib/lua/luci/dispatcher.lua:262: in function 'dispatch'
Nov 15 11:16:27 thorgal uhttpd[1440]: #011/usr/lib/lua/luci/dispatcher.lua:127: in function </usr/lib/lua/luci/dispatcher.lua:126>

dibdot, 2018-11-15:
Wrong repo, please report LuCI issues in the corresponded luci repo. 
Anyway to fix your issue, reset your LuCI caches `rm -rf /tmp/luci-*`

dibdot, 2018-11-17:
No more feedback - seems to be fixed.

closed 2018-11-17 by dibdot