openwrt/packages #7457: simple adblock problem in openwrt

jkle112, 2018-11-17:
running error 
[image]
[image]

dibdot, 2018-11-17:
Please always inform the maintainer: @stangri

stangri, 2018-11-20:
@jkle112 -- set verbosity to 2 and check the logs to see which list failed to download. Download errors are not the package bugs, provided you find out which list is failing to download, you may want to seek additional help at forums.

dibdot, 2018-11-21:
as @stangri said, probably a temporary download issue.

closed 2018-11-21 by dibdot

jkle112, 2018-12-09:
The same problem in LEDE firmware!

[image]