openwrt/packages #7470: strongswan: OpenWrt 18.06 branch missing upstream security fixes

hailfinger, 2018-11-18:
Maintainer: @dedeckeh @stintel 
Environment: OpenWrt 18.06 branch

Description:
AFAICS the patches for CVE-2018-16151, CVE-2018-16152 and CVE-2018-17540 are missing from the strongswan 5.6.3 packages as shipped in OpenWrt 18.06.
strongswan 5.7.1 as shipped in master is not affected.
Vendor info:
https://www.strongswan.org/blog/2018/09/24/strongswan-vulnerability-(cve-2018-16151,-cve-2018-16152).html
https://www.strongswan.org/blog/2018/10/01/strongswan-vulnerability-(cve-2018-17540).html
Patches (both patches needed):
https://download.strongswan.org/patches/27_gmp_pkcs1_verify_patch/
https://download.strongswan.org/patches/28_gmp_pkcs1_overflow_patch/

dedeckeh, 2018-11-19:
@stintel Any objection if I cherry-pick https://github.com/openwrt/packages/commit/fe973d181b9d1825d547f2ee806fe1e4e22bb301 to 18.906 branch ?

stintel, 2018-11-19:
@dedeckeh ACK

dedeckeh, 2018-11-19:
Done in commit https://github.com/openwrt/packages/commit/b6f6df1d771ebbfe1a4ca0b498443754ccd9ef9c

closed 2018-11-19 by dedeckeh