openwrt/packages #7472: 6in4 - Troubles with multiple DHCP wans and secure updates

SeSeKenny, 2018-11-19:
Maintainer: @jow- 
Environment: any, OWRT 18.06.1

Description:
I have been dealing with a weird issue for sometime now and thought I would try and get some traction here. Basically I have 2 dhcp based wans and 1 static wan. If I avoid all dynamic updating I can get things to work (although broken local addressing gets assigned to the interface (link/sit address)). What I am hoping to do is assign an account/blocks to each wan with dynamic updating still intact.

I believe the problems are 2 fold,

    my wan seems to swing between any interface I have with default routes (I would think this should reply with the lowest metric'd default route but :/) as returned via the __network_wan function. This ultimately changes the ipaddr (? my bash skills are bad) being utilized by 6in4.sh

    uclient-fetch as utilized in 6in4.sh does not appear to be using any sort of source address or interface pinning, meaning it is impossible to pin more than 1 connection with this scripting (even now I am putting in static routes for the update addressing to force it out the interface I desire)

I realize I could script my way to success outside of luci and the 6in4 package/uci network architecture, but i feel like this can be solved quite easily with some trivial line changes.

I am thinking putting two optional parameters against the 6in4 interface type. 1 being the interface name which will be used for ipaddr and uclient-fetch resolution. 2 being perhaps the index of ipv4-address so that a static interface defined with multiple addresses can be used as well.

As an aside, personally I think that any tunnel type connection should have these options. It would make things much more flexible and basically put feature parity on some of these things higher than that of the pfsenses of the world :)

SeSeKenny, 2018-11-19:
Looking at this a bit further it looks like an undocumented (for this interface type anyways) parameter tunlink may be partially implemented? libuclient also appears to not have the option for source interface/address selection perhaps some conditional logic for using wget instead if the parameter is specified?

neheb, 2020-01-28:
6in4 issues should be reported under https://bugs.openwrt.org/

closed 2020-01-28 by neheb