openwrt/packages #7473: upmpdcli crashes on sunxi

rvbglas, 2018-11-19:
Maintainer: @ignisf
Environment: sunxi, H2+, Orange Pi Zero

Description: upmpdcli crashes on start when creating worker threads

```
Thread 1 "upmpdcli" received signal SIGILL, Illegal instruction.
__pthread_timedjoin_np (t=0xb6c71d5c, res=res@entry=0x0, at=at@entry=0x0) at src/thread/pthread_join.c:15
15			if (state >= DT_DETACHED) a_crash();
(gdb) backtrace 
#0  __pthread_timedjoin_np (t=0xb6c71d5c, res=res@entry=0x0, at=at@entry=0x0) at src/thread/pthread_join.c:15
#1  0xb6fd6870 in __pthread_join (t=<optimized out>, res=res@entry=0x0) at src/thread/pthread_join.c:28
#2  0xb6fd6614 in __pthread_detach (t=<optimized out>) at src/thread/pthread_detach.c:11
#3  0xb6c737a4 in CreateWorker (tp=tp@entry=0xb6d38130 <gSendThreadPool>) at src/ThreadPool.c:654
#4  0xb6c74574 in ThreadPoolInit (tp=0xb6d38130 <gSendThreadPool>, tp@entry=0xb6d21cb8 <UpnpInit+356>, attr=attr@entry=0xbefff7a8) at src/ThreadPool.c:751
#5  0xb6d21cb8 in UpnpInitThreadPools () at src/api/upnpapi.c:291
#6  UpnpInitPreamble () at src/api/upnpapi.c:372
#7  UpnpInit (HostIP=0xbefff850 "192.168.1.161", DestPort=DestPort@entry=49152) at src/api/upnpapi.c:464
#8  0xb6f40f4c in UPnPP::LibUPnP::LibUPnP (this=0xb6d386f0, serveronly=<optimized out>, hwaddr=0xbefffb6c, ifname=..., inip=..., port=49152) at libupnpp/upnpplib.cxx:144
#9  0xb6f422d0 in UPnPP::LibUPnP::getLibUPnP (serveronly=serveronly@entry=true, hwaddr=0xbefffb6c, hwaddr@entry=0xbefffb64, ifname=..., ip=..., port=49152, port@entry=6600) at libupnpp/upnpplib.cxx:78
#10 0x0001ab3c in main (argc=0, argv=<optimized out>) at src/main.cxx:493

```

diizzyy, 2018-12-30:
@rvbglas @ignisf 
Can you give it a try applying these patches?
https://github.com/diizzyy/packages/commit/6c35f55cdf713a0f0ba2a07ebc8b0d4bf22261e8.patch
https://github.com/diizzyy/packages/commit/c2c4054d21cbb4fe4bdf792effe7c1ea313d6b52.patch
https://github.com/diizzyy/packages/commit/a6638b6627dd20560865a3a4fb414919c9871fc1.patch

diizzyy, 2019-01-03:
Illegal instruction here as well (mvebu) :(

Current version (1.3.8) fails to compile here:
https://opensourceprojects.eu/p/upmpdcli/code/ci/b2af539c29c3aac57a78465aae874595aaa93f03/tree/src/mediaserver/cdplugins/curlfetch.cpp#l14

diizzyy, 2019-01-13:
```
Starting program: /usr/bin/upmpdcli -c /etc/upmpdcli.conf
:4:src/mediaserver/contentdirectory.cxx:179::ContentDirectory: not creating entry for gmusic because gmusicuser is not defined in configuration
:4:src/mediaserver/contentdirectory.cxx:179::ContentDirectory: not creating entry for qobuz because qobuzuser is not defined in configuration
:4:src/mediaserver/contentdirectory.cxx:179::ContentDirectory: not creating entry for tidal because tidaluser is not defined in configuration
:4:src/mediaserver/contentdirectory.cxx:179::ContentDirectory: not creating entry for uprcl because uprcluser is not defined in configuration
[New LWP 2565]

Thread 1 "upmpdcli" received signal SIGILL, Illegal instruction.
__pthread_timedjoin_np (t=0xb6b6ed5c, res=res@entry=0x0, at=at@entry=0x0) at src/thread/pthread_join.c:15
15      src/thread/pthread_join.c: No such file or directory.
(gdb) bt
#0  __pthread_timedjoin_np (t=0xb6b6ed5c, res=res@entry=0x0, at=at@entry=0x0) at src/thread/pthread_join.c:15
#1  0xb6fd68f8 in __pthread_join (t=<optimized out>, res=res@entry=0x0) at src/thread/pthread_join.c:28
#2  0xb6fd669c in __pthread_detach (t=<optimized out>) at src/thread/pthread_detach.c:11
#3  0xb6e307a4 in CreateWorker (tp=tp@entry=0xb6e73130 <gSendThreadPool>) at src/ThreadPool.c:654
#4  0xb6e31574 in ThreadPoolInit (tp=0xb6e73130 <gSendThreadPool>, tp@entry=0xb6e5ccb8 <UpnpInit+356>, attr=attr@entry=0xbefff690) at src/ThreadPool.c:751
#5  0xb6e5ccb8 in UpnpInitThreadPools () at src/api/upnpapi.c:291
#6  UpnpInitPreamble () at src/api/upnpapi.c:372
#7  UpnpInit (HostIP=0x0, DestPort=DestPort@entry=1) at src/api/upnpapi.c:464
#8  0xb6f612f8 in UPnPP::LibUPnP::LibUPnP (this=0xb6ffcbf0, serveronly=<optimized out>, hwaddr=<optimized out>, ifname=..., inip=..., port=0) at libupnpp/upnpplib.cxx:163
#9  0xb6f61e8c in UPnPP::LibUPnP::getLibUPnP (serveronly=serveronly@entry=true, hwaddr=0xbefffaf0, hwaddr@entry=0xbefffae8, ifname=..., ip=..., port=port@entry=0) at libupnpp/upnpplib.cxx:80
#10 0x0001e6a0 in main (argc=<optimized out>, argv=<optimized out>) at src/main.cxx:603
(gdb) disas
Dump of assembler code for function __pthread_timedjoin_np:
   0xb6fd6818 <+0>:     push    {r4, r5, r6, r7, r8, r9, lr}
   0xb6fd681c <+4>:     sub     sp, sp, #20
   0xb6fd6820 <+8>:     mov     r5, r0
   0xb6fd6824 <+12>:    mov     r6, r1
   0xb6fd6828 <+16>:    mov     r7, r2
   0xb6fd682c <+20>:    bl      0xb6fd79a8 <__pthread_testcancel>
   0xb6fd6830 <+24>:    add     r1, sp, #12
   0xb6fd6834 <+28>:    mov     r0, #1
   0xb6fd6838 <+32>:    bl      0xb6fd762c <__pthread_setcancelstate>
   0xb6fd683c <+36>:    ldr     r1, [sp, #12]
   0xb6fd6840 <+40>:    cmp     r1, #0
   0xb6fd6844 <+44>:    bne     0xb6fd6850 <__pthread_timedjoin_np+56>
   0xb6fd6848 <+48>:    mov     r0, r1
   0xb6fd684c <+52>:    bl      0xb6fd762c <__pthread_setcancelstate>
   0xb6fd6850 <+56>:    mov     r4, #0
   0xb6fd6854 <+60>:    add     r8, r5, #36     ; 0x24
   0xb6fd6858 <+64>:    mov     r9, r4
   0xb6fd685c <+68>:    ldr     r1, [r5, #36]   ; 0x24
   0xb6fd6860 <+72>:    cmp     r1, #0
   0xb6fd6864 <+76>:    bne     0xb6fd688c <__pthread_timedjoin_np+116>
   0xb6fd6868 <+80>:    ldr     r0, [sp, #12]
   0xb6fd686c <+84>:    mov     r1, #0
   0xb6fd6870 <+88>:    bl      0xb6fd762c <__pthread_setcancelstate>
   0xb6fd6874 <+92>:    cmp     r4, #22
   0xb6fd6878 <+96>:    cmpne   r4, #110        ; 0x6e
   0xb6fd687c <+100>:   bne     0xb6fd68c0 <__pthread_timedjoin_np+168>
   0xb6fd6880 <+104>:   mov     r0, r4
   0xb6fd6884 <+108>:   add     sp, sp, #20
   0xb6fd6888 <+112>:   pop     {r4, r5, r6, r7, r8, r9, pc}
   0xb6fd688c <+116>:   cmp     r4, #110        ; 0x6e
   0xb6fd6890 <+120>:   cmpne   r4, #22
   0xb6fd6894 <+124>:   beq     0xb6fd6868 <__pthread_timedjoin_np+80>
   0xb6fd6898 <+128>:   cmp     r1, #2
   0xb6fd689c <+132>:   ble     0xb6fd68a4 <__pthread_timedjoin_np+140>
=> 0xb6fd68a0 <+136>:   udf     #0
   0xb6fd68a4 <+140>:   str     r9, [sp]
   0xb6fd68a8 <+144>:   mov     r3, r7
   0xb6fd68ac <+148>:   mov     r2, #0
   0xb6fd68b0 <+152>:   mov     r0, r8
   0xb6fd68b4 <+156>:   bl      0xb6fd45dc <__timedwait_cp>
   0xb6fd68b8 <+160>:   mov     r4, r0
   0xb6fd68bc <+164>:   b       0xb6fd685c <__pthread_timedjoin_np+68>
   0xb6fd68c0 <+168>:   dmb     ish
   0xb6fd68c4 <+172>:   cmp     r6, #0
   0xb6fd68c8 <+176>:   ldrne   r3, [r5, #76]   ; 0x4c
   0xb6fd68cc <+180>:   strne   r3, [r6]
   0xb6fd68d0 <+184>:   ldr     r0, [r5, #48]   ; 0x30
   0xb6fd68d4 <+188>:   cmp     r0, #0
   0xb6fd68d8 <+192>:   moveq   r4, r0
   0xb6fd68dc <+196>:   beq     0xb6fd6880 <__pthread_timedjoin_np+104>
   0xb6fd68e0 <+200>:   ldr     r1, [r5, #52]   ; 0x34
   0xb6fd68e4 <+204>:   mov     r4, #0
   0xb6fd68e8 <+208>:   bl      0xb6fb6480 <__munmap>
   0xb6fd68ec <+212>:   b       0xb6fd6880 <__pthread_timedjoin_np+104>
End of assembler dump.
```
So "udf" --> undefined is bad, not sure how to fix however...

ignisf, 2019-01-19:
Hi all, sorry for the late response. I've been unwilling to update this package as the new versions in my opinion don't add enough new functionality to warrant the addition of their dependencies... I am not sure if I can help with this current issue though

diizzyy, 2019-01-27:
Works with the following PRs (I haven't tested old versions):

https://github.com/openwrt/packages/pull/8043
https://github.com/openwrt/packages/pull/8044
https://github.com/openwrt/packages/pull/8045
https://github.com/openwrt/packages/pull/8046
https://github.com/openwrt/packages/pull/8047

diizzyy, 2019-01-31:
@thess @hnyman 
Close, works now :-)

diizzyy, 2019-03-28:
Resolved, closing

closed 2019-03-28 by diizzyy