openwrt/packages #7474: mwan3: About policy route

leotian111, 2018-11-19:
Hi, I'm testing newest version 2.7.6 of mwan3.
But I can't succeed through lan.  I must modify from "iif $2 lookup $id" to "iif $2 lookup main" in shell function mwan3_create_iface_rules. then we can work in br-lan

I wonder if there are any special settings! 
thks!!

dibdot, 2018-11-20:
@feckert

leotian111, 2018-11-20:
@feckert

leotian111, 2018-11-20:
@dibdot  
thks

leotian111, 2018-11-20:
@feckert
Hi, Can I ask you a question about mwan3-2.7.6?

leotian111, 2018-11-20:
@feckert
Hi, I'm testing newest version 2.7.6 of mwan3.
But I can't succeed through lan. I must modify from "iif $2 lookup $id" to "iif $2 lookup main" in shell function mwan3_create_iface_rules. then we can work in br-lan

I wonder if there are any special settings!
thks!!

leotian111, 2018-11-20:
I have known reason, mwan3rtmon not work proper,  not update direct route into $id rules!

closed 2018-11-20 by leotian111

feckert, 2018-11-21:
@leotian111 did you solved your issue?

leotian111, 2018-11-21:
yes, thank you,  i see the shell function mwan3_rtmon_ipv4 detailed and found that we must have option family in file /etc/config/mwan3,  after adding it the problem solved!