openwrt/packages #7494: Time based rule on mwan3

speedster-pta, 2018-11-21:
Please make sure that the issue subject starts with `<package-name>: `

This repo here is only for packages maintained in this repo.  For base packages residing in the same repo as the build system and maintained by core devs, please consider opening tickets there for more timely responses

 - OpenWrt base system: https://bugs.openwrt.org
 - Most LuCI packages: https://github.com/openwrt/luci/issues

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

```
Formating code blocks by wrapping them with pairs of ```
```

gitoriu, 2018-11-28:
+1 for time based rules, like  firwall iptables time based rules, options for:

- Week Days
- Month Days
- Start Time (hh:mm:ss)
- Stop Time (hh:mm:ss)
- Start Date (yyyy-mm-dd)
- Stop Date (yyyy-mm-dd=

feckert, 2018-11-30:
I think this is a good feature request I will have to think about this.

gishanc, 2025-08-08:
This is the most wanted feature right now. Did you think about it?