openwrt/packages #7501: luci-mwan3 - adding ubus status => online + offline to status page)

camelcamro, 2018-11-22:
Please make sure that the issue subject starts with `<package-name>: `

@feckert
tested via ubus and is fine to collect the wanted informations to show them on the detail status page of mwan3.

ubus call mwan3 status
to get infos for the mwan3 status interface ONLINE / DOWNTIME in sec.

ubus call network.interface.4G status
to get infos for the mwan3 status interface UPTIME "online" in sec. of interface itself

would be nice, if you can now add it to the luci-mwan3 package to get the online+downline time + interface uptime
like as described above:

the result outcome should be to see on mwan3 status page in luci:
====================
Interface status:
interface 4G is unknown and tracking is down
interface 4GA is unknown and tracking is down
interface VPN_ROUT02 is online 00h:01m:18s , uptime 00h:34m:58s and tracking is active
interface VPN_ROUT03 is offline 00h:03m:45s , uptime 02h:24m:28s and tracking is active
interface VPN_ROUT04 is online 00h:00m:14s , uptime 00h:34m:58s and tracking is active

feckert, 2018-12-30:
@camelcamro I have added this future to luci-app-mwan3 so I think we could close this feature request

camelcamro, 2018-12-31:
this shows then on the detail page: like:
192.168.1.3/cgi-bin/luci/admin/status/mwan/detail

camelcamro, 2019-01-08:
yes, but only in the OVERVIEW page, and there is only the ONLINE timem, which is OK for there.
But in the "detail tab" - there is the infos missing like as i described ...
maybe not the best method to show the values, as you have now a ubus call to get the information, but mine is fine working and is looking like that:
on "mwan3 DETAIL" tab (/cgi-bin/luci/admin/status/mwan/detail)

it is looking like that, which gives much more info to see:

Interface status:
 interface 4G is online 07h:37m:56s , uptime 17h:28m:01s and tracking is active
 interface VPN_ROUT03 is online 07h:37m:28s , uptime 11h:44m:55s and tracking is active
 interface VPN_ROUT04 is unknown and tracking is down
 interface VPN_ROC is unknown and tracking is down
 interface VPN_ROC_AT is online 07h:36m:57s , uptime 17h:16m:24s and tracking is active
 interface VPN_OTA is online 07h:36m:25s , uptime 17h:28m:01s and tracking is active
 interface VPN_I_AT is unknown and tracking is down
 interface VPN_I_HU is online 07h:35m:54s , uptime 14h:30m:31s and tracking is active

pls add this infos to the luci "detail" view.

feckert, 2019-05-25:
This is a output from the mwan3 packages itself. So we have add this to the mwan3 tool not to Luci

closed 2019-05-25 by feckert

camelcamro, 2019-05-25:
i guess it is fine as it is ...
it is only needed for the detail page in luci, nothing else.
otherwise it would be an overkill to have the detail information everywhere.
i would recommend as i provided only to add this to luci 
see my patch as given from "camelcamro commented on 31 Dec 2018"

see: /lib/mwan3/mwan3.sh - around line 1071

	if [ -z "$id" -o -z "$device" ]; then
		result="unknown"
	elif [ -n "$($IP rule | awk '$1 == "'$(($id+1000)):'"')" ] && \
		[ -n "$($IP rule | awk '$1 == "'$(($id+2000)):'"')" ] && \
		[ -n "$($IPT -S mwan3_iface_in_$1 2> /dev/null)" ] && \
		[ -n "$($IP route list table $id default dev $device 2> /dev/null)" ]; then
+ # camel
- 		result="$(mwan3_get_iface_hotplug_state $1)"
+		ifuptime=`/sbin/ifstatus $1 | grep 'uptime' | grep -o '[0-9]\+'`
+		ifonline=$(($(date +%s) - $(date +%s -r /var/run/mwan3/iface_state/$1)))
+		result="$(mwan3_get_iface_hotplug_state $1) $(printf '%02dh:%02dm:%02ds\n' $(($ifonline/3600)) $(($ifonline%3600/60)) $(($ifonline%60))) , uptime $(printf '%02dh:%02dm:%02ds\n' $(($ifuptime/3600)) $(($ifuptime%3600/60)) $(($ifuptime%60)))"

	elif [ -n "$($IP rule | awk '$1 == "'$(($id+1000)):'"')" ] || \