openwrt/packages #7519: mwan3: Comparing two wan's delay, And make the lowest delay of wan as main output interface dynamicly!

leotian111, 2018-11-26:
@feckert
hi, Do you any plan to realize the following functionality?
we have two LTES on our board.  the 2 SIMs are from different ISPs,  we want to use the lte with lowest delay  as the only output interface.  do you have any better methods to compare two lte's delay, And then make the decision dynamically!
Thanks a lot!

obsy, 2018-11-26:
Or other things like roaming, signal strength, technology etc.

feckert, 2018-11-26:
I don't think that's suitable for the mwan3 package.
These are mobile phone parameters.
But you could think about implementing something like this in a separate service that makes the decision and then tells this to mwan3.

obsy, 2018-11-26:
This is probably best option, in this way we can also implement https://github.com/openwrt/packages/issues/7494

leotian111, 2018-11-26:
@feckert 
Thank you!
firstly there is not relavent lte mobile parameters.  we want to make the decision 
 only according to the tcp/ip delay.  ok i'll write a new mwan3track by myself! 
Thank you again if you have any plans about this!

feckert, 2018-11-26:
@leotian111 
Ok if you want to to network decision this could go into mwan3track.
But I though you want to use LTE parameters