openwrt/packages #7526: mwan3: httping not detecting when interface up again

mistoll, 2018-11-27:
Maintainer: @feckert 
Environment: Linksys WRT1200AC / ARMv7 Processor rev 1 (v7l) / OpenWrt 18.06.1 r7258-5eb055306f 

Description:

I'm using mwan3 with a wired and a mobile broadband connection. The mobile broadband is used as a fallback, when the wired connection is unavailable. That works fine using standard ping. However the wired connection isn't under my control and somtimes there's a firewall there, blocking everything except of HTTP traffic.

So I tried to use httping. With httping as a tracking method the interface state is detected as online, as long the connection is available. When the connection is disconnected, the interface state goes to offline. But when the connection is working again, the interface won't go online.

Test output in that state, on a network where both http and ping are allowed:
```
root@charlie-wlan:~# ifconfig
...
eth1      Link encap:Ethernet  HWaddr ...
inet addr:10.31.0.217  Bcast:10.31.0.255  Mask:255.255.255.0
...

root@charlie-wlan:~# ping -I eth1 1.1.1.1 -c 1
PING 1.1.1.1 (1.1.1.1): 56 data bytes
64 bytes from 1.1.1.1: seq=0 ttl=56 time=9.875 ms
--- 1.1.1.1 ping statistics ---
1 packets transmitted, 1 packets received, 0% packet loss
round-trip min/avg/max = 9.875/9.875/9.875 ms

root@charlie-wlan:~# httping -y 10.31.0.217 1.1.1.1
PING 1.1.1.1:80 (/):
connect time out
```

feckert, 2018-11-30:
The following lines are responsible for tracking https://github.com/openwrt/packages/blob/07e17bfe4a8760483548722b68edba2acd87902c/net/mwan3/files/lib/mwan3/mwan3.sh#L711-L722

I am not sure why this would not work with httping. Does it has todo with connection tracking?

I have not setup a httping tracking for a while. But I have to verify this next week.

feckert, 2018-11-30:
I think the problem is that we need a SRC_IP
https://github.com/openwrt/packages/blob/07e17bfe4a8760483548722b68edba2acd87902c/net/mwan3/files/usr/sbin/mwan3track#L140-L150

mistoll, 2018-12-01:
I don't think there's something wrong with the mwan3 tracking implementation. When I run the commands directly the result is the same (See listing in first comment). Thus I think it's more an issue of routing.

I can do some more tests and publish the results. I already did some experiments with nping, which also supports tcp connections and binding to a device. I think it behaved the same way as httping, however I didn't document it. I'll retry and publish it.

feckert, 2018-12-01:
> I can do some more tests and publish the results

that would be great :+1: thanks for your research.

mistoll, 2018-12-03:
I did some tests, see attachment for detailed protocol ([mwan3.txt](https://github.com/openwrt/packages/files/2640984/mwan3.txt))

* The tests were done on a different external network than the first test, so is seems not to be an issue of the external network.
* The value of flush_conntrack has no influence. 'never', 'always' and no value give the same results.
* Also tried nping (--tcp and --icmp), which fails in the same cases as httping, however there's more information.
* Most interesing is the influence of mwan3.globals.local_source. All tests were done with this config set to lan. If set to none httping behaves as expected.

So setting mwan3.globals.local_source=none could be a good workaround. IMO ping and httping behave differently when binding to an interface. Maybe ping ignores the default route and httping (and nping) doesn't. Have a look at the nping output, that could be helpfull:
```
root@charlie-wlan:~# nping -c 1 -e eth1 1.1.1.1 --tcp
sendto in send_ip_packet_sd: sendto(4, packet, 40, 0, 1.1.1.1, 16) => Network unreachable
Offending packet: TCP 192.168.1.18:48229 > 1.1.1.1:80 S ttl=64 id=43516 iplen=40  seq=905292359 win=1480 
SENT (0.0922s) TCP 192.168.1.18:48229 > 1.1.1.1:80 S ttl=64 id=43516 iplen=40  seq=905292359 win=1480 
```

feckert, 2018-12-04:
Thanks again  for the very well prepared logs.

> Most interesing is the influence of mwan3.globals.local_source. All tests were done with this config set to lan. If set to none httping behaves as expected.

The local_source issue is deprecated in master now. This should not be used anymore. There were a lot of problems with this config option. Does nping work too?

The option **local_source** was used to also have a fail-over and load-balancing on router initiated traffic. But this introduced a lot problems. So it was removed and some changes where made (see rt_mon) so that also router initiated traffic has fail-over and load-balancing as well. So this should work out of the box.

mistoll, 2018-12-04:
Is the default (non deprecated) beahavior the same as when local_source is 'none'? In that case this issue is solved and can be closed.

AFAIK nping works too. I experimented with nping, but didn't finish it, maybe you can make use of this: 6673c2c46df7e4dfcba5fe0e49dc9df82742f6f9

feckert, 2018-12-06:
> Is the default (non deprecated) beahavior the same as when local_source is 'none'? In that case this issue is solved and can be closed.

On master **local_source** is removed so you do not need this anymore.
Load-balancing and fail-over works out of the box for router initiated traffic.

On openwrt-18.06 **local_source** is still included and is set to **lan** per default.
if you set this to **none** then you have no fail-over and load-balancing for router initiated traffic.

> AFAIK nping works too. I experimented with nping, but didn't finish it, maybe you can make use of this: 6673c2c

I will add this maybe next week to mwan3 thanks.

mistoll, 2018-12-07:
Personally I'll wait for the next release.

IMO this is solved. Thanks for your support.

closed 2018-12-07 by mistoll