openwrt/packages #7559: travis: build against xenial

pprindeville, 2018-12-02:
Maintainer: @champtar 
Environment: n/a

Description:

Now that Travis supports Xenial (16.04), can we build under that instead of Trusty Tahr (14.04)?

thess, 2018-12-02:
I don't know what Etienne thinks, but I would rather proceed with CircleCI and drop Travis altogether. The current Travis app we have setup for our repo will no longer be supported in January.

thess, 2018-12-05:
Travis CI is currently integrated as a "service" in the repo. GitHub is discontinuing this type of service integration in January (https://developer.github.com/changes/2018-04-25-github-services-deprecation/). If it is desirable to use Travis over CircleCI then our current integration needs to be updated and re-installed. I am not planning to do this and I don't think @champtar is either.

My current plan is to drop Travis integration in a week or 2 after some CircleCI updates and testing (PR #7420). Additionally, I am not sure the DCO test is providing anything over our CircleCI tests now.

champtar, 2018-12-05:
Hi @thess

My plan was always to drop Travis
Regarding DCO I would like to make it required, so we never merge commit without SoB.
Also with DCO check you know why it fail right away without doing 10 clicks to get to the logs

thess, 2018-12-05:
OK - we keep DCO.

champtar, 2018-12-05:
BTW when I say I would like to make it required, I'm not admin so if people agree I need an admin to flip the switch ;)

thess, 2018-12-05:
Roger that...

pprindeville, 2018-12-10:
Sorry, “DCO” and “SoB” are what? Also what distro is CircleCI using?

champtar, 2018-12-10:
Hi @pprindeville,
DCO == developer certificate of origin
SoB == Signed of by
CircleCi is bring your own container, right now we use Debian 9.

champtar, 2018-12-10:
https://github.com/openwrt/packages/blob/master/.circleci/

thess, 2018-12-10:
DCO: https://probot.github.io/apps/dco/

Debian 9.6 when I rebuilt the container last week. Container is in our public Docker Hub repo (see note here: https://github.com/openwrt/packages/issues/7420#issuecomment-444640533) and link provided above.

thess, 2018-12-11:
Travis is now disconnected - Closing this

closed 2018-12-11 by thess