openwrt/packages #7624: Enabling mandatory PR checks

thess, 2018-12-09:
Maintainer: @openwrt/package-maintainers / @openwrt/developer 
Environment: Package PR handling in this repo

Discussion:

CircleCI PR checks are working pretty well now. Going forward, I am planning to enable mandatory CI checks before a PR can be merged. Commits must first be pushed to another branch, then merged or pushed directly to the branch with these rules enabled after the status checks have passed. This process is pretty much as we operate now. 

Checks that will be required are DCO (signature check) and CircleCI 'buildpr' initially on the master and openwrt-18.96 branches. The CircleCI script verifies that the source package is available and tagged with the correct hash. It also checks that the PR does not contain merge commits and validates that the SoB looks like a real name and matches the commit(s) author name. Test builds are performed using the current release (branch dependent)  ar71xx/generic SDK. 

The Travis CI integration will be turned off as it is redundant and is currently a deprecated integration which will cease working next month anyway.

Bypassing these checks, if necessary, can be merged by a developer in the "admin" group.

champtar, 2018-12-10:
Thanks @thess for pushing this.
Maybe you can enable DCO on all branches?
I don't think this impact many packages, but svn doesn't produce reproducible tarball, so make check will always fail for now for those packages.

yousong, 2018-12-10:
Thank you for taking care of this.

I'd like to add a minor note that there are packages not available for ar71xx target.   utils/qemu is one at the moment.  It's rare and surely we can cope with that ;)

thess, 2018-12-10:
@champtar - Yes, enabling DCO for all branches. I'll check with @jow to see if we can do something similar for Luci and possibly other repos. 

svn non-reproducible tarballs must be problematic for the builds too however, hash checks are warnings currently...

@yousong - I would like to see if we could build more than 1 target in CI - It would help with a lot of corner-case issues that pop up from time-to-time. Also, the current script considers a package without a modified Makefile to be OK. I think in a lot of situations, this is not the case - for example, an updated patch or init script without bumping PKG_RELEASE.

I am really interested in furthering CI at least for our package repos. The buildbots take care of the targets pretty well. What would really make our lives better would be build failure notifications filtered by specific packages (and targets). Our current collections of "faillogs" are sorted by target and not by package. I think by-package info is what most devos want.

champtar, 2018-12-10:
> @champtar - Yes, enabling DCO for all branches. I'll check with @jow to see if we can do something similar for Luci and possibly other repos.

Thanks !

> svn non-reproducible tarballs must be problematic for the builds too however, hash checks are warnings currently...

If you don't use HASH it's not problematic

> @yousong - I would like to see if we could build more than 1 target in CI - It would help with a lot of corner-case issues that pop up from time-to-time.

We can definitely configure more taget in //, but this use more of CircleCI free resources, which are shared by the open source community

> Also, the current script considers a package without a modified Makefile to be OK. I think in a lot of situations, this is not the case - for example, an updated patch or init script without bumping PKG_RELEASE.

If a file is modified, check that the Makefile in the same directory or upper directory is modifed
this will introduce some false positive but remove some manual review

> I am really interested in furthering CI at least for our package repos. The buildbots take care of the targets pretty well. What would really make our lives better would be build failure [...]

buildbots should send and email to the PKG_MAINTAINER + cc a mailinglist openwrt-buildfailures

KanjiMonster, 2018-12-11:
As long as we wait for https://github.com/probot/dco/pull/98 being applied before enforcing DCO, I'm fine with it.

thess, 2018-12-11:
@KanjiMonster - While I agree with your concern, this should not be a block to going forward. If there turns out to be too many exceptions to handle manually, we can always disable the offending check.

Thanks for the feedback!

karlp, 2018-12-17:
Does this integration only pickup from PRs?  I can't push directly at the moment, despite having correctly formed SoB lines.  

```
karlp@maxo:~/src/openwrt-packages (openwrt-18.06)$ git push upstream 
Host key fingerprint is SHA256:nThbg6kXUpJWGl7E1IGOCspRomTxdCARLviKw6E5SY8
+---[RSA 2048]----+
| =+o...+=o..     |
|o++... *o .      |
|*.o.  *o.        |
|oo.  ..o.= .     |
|.+o. .. S =      |
|*=+ .  o = .     |
|OE .  . o        |
| o     .         |
|                 |
+----[SHA256]-----+
Counting objects: 5, done.
Delta compression using up to 4 threads.
Compressing objects: 100% (5/5), done.
Writing objects: 100% (5/5), 643 bytes | 321.00 KiB/s, done.
Total 5 (delta 3), reused 0 (delta 0)
remote: Resolving deltas: 100% (3/3), completed with 3 local objects.
remote: error: GH006: Protected branch update failed for refs/heads/openwrt-18.06.
remote: error: Required status check "DCO" is expected.
To github.com:openwrt/packages.git
 ! [remote rejected]     openwrt-18.06 -> openwrt-18.06 (protected branch hook declined)
error: failed to push some refs to 'git@github.com:openwrt/packages.git'
karlp@maxo:~/src/openwrt-packages (openwrt-18.06)$ git log -p
commit 13a4dcd6147c43e9739b1f75e5d08bab2c8b607f (HEAD -> openwrt-18.06, mos155-1806)
Author: Karl Palsson <karlp@etactica.com>
Date:   Mon Dec 17 10:55:34 2018 +0000

    net/mosquitto: bump to 1.5.5
    
    Security and bug fix.  Full changelog available at: https://mosquitto.org/ChangeLog.txt
    
    Signed-off-by: Karl Palsson <karlp@etactica.com>

diff --git a/net/mosquitto/Makefile b/net/mosquitto/Makefile
<snipped>
```

thess, 2018-12-17:
I'm not entirely sure what the problem may be here. You should be able to push directly to the repo as a "Package Maintainer" member after status checks. I do think you need to do that from a different branch than the protected one. The following statement in the github help seems to imply that.
```
When enabled, commits must first be pushed to another branch, then merged or pushed directly to a branch that matches this rule after status checks have passed.
````
Try pushing from a branch not directly protected via `push upstream <yourbranch>:openwrt-18.06`

karlp, 2018-12-17:
nope, same thing
```
karlp@maxo:~/src/openwrt-packages (mos155-1806)$ git push upstream mos155-1806:openwrt-18.06 
Host key fingerprint is SHA256:nThbg6kXUpJWGl7E1IGOCspRomTxdCARLviKw6E5SY8
+---[RSA 2048]----+
| =+o...+=o..     |
|o++... *o .      |
|*.o.  *o.        |
|oo.  ..o.= .     |
|.+o. .. S =      |
|*=+ .  o = .     |
|OE .  . o        |
| o     .         |
|                 |
+----[SHA256]-----+
Counting objects: 5, done.
Delta compression using up to 4 threads.
Compressing objects: 100% (5/5), done.
Writing objects: 100% (5/5), 643 bytes | 643.00 KiB/s, done.
Total 5 (delta 3), reused 0 (delta 0)
remote: Resolving deltas: 100% (3/3), completed with 3 local objects.
remote: error: GH006: Protected branch update failed for refs/heads/openwrt-18.06.
remote: error: Required status check "DCO" is expected.
To github.com:openwrt/packages.git
 ! [remote rejected]     mos155-1806 -> openwrt-18.06 (protected branch hook declined)
error: failed to push some refs to 'git@github.com:openwrt/packages.git'
karlp@maxo:~/src/openwrt-packages (mos155-1806)$
```

tripolar, 2018-12-17:
But which security benefits does commit to a new branch and push from
this branch to another branch have? Why doesn't it check the SoB lines
at the time the commit is pushed?

 On Mon, 17 Dec 2018 06:29:19 -0800
Ted Hess <notifications@github.com> wrote:

> @karlp - If the issue isn't about which branch you are trying to
> merge from, there is another branch protection setting which may be
> affecting what you are trying to do. Let me know after you try
> pushing from a  "feature branch".
>

thess, 2018-12-17:
@karlp - try it now. I added openwrt/package-maintainers to the `Restrict who can push to matching branches` list. The text in this option is a bit ambiguous. I guess "admins" can always push and "writers" must be excepted - subject to status checks.

Thankyou for your patience, I'm still figuring out Github project management subtleties.

thess, 2018-12-17:
@tripolar - Beats me. They write the rules.

tripolar, 2018-12-17:
On Mon, 17 Dec 2018 15:02:10 +0000 (UTC)
Ted Hess <notifications@github.com> wrote:

> @tripolar - Beats me. They write the rules.
> 

Was this security by obscurity in place before microsoft bought Github?

karlp, 2018-12-17:
doesn't seem to work?
First is after merging the branch locally, and attempting to push, as I've done historically,
second is attempting to push the branch as is to the openwrt-1806.
```
karlp@maxo:~/src/openwrt-packages (openwrt-18.06)$ git push upstream 
<snip>
remote: error: GH006: Protected branch update failed for refs/heads/openwrt-18.06.
remote: error: Required status check "DCO" is expected.
To github.com:openwrt/packages.git
 ! [remote rejected]     openwrt-18.06 -> openwrt-18.06 (protected branch hook declined)
error: failed to push some refs to 'git@github.com:openwrt/packages.git'
karlp@maxo:~/src/openwrt-packages (openwrt-18.06)$ git checkout mos155-1806 
Switched to branch 'mos155-1806'
karlp@maxo:~/src/openwrt-packages (mos155-1806)$ git push upstream mos155-1806:openwrt-18.06 
<snip>
remote: Resolving deltas: 100% (3/3), completed with 3 local objects.
remote: error: GH006: Protected branch update failed for refs/heads/openwrt-18.06.
remote: error: Required status check "DCO" is expected.
To github.com:openwrt/packages.git
 ! [remote rejected]     mos155-1806 -> openwrt-18.06 (protected branch hook declined)
error: failed to push some refs to 'git@github.com:openwrt/packages.git'
karlp@maxo:~/src/openwrt-packages (mos155-1806)$ 
```

karlp, 2018-12-17:
I'll make a PR, but only to try and get this resolved :)

It's just a single commit:
```
karlp@maxo:~/src/openwrt-packages (mos155-1806)$ git cherry -v upstream/openwrt-18.06 
+ 13a4dcd6147c43e9739b1f75e5d08bab2c8b607f net/mosquitto: bump to 1.5.5
karlp@maxo:~/src/openwrt-packages (mos155-1806)$ git checkout openwrt-18.06 
Switched to branch 'openwrt-18.06'
Your branch is ahead of 'upstream/openwrt-18.06' by 1 commit.
  (use "git push" to publish your local commits)
karlp@maxo:~/src/openwrt-packages (openwrt-18.06)$ git cherry -v upstream/openwrt-18.06 
+ 13a4dcd6147c43e9739b1f75e5d08bab2c8b607f net/mosquitto: bump to 1.5.5
commit 13a4dcd6147c43e9739b1f75e5d08bab2c8b607f (HEAD -> openwrt-18.06, remake-gh/rme-gate-v2.10.1, mos155-1806)
Author:     Karl Palsson <karlp@etactica.com>
AuthorDate: Mon Dec 17 10:55:34 2018 +0000
Commit:     Karl Palsson <karlp@etactica.com>
CommitDate: Mon Dec 17 10:55:34 2018 +0000

    net/mosquitto: bump to 1.5.5
    
    Security and bug fix.  Full changelog available at: https://mosquitto.org/ChangeLog.txt
    
    Signed-off-by: Karl Palsson <karlp@etactica.com>

diff --git a/net/mosquitto/Makefile b/net/mosquitto/Makefile
<diff contents snipped>
```

See https://github.com/openwrt/packages/pull/7699 for the push of the same branch.

thess, 2018-12-17:
OK - You shouldn't have any trouble merging from the PR. If we cannot allow direct pushes from project maintainers, I'll disable DCO checks for now. I need a non-owner account to debug this.

karlp, 2018-12-17:
I can push to 1806 now, but pushing to master is still locked.  I tried making an explicit PR, hoping it would just match on the commit sha, and once it had been approved on the PR I could push, but still no go there.

blogic, 2018-12-17:
use they cloud they said, it'll be amazing. I [vote] for not using github :-)

thess, 2018-12-17:
@blogic - Really? Is it just because of MS or there is something missing?

tripolar, 2018-12-17:
how about switching to signed commits instead of this unreliable hooks?

tripolar, 2018-12-17:
> use they cloud they said, it'll be amazing. I [vote] for not using github :-)

me too, let's leave github

blogic, 2018-12-17:
@thess its a walled garden, it just makes my bio correctness filter go f00 ...

thess, 2018-12-17:
For now, until things are sorted, let's disable protected branches.

OK, done - life can return to normal. Bitching about github is still permitted.
Will continue to investigate CI and status checks. I for one, have no interest in moving this beast elsewhere.

cshoredaniel, 2018-12-20:
@thess is it possible to more detail about expected run check that have to at least be filled out before the checks pass? (presuming 'Member's can override check pass requirements if there is good reason).

hnyman, 2019-03-17:
> life can return to normal. Bitching about github is still permitted.

@thess 
did you unprotect also lede-17.01 ?
I have now run into this error with 17.01 when backporting a simple commit as discussed in https://github.com/openwrt/packages/pull/8370#issuecomment-473673836 

```
remote: error: GH006: Protected branch update failed for refs/heads/lede-17.01.
remote: error: Required status check "DCO" is expected.
```

danielg4, 2019-03-17:
> > life can return to normal. Bitching about github is still permitted.
> 
> @thess
> did you unprotect also lede-17.01 ?
> I have now run into this error with 17.01 when backporting a simple commit and pushing it, as discussed in [#8370 (comment)](https://github.com/openwrt/packages/pull/8370#issuecomment-473673836)
> 
> ```
> remote: error: GH006: Protected branch update failed for refs/heads/lede-17.01.
> remote: error: Required status check "DCO" is expected.
> ```

This is related to the problem reported in #7888 also.

thess, 2019-03-17:
@danielg4 - lede-17.01 should be OK now. It shouldn't have been protected from remote push by legitimate writers.

neheb, 2021-08-28:
Is this resolved?

thess, 2021-08-29:
I suppose -- others are managing this now.

closed 2021-08-29 by thess