openwrt/packages #7667: FreeRadius 3: /usr/lib/ Error

ahmar20, 2018-12-14:
TP-Link TD-W8980 V1 - OpenWrt v18.06.1

Using FreeRaius3 on Openwrt is not working with `radiusd` command, apparently it looks for its libraries in `/usr/lib` directory. But the libraries get stored in `/usr/lib/freeradius3` directory. Can anyone fix this? I am sure this should be a couple of lines of code which tells radiusd to look for libraries in the correct place. While using `/etc/init.d/radiusd`, radiusd can be started successfully (if you have setup everything correctly) but there is no debugging option that is available through `radiusd -X`. If someone can point out what to look for I can probably change the code myself and submit a pull request. Thanks

```
root@LEDE:~# radiusd -X
Error loading shared library libfreeradius-server.so: No such file or directory (needed by /usr/sbin/radiusd)
Error loading shared library libfreeradius-radius.so: No such file or directory (needed by /usr/sbin/radiusd)
Error loading shared library libfreeradius-eap.so: No such file or directory (needed by /usr/sbin/radiusd)
Error relocating /usr/sbin/radiusd: fr_cursor_next_by_da: symbol not found
Error relocating /usr/sbin/radiusd: fr_event_loop: symbol not found
Error relocating /usr/sbin/radiusd: fr_packet_list_num_elements: symbol not found
Error relocating /usr/sbin/radiusd: fr_pair_copy: symbol not found
Error relocating /usr/sbin/radiusd: cf_section_to_item: symbol not found
Error relocating /usr/sbin/radiusd: cf_file_read: symbol not found
Error relocating /usr/sbin/radiusd: fr_talloc_fault_setup: symbol not found
Error relocating /usr/sbin/radiusd: ip_ntoh: symbol not found
Error relocating /usr/sbin/radiusd: fr_cursor_next_by_num: symbol not found
Error relocating /usr/sbin/radiusd: cf_data_remove: symbol not found
Error relocating /usr/sbin/radiusd: cf_section_name1: symbol not found
Error relocating /usr/sbin/radiusd: fr_ntop: symbol not found
Error relocating /usr/sbin/radiusd: rad_check_lib_magic: symbol not found
Error relocating /usr/sbin/radiusd: xlat_unregister_module: symbol not found
Error relocating /usr/sbin/radiusd: fr_set_dumpable: symbol not found
Error relocating /usr/sbin/radiusd: rad_recv_header: symbol not found
Error relocating /usr/sbin/radiusd: fr_packet_list_socket_del: symbol not found
Error relocating /usr/sbin/radiusd: fr_prints: symbol not found
Error relocating /usr/sbin/radiusd: fr_debug_state: symbol not found
Error relocating /usr/sbin/radiusd: fr_int2str: symbol not found
Error relocating /usr/sbin/radiusd: fr_cursor_remove: symbol not found
Error relocating /usr/sbin/radiusd: radlog: symbol not found
Error relocating /usr/sbin/radiusd: debug_pair: symbol not found
Error relocating /usr/sbin/radiusd: rad_getpwuid: symbol not found
Error relocating /usr/sbin/radiusd: radius_evaluate_cond: symbol not found
Error relocating /usr/sbin/radiusd: paircompare_unregister_instance: symbol not found
Error relocating /usr/sbin/radiusd: fr_syserror: symbol not found
Error relocating /usr/sbin/radiusd: fr_heap_peek: symbol not found
Error relocating /usr/sbin/radiusd: request_alloc_fake: symbol not found
Error relocating /usr/sbin/radiusd: fr_assignment_op: symbol not found
Error relocating /usr/sbin/radiusd: cf_item_to_section: symbol not found
Error relocating /usr/sbin/radiusd: dict_valnamebyattr: symbol not found
Error relocating /usr/sbin/radiusd: fr_pton: symbol not found
Error relocating /usr/sbin/radiusd: fr_pair_delete_by_num: symbol not found
Error relocating /usr/sbin/radiusd: radlog_debug_enabled: symbol not found
Error relocating /usr/sbin/radiusd: fr_pair_make: symbol not found
Error relocating /usr/sbin/radiusd: cf_section_sub_find_name2: symbol not found
Error relocating /usr/sbin/radiusd: xlat_register: symbol not found
Error relocating /usr/sbin/radiusd: rad_send: symbol not found
Error relocating /usr/sbin/radiusd: rate_limit_enabled: symbol not found
Error relocating /usr/sbin/radiusd: tmpl_prints: symbol not found
Error relocating /usr/sbin/radiusd: fr_socket: symbol not found
Error relocating /usr/sbin/radiusd: fr_ipaddr_cmp: symbol not found
Error relocating /usr/sbin/radiusd: rad_packet_ok: symbol not found
Error relocating /usr/sbin/radiusd: fr_fifo_free: symbol not found
Error relocating /usr/sbin/radiusd: fr_rand: symbol not found
Error relocating /usr/sbin/radiusd: fr_pair_steal: symbol not found
Error relocating /usr/sbin/radiusd: fr_pair_list_afrom_str: symbol not found
Error relocating /usr/sbin/radiusd: fr_fifo_pop: symbol not found
Error relocating /usr/sbin/radiusd: fr_packet_list_socket_freeze: symbol not found
Error relocating /usr/sbin/radiusd: ip_hton: symbol not found
Error relocating /usr/sbin/radiusd: cf_section_lineno: symbol not found
Error relocating /usr/sbin/radiusd: rad_copy_packet: symbol not found
Error relocating /usr/sbin/radiusd: rad_sign: symbol not found
Error relocating /usr/sbin/radiusd: rad_recv_discard: symbol not found
Error relocating /usr/sbin/radiusd: cf_new_escape: symbol not found
Error relocating /usr/sbin/radiusd: xlat_from_tmpl_attr: symbol not found
Error relocating /usr/sbin/radiusd: dict_valbyattr: symbol not found
Error relocating /usr/sbin/radiusd: radlog_init: symbol not found
Error relocating /usr/sbin/radiusd: cf_log_info: symbol not found
Error relocating /usr/sbin/radiusd: fr_packet_list_create: symbol not found
[... 214 lines cut ...]
Error relocating /usr/sbin/radiusd: cf_pair_filename: symbol not found
Error relocating /usr/sbin/radiusd: fr_cursor_next: symbol not found
Error relocating /usr/sbin/radiusd: rbtree_finddata: symbol not found
Error relocating /usr/sbin/radiusd: cf_pair_find_next: symbol not found
Error relocating /usr/sbin/radiusd: cf_item_add: symbol not found
Error relocating /usr/sbin/radiusd: fr_ipaddr2sockaddr: symbol not found
Error relocating /usr/sbin/radiusd: cf_subsection_find_next: symbol not found
Error relocating /usr/sbin/radiusd: fr_event_fd_insert: symbol not found
Error relocating /usr/sbin/radiusd: fr_cursor_insert: symbol not found
Error relocating /usr/sbin/radiusd: fr_hash_update: symbol not found
Error relocating /usr/sbin/radiusd: root_config: symbol not found
Error relocating /usr/sbin/radiusd: cf_section_name: symbol not found
Error relocating /usr/sbin/radiusd: map_cast_from_hex: symbol not found
Error relocating /usr/sbin/radiusd: log_dates_utc: symbol not found
Error relocating /usr/sbin/radiusd: fr_dns_lookups: symbol not found
Error relocating /usr/sbin/radiusd: fr_max_attributes: symbol not found
Error relocating /usr/sbin/radiusd: default_log: symbol not found
Error relocating /usr/sbin/radiusd: log_dates_utc: symbol not found
Error relocating /usr/sbin/radiusd: log_dates_utc: symbol not found
Error relocating /usr/sbin/radiusd: fr_dns_lookups: symbol not found
Error relocating /usr/sbin/radiusd: fr_max_attributes: symbol not found

root@LEDE:~# /etc/init.d/radiusd start

```

micmac1, 2018-12-27:
Hello @ahmar20 

LD_LIBRARY_PATH=/usr/lib/freeradius3 radiusd -h

Also explained in /etc/freeradius3/radiusd.conf.

Regards,
Seb

ahmar20, 2018-12-28:
Alright, I am not working on this at the moment so I will just close the issue for now. If something pops up in future, I'll see what I can do. Thanks

closed 2018-12-28 by ahmar20