openwrt/packages #7677: Stubby: Does not automatically start

adagari, 2018-12-15:
Maintainer: @iamperson347 , @jonathanunderwood 
Environment: mvebu, WRT3200ACM, snapshot

Description:
Stubby fails to start upon reboot. Current systems runs odhcpd, unbound, and stubby. Dnsmasq has been removed. The startup trigger is set to default 'wan'. System logs do not show Stubby tried to start at all.

Restarting the interface will cause Stubby to start. Stubby can also be started by issuing the "service stubby start" command.

jonathanunderwood, 2018-12-18:
That is strange. I can't reproduce it locally. Do you see the same thing with the 18.06 release (using the latest stubby package init so you get the trigger behaviour)?

jonathanunderwood, 2018-12-18:
This could well be a bug with procd; it sounds like early signals are not getting delivered.

I can't immediately see why removing dnsmasq would cause this issue. But it would be worth taking it step by step: install snapshot as stock and confirm whether or not the bug manifests. If it doesn't, then remove dnsmasq and see if it manifests then.

dibdot, 2018-12-18:
@adagari just for testing reduce the START time in the stubby init script (/etc/init.d/stubby) from 50 to 30, reboot and check if the startup issue disappears ...

jonathanunderwood, 2019-01-03:
@adagari - any update on this?

@dibdot - would you advise reducing the START time to 30 in the package?

kesslerm, 2019-01-03:
On a lantiq wbmr-hp-g300h with the latest snapshot from openwrt master, stubby does not appear to start at all, neither automatically after boot, nor manually through `/etc/init.d/stubby start`.

The generated file `/var/etc/stubby/stubby.yml` appears to have an updated timestamp after a reboot or manual start, but there is no sign that `/usr/sbin/stubby` was actually started. It never leaves any log entries in syslog. Changing `START` to a lower value does not appear to make any difference.

The `stubby` executable seems to work when run directly or through a minimal init file, so it appears that there is something in the packages init file that prevents the programme from running.

jonathanunderwood, 2019-01-03:
Thanks @kesslerm.

Can you add `PROCD_DEBUG=1` to the init file and retry? 

Also, add/uncomment `option log_level 7` in `stubby.conf`, even though I don't think that'll tell you anything.

dibdot, 2019-01-03:
@jonathanunderwood yes, we had a similar issue with dnscrypt-proxy init, see here for reference: #4675

jonathanunderwood, 2019-01-03:
Thanks @dibdot !

@kesslerm - when you changed to `START=30` did you also issue `/etc/init.d/stubby disable && /etc/init.d/stubby enable` ?

kesslerm, 2019-01-03:
```
PROCD_DEBUG=1 /etc/init.d/stubby start
{ "name": "stubby", "script": "\/etc\/init.d\/stubby", "instances": { "stubby": { "command": [ "\/usr\/sbin\/stubby", "-C", "\/var\/etc\/stubby\/stubby.yml", "-v", "7" ], "respawn": [ "3600", "5", "5" ], "file": [ "\/var\/etc\/stubby\/stubby.yml" ], "stdout": true, "stderr": true, "pidfile": "\/var\/run\/stubby.pid", "user": "stubby" } }, "triggers": [ [ "interface.*.up", [ "if", [ "eq", "interface", "wan" ], [ "run_script", "\/etc\/init.d\/stubby", "start" ] ], 2000 ], [ "config.change", [ "if", [ "eq", "package", "stubby" ], [ "run_script", "\/etc\/init.d\/stubby", "reload" ] ], 2000 ] ], "data": { } }
```

The above looks sensible to me. Still no output from `stubby` appearing in the syslog. And yes, I did disable and re-enable the service, and also stopped it explicitly prior to restarting.

jonathanunderwood, 2019-01-03:
@kesslerm and I presume everything works fine on 18.06.1, with the latest init file from master?

kesslerm, 2019-01-03:
@jonathanunderwood I can't test that easily right now, as I've been running this router on master builds and haven't checked that my setup is still compatible with 18.06.1.

jonathanunderwood, 2019-01-03:
@kesslerm does stubby start if you restart the interface?

kesslerm, 2019-01-03:
Shortly after restarting the interface, it appears that the init script is called (`/bin/sh /etc/rc.common /etc/init.d/stubby start`), and the `/var/etc/stubby/stubby.yml` file is updated (changed timestamp). `stubby` itself is not running afterwards, just like when directly calling `/etc/init.d/stubby start`.

jonathanunderwood, 2019-01-04:
OK, thanks. I'll endeavour to reproduce this locally (haven't managed to so far). 

@didbot any suggestions? I am thinking this may be a big in procd ?

jonathanunderwood, 2019-01-04:
This might be a clue, or a red herring:

If I change the line in the init file reading:

    procd_set_param command "$stubby" -C "$stubby_config"
 
to:

    procd_set_param command "$stubby -C $stubby_config"

(note the difference in quoting) I see the behaviour that is being reported here. So, `procd_set_param command` is quite fussy about quoting. I wonder if something has changed there.

kesslerm, 2019-01-04:
The service appears to be in a stuck state, as shown here:
```
ubus call service list '{"name":"stubby", "verbose": true}'
{
        "stubby": {
                "instances": {
                        "stubby": {
                                "running": false,
                                "command": [
                                        "\/usr\/sbin\/stubby",
                                        "-C",
                                        "\/var\/etc\/stubby\/stubby.yml"
                                ],
                                "term_timeout": 5,
                                "pidfile": "\/var\/run\/stubby.pid"
                        }
                },
                "triggers": [
                        [
                                "interface.*.up",
                                [
                                        "if",
                                        [
                                                "eq",
                                                "interface",
                                                "wan"
                                        ],
                                        [
                                                "run_script",
                                                "\/etc\/init.d\/stubby",
                                                "start"
                                        ]
                                ],
                                2000
                        ],
                        [
                                "config.change",
                                [
                                        "if",
                                        [
                                                "eq",
                                                "package",
                                                "stubby"
                                        ],
                                        [
                                                "run_script",
                                                "\/etc\/init.d\/stubby",
                                                "reload"
                                        ]
                                ],
                                2000
                        ]
                ]
        }
}
```
Note `"running": false"` and the lack of `"pid"` and `"respawn"` in the instance record. In this state, it's no longer possible to stop or restart an instance. [This bug report](https://bugs.openwrt.org/index.php?do=details&task_id=1481) might be related.

dibdot, 2019-01-04:
@kesslerm your service dump looks quite usual/OK. Information for PID and RESPAWN timeout will only be shown for running services. I would doublecheck two things: your config file (stubby.yml) and maybe something wrong with your user ... if I remember correctly, stubby runs not as root!?

kesslerm, 2019-01-04:
@dibdot, indeed, this issue appears to be related to file permissions on files/directories that stubby tries to access. When running as root (removed the user setting from the init file), the service starts correctly after a reboot. I thought I had tried this before, but didn't perform a reboot, and the change to the init file was likely unsuccessfull once the service was in the 'hanging' state.

jonathanunderwood, 2019-01-04:
Unless you have stubby configured to be doing DNSSEC validation, the only file stubby should need access to is the generated config, which should look like:

```
# ls -l /var/etc/stubby/stubby.yml 
-rw-r--r--    1 root     root           688 Jan  4 03:28 /var/etc/stubby/stubby.yml
```

mistepien, 2019-01-05:
If change "procd_set_param user stubby" to "procd_set_param user nobody" stubby works fine and starts automagically.

mistepien, 2019-01-05:
btw, is it important that the home of stubby user is /var/run/stubby which does not exit and is not created during startup?

mistepien, 2019-01-14:
I have ramips ubnt-x with r9057-8c6f00e and stubby works just fine.
System image made with imagebuilder.

jonathanunderwood, 2019-01-23:
I've been trying to reproduce this in a x86 VM, and on my lantiq xrx200 device and have had no luck.

Stubby 0.2.4 has now been merged into the master branch, so it would be worth seeing if that's fixed the issue.

mistepien, 2019-01-24:
It works fine as it should, there are no more problest with start during the boot.

dibdot, 2019-01-24:
Great, let's close this.

closed 2019-01-24 by dibdot

the-solipsist, 2019-11-08:
Just leaving a note for those who might stumble upon this issue that one other way this happens is when the trigger (set in `/etc/config/stubby`) doesn't fire.  So, if you don't use `wan` as the interface name for your WAN or have renamed it, then you'll run into this problem.

So you'll need to change `option trigger 'wan'` in that case.

Ammi55, 2022-04-13:
> Just leaving a note for those who might stumble upon this issue that one other way this happens is when the trigger (set in `/etc/config/stubby`) doesn't fire. So, if you don't use `wan` as the [...]

Thanks, this resolved the issue for me.

Coralesoft, 2023-10-31:
> /etc/config/stubby

This solved my issue as well, I changed the option trigger 'wan' to 'WAN' and perfect startup
thankyou