openwrt/packages #7694: lxc still being maintained?

n8v8R, 2018-12-16:
@rmilecki Having looked at the package history and its current version 2.1.1 and compared to upstream version 3.0.3 with various bug fixes and security patches I am curious thus whether the package is still actively maintained?

stintel, 2018-12-28:
@ratkaj ?

ratkaj, 2019-01-03:
Hi @n8v8R, @stintel,

yes it is. Trying to find the time to update for a while now, didn't think there was any interest.
I'll try to bump within a week.

n8v8R, 2019-01-03:
@ratkaj Thank you for the feedback and time/effort put into maintenance :+1: 

Probably the user base is rather small indeed considering the hardware requirements (external disk to avoid wearing the NAND, sufficient RAM and CPU power). But a few routers do match such requirements and perhaps more to come. Also the number of user interested in lxc might increase over time - currently it seems not be widely known.

n8v8R, 2019-01-03:
@cshoredaniel https://discuss.linuxcontainers.org/t/lxd-3-0-3-has-been-released/3359

> Support and upgrade
> 
> LXD 3.0.3 is supported until June 2023 and **is our current LTS release,** users are encouraged to update to the latest bugfix releases as they’re made available.

cshoredaniel, 2019-01-03:
Hmmm....I'm surprised, I must have misread the website (or it's not yet been updated).

cshoredaniel, 2019-01-03:
https://linuxcontainers.org/lxc/introduction/ gives the wrong impression -- 1.whatever and 2.whatever are the old LTS not the current LTS.  Should file a bug with them to update the site to be more clear.

alump, 2019-02-10:
2.1 never even was a LTS version? 2.0 was/is the LTS version before 3.0. 2.1 support has actually ended on September 2018 already, so we have been running on unsupported version since then. Right?

n8v8R, 2019-02-10:
@alump https://discuss.linuxcontainers.org/t/lxc-2-1-has-been-released/487

> Note that this isn’t a LTS release and we’ll therefore only be supporting LXC 2.1 for a year.

cshoredaniel, 2019-02-10:
Yeah their 'introduction' page really needs some love.

n8v8R, 2019-02-10:
The release history seems better documented here https://linuxcontainers.org/lxc/news/

But that aside what is happening (or going) to happen with the maintenance of LXC in OpenWRT?

dangowrt, 2019-02-11:
@lperkov ?

n8v8R, 2019-02-12:
One of many security patches missed in an unmaintained LXC package.

https://brauner.github.io/2019/02/12/privileged-containers.html

AreYouLoco, 2019-02-14:
I am also looking for the upstream update of lxc package. I am interested in non-privilidged containers option.

BKPepe, 2019-03-28:
@ratkaj Is there anything new about this issue? Did you have time to look at it?

n8v8R, 2019-06-04:
Not sure who did the effort but just found

`cat /usr/lib/opkg/info/lxc.control`
Package: lxc
Version: **3.0.3-0.0**
Depends: libc
Source: feeds/packages/utils/lxc
License: LGPL-2.1+ BSD-2-Clause GPL-2.0
Section: utils
Maintainer: Marko Ratkaj <marko.ratkaj@sartura.hr>
Architecture: arm_cortex-a9_vfpv3
Installed-Size: 104

stintel, 2019-06-04:
Wonder where that came from ... https://github.com/openwrt/packages/blob/master/utils/lxc/Makefile#L11

n8v8R, 2019-06-04:
@stintel just discovered that it originates from downstream TurrisOS

AreYouLoco, 2019-06-04:
Great! I wonder if the update will go upstream so others can benefit from it.

BKPepe, 2019-06-04:
Guys, we (as Turris guys) already said on the [forum](https://forum.turris.cz/t/turris-os-4-0-beta1-is-released/10107/31?u=pepe) that we are working on it and we would like to upstream it. It's WIP and it requires some more changes before it can be upstreamed. You can check [the issue](https://gitlab.labs.nic.cz/turris/turris-os-packages/issues/354), which is on Turris Gitlab.

@n8v8r: I would like to kindly ask you to avoid using words like downstream and so on. It will just confuse other users. Since Turris OS 4.x we are based on top of OpenWrt with patches and our feed. Some of it, it couldn't be upstream at all as it is CZ.NIC (if you prefer Turris) specific.

Our work is open source. If you are interested in helping us, you can. ;-) Patches are always welcome. You can even see that me and also others are working on sending patches to OpenWrt as much as we can. 

Don't confuse the issue with offtopic as it is not related to the issue at all.

cshoredaniel, 2019-06-05:
@BKPepe Not sure why you take issue with downstream -- OpenWrt is upstream therefore you are downstream.  If seen the terms used in other projects (e.g. in reference to Ubuntu derivatives as downstream).  In open source it is understood downstream is not an exactly replica of upstream, and has unique things/target audience/situation that may not be applicable to upstream.

stintel, 2020-01-16:
@ratkaj any progress? The current version of LXC no longer builds due to 71b663b335da. I fixed the missing header file in c01462dcba15ad4343c6e709f5244bc25f3a01ba, but it seems LXC 2.1.1 is incompatible with libseccomp 2.4.2:
https://gist.github.com/4811e874a746004f32a327c72eae0384

n8v8R, 2020-03-25:
meantime source development released 4.0 as LTS branch

https://github.com/lxc/lxc/releases/tag/lxc-4.0.0

n8v8R, 2020-04-07:
LTS 4.0.1 point release

https://github.com/lxc/lxc/releases/tag/lxc-4.0.1

erdoukki, 2020-04-11:
All tweaks needed for 19.07.x with lxc-2.1.1 are tested here ;
https://forum.openwrt.org/t/request-for-lxc-support-in-mvebu/58767

I shared them here also : 
https://github.com/erdoukki/openwrt-LXC/commit/aa7234637c039fa9559a8cad591995d28087faa8

n8v8R, 2020-04-16:
LTS 4.0.2 point release

https://github.com/lxc/lxc/releases/tag/lxc-4.0.2
https://discuss.linuxcontainers.org/t/lxc-4-0-2-lts-has-been-released/7449

robimarko, 2020-05-19:
We have updated the LXC package yesterday.
https://github.com/openwrt/packages/commit/9f43594e3a28deeea18cfa40acd0b1388e64f600

n8v8R, 2020-05-19:
@robimarko thank you! Hope that it will make soon(er) into the stable branch as well.

Are _unprivileged_ containers also supported now?

closed 2020-05-19 by n8v8R