openwrt/packages #7698: reaver: stucking at "Waiting for beacon from XX:XX:XX:XX:XX:XX"

nurulhuda859, 2018-12-17:
Maintainer: @yousong (find it by checking history of the package Makefile)
Environment: (Atheros AR9344 rev 2, TP-Link CPE210 v1.0, OpenWrt 18.06.1 r7258-5eb055306f / LuCI openwrt-18.06)

Description:
I installed reaver, aircrack-ng and airmon-ng using Luci. Created a wireless network named it 'wlan1' from Network-->Wireless-->Add.
Turnde on monitor mode using airmon-ng
```
# airmon-ng start wlan1
ps: unrecognized option: A
BusyBox v1.28.3 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.28.3 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output
ps: unrecognized option: a
BusyBox v1.28.3 () multi-call binary.

Usage: ps 

Show list of processes

	w	Wide output

Interface	Chipset		Driver

wlan0		Atheros AR9340	ath9k - [phy0]
wlan1		Atheros AR9340	ath9k - [phy0]
				(monitor mode enabled on mon0)

```
and then I ran reaver
```
# reaver -i mon0 -b XX:XX:XX:XX:XX:XX -vv

Reaver v1.6.4 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Waiting for beacon from XX:XX:XX:XX:XX:XX

```
It stuck here forever.

yousong, 2018-12-17:
sorry, not an expert on reaver or aircrack-ng.  You may want to debug further with strace to see what reaver was waiting for, then a better chance would be reporting that to the upstream project [1].

 [1] https://github.com/t6x/reaver-wps-fork-t6x/

yousong, 2018-12-17:
As for the repeated warning from `ps` about unrecognized option `a`.  It's because airmon-ng requires `ps` command from procps-ng.  busybox `ps` command does not suffice.  Many people just use `iw` command to create a monitor mode interface instead [1].

I will keep this issue open for a few days in case further progress needed to be tracked here.  Thank you.

 [1] https://github.com/openwrt/packages/pull/1693#issuecomment-133818911

nurulhuda859, 2018-12-17:
How can I solve this problem?

yousong, 2018-12-17:
Install strace and post output of `strace -ff -tt reaver -i mon0 -b XX:XX:XX:XX:XX:XX -vv` (as attachment if it's too big) please.

nurulhuda859, 2018-12-17:
Wait.. I am sending you these data as soon as I can                                                                                                                                                                                                                                                                                                                                        Sent from my BlackBerry 10 smartphone.                                                                                                                                                                                                                From: Yousong ZhouSent: Monday, December 17, 2018 6:14 PMTo: openwrt/packagesReply To: openwrt/packagesCc: Nurul Huda; AuthorSubject: Re: [openwrt/packages] reaver: stucking at "Waiting for beacon from XX:XX:XX:XX:XX:XX" (#7698)Install strace and post output of strace -ff -tt reaver -i mon0 -b XX:XX:XX:XX:XX:XX -vv (as attachment if it's too big) please.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/openwrt/packages/issues/7698#issuecomment-447824497",
"url": "https://github.com/openwrt/packages/issues/7698#issuecomment-447824497",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

nurulhuda859, 2018-12-17:
I appreciate your help here is what I got after running your given command.
```
# strace -ff -tt reaver mon0 -b 84:16:F9:52:9C:1E -vv
18:28:22.109719 execve("/usr/bin/reaver", ["reaver", "mon0", "-b", "84:16:F9:52:9C:1E", "-vv"], 0x7fa90838 /* 12 vars */) = 0
18:28:22.113391 set_thread_area(0x77fc9dc0) = 0
18:28:22.115087 set_tid_address(0x77fc2d28) = 20375
18:28:22.117606 open("/etc/ld-musl-mips-sf.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
18:28:22.119577 open("/lib/libpcap.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
18:28:22.121610 open("/usr/local/lib/libpcap.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
18:28:22.123196 open("/usr/lib/libpcap.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
18:28:22.125188 fcntl64(3, F_SETFD, FD_CLOEXEC) = 0
18:28:22.126611 fstat64(3, {st_mode=S_IFREG|0644, st_size=177336, ...}) = 0
18:28:22.129275 read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\3\0\10\0\0\0\1\0\0=\360\0\0\0004"..., 936) = 936
18:28:22.131386 mmap2(NULL, 245760, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x77ee2000
18:28:22.133441 mmap2(0x77f1c000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x2a000) = 0x77f1c000
18:28:22.135749 close(3)                = 0
18:28:22.137146 open("/lib/libgcc_s.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
18:28:22.141023 fcntl64(3, F_SETFD, FD_CLOEXEC) = 0
18:28:22.142940 fstat64(3, {st_mode=S_IFREG|0644, st_size=78096, ...}) = 0
18:28:22.145534 read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\3\0\10\0\0\0\1\0\0(p\0\0\0004"..., 936) = 936
18:28:22.147569 mmap2(NULL, 147456, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x77ebe000
18:28:22.149697 mmap2(0x77ee1000, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x13000) = 0x77ee1000
18:28:22.152726 close(3)                = 0
18:28:22.155594 mprotect(0x77f1c000, 4096, PROT_READ) = 0
18:28:22.158494 mprotect(0x463000, 4096, PROT_READ) = 0
18:28:22.161181 brk(NULL)               = 0x585000
18:28:22.162490 brk(0x590000)           = 0x590000
18:28:22.163988 writev(2, [{iov_base="\nReaver v1.6.4 WiFi Protected Se"..., iov_len=48}, {iov_base=NULL, iov_len=0}], 2
Reaver v1.6.4 WiFi Protected Setup Attack Tool
) = 48
18:28:22.166899 writev(2, [{iov_base="", iov_len=0}, {iov_base="Copyright (c) 2011, Tactical Net"..., iov_len=88}], 2Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

) = 88
18:28:22.169782 writev(2, [{iov_base="", iov_len=0}, {iov_base="Required Arguments:\n", iov_len=20}], 2Required Arguments:
) = 20
18:28:22.172666 writev(2, [{iov_base="", iov_len=0}, {iov_base="\t-i, --interface=<wlan>         "..., iov_len=75}], 2 -i, --interface=<wlan>          Name of the monitor-mode interface to use
) = 75
18:28:22.175428 writev(2, [{iov_base="", iov_len=0}, {iov_base="\t-b, --bssid=<mac>              "..., iov_len=56}], 2 -b, --bssid=<mac>               BSSID of the target AP
) = 56
18:28:22.177632 writev(2, [{iov_base="", iov_len=0}, {iov_base="\nOptional Arguments:\n", iov_len=21}], 2
Optional Arguments:
) = 21
18:28:22.179983 writev(2, [{iov_base="", iov_len=0}, {iov_base="\t-m, --mac=<mac>                "..., iov_len=56}], 2 -m, --mac=<mac>                 MAC of the host system
) = 56
18:28:22.182926 writev(2, [{iov_base="", iov_len=0}, {iov_base="\t-e, --essid=<ssid>             "..., iov_len=56}], 2 -e, --essid=<ssid>              ESSID of the target AP
) = 56
18:28:22.185678 writev(2, [{iov_base="", iov_len=0}, {iov_base="\t-c, --channel=<channel>        "..., iov_len=87}], 2 -c, --channel=<channel>         Set the 802.11 channel for the interface (implies -f)
) = 87
18:28:22.188122 writev(2, [{iov_base="", iov_len=0}, {iov_base="\t-o, --out-file=<file>          "..., iov_len=68}], 2 -o, --out-file=<file>           Send output to a log file [stdout]
) = 68
18:28:22.190401 writev(2, [{iov_base="", iov_len=0}, {iov_base="\t-s, --session=<file>           "..., iov_len=65}], 2 -s, --session=<file>            Restore a previous session file
) = 65
18:28:22.193169 writev(2, [{iov_base="", iov_len=0}, {iov_base="\t-C, --exec=<command>           "..., iov_len=91}], 2 -C, --exec=<command>            Execute the supplied command upon successful pin recovery
) = 91
18:28:22.196043 writev(2, [{iov_base="", iov_len=0}, {iov_base="\t-D, --daemonize                "..., iov_len=50}], 2 -D, --daemonize                 Daemonize reaver
) = 50
18:28:22.198300 writev(2, [{iov_base="", iov_len=0}, {iov_base="\t-f, --fixed                    "..., iov_len=57}], 2 -f, --fixed                     Disable channel hopping
) = 57
18:28:22.201027 writev(2, [{iov_base="", iov_len=0}, {iov_base="\t-5, --5ghz                     "..., iov_len=58}], 2 -5, --5ghz                      Use 5GHz 802.11 channels
) = 58
18:28:22.204648 writev(2, [{iov_base="", iov_len=0}, {iov_base="\t-v, --verbose                  "..., iov_len=86}], 2 -v, --verbose                   Display non-critical warnings (-vv or -vvv for more)
) = 86
18:28:22.207377 writev(2, [{iov_base="", iov_len=0}, {iov_base="\t-q, --quiet                    "..., iov_len=64}], 2 -q, --quiet                     Only display critical messages
) = 64
18:28:22.210187 writev(2, [{iov_base="", iov_len=0}, {iov_base="\t-h, --help                     "..., iov_len=43}], 2 -h, --help                      Show help
) = 43
18:28:22.212379 writev(2, [{iov_base="", iov_len=0}, {iov_base="\nAdvanced Options:\n", iov_len=19}], 2
Advanced Options:
) = 19
[... cut ...]
18:28:22.264351 writev(2, [{iov_base="", iov_len=0}, {iov_base="\t-Z                             "..., iov_len=54}], 2 -Z                              Run pixiedust attack
) = 54
18:28:22.266613 writev(2, [{iov_base="\nExample:\n\treaver -i wlan0mon -b"..., iov_len=56}, {iov_base=NULL, iov_len=0}], 2
Example:
        reaver -i wlan0mon -b 00:90:4C:C1:AC:21 -vv

) = 56
18:28:22.270126 close(1)                = 0
18:28:22.271789 exit_group(1)           = ?
18:28:22.274102 +++ exited with 1 +++
```

nurulhuda859, 2018-12-17:
Sorry I am doing that again....

yousong, 2018-12-17:
`strace -f -o o.txt ...` if it's too big to copy from screen.  Copy `o.txt` when it hangs.

nurulhuda859, 2018-12-17:
Ii is generating same code in a loop.

yousong, 2018-12-17:
You will need to find out what file descriptor number 3 is by going backwards to see how it was open.  Correlate that to the C code, or report your findings to the upstream project.

nurulhuda859, 2018-12-17:
I ran `strace -ff -tt -o dlog.txt reaver -i mon0 -b 84:16:F9:52:9C:1E -vv` and is hanging in here
```
# strace -ff -tt -o dlog.txt reaver -i mon0 -b 84:16:F9:52:9C:1E -vv

Reaver v1.6.4 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Waiting for beacon from 84:16:F9:52:9C:1E
```

nurulhuda859, 2018-12-17:
The process is steel hanging their. I copied the file from the midpoint. Here it is
[o.txt](https://github.com/openwrt/packages/files/2686097/o.txt)

yousong, 2018-12-17:
Have you tried putting `mon0` into correct channel?  airmon-ng is different from airodump-ng which switches between different channels for scanning.

nurulhuda859, 2018-12-17:
I which command should I put mon0 in correct channel.
NOTE: I am currently using the router as client
.

yousong, 2018-12-17:
maybe `-c` will suffice, but I am running out of better advice ;)  It seems to me reaver is there waiting for beacon frames without success.  It did not hang, it just had nothing to report.

nurulhuda859, 2018-12-17:
I tried `-c ` with correct channel in reaver but same result. But why reaver has nothing to replay?

neheb, 2018-12-20:
One note: Monitor mode can be activated from LuCI directly instead of using airmon-ng.

nurulhuda859, 2018-12-21:
I have no problem with monitor mode but with reaver. I tried earlier version of reaver and it worked (although not avail to test it).

yousong, 2018-12-21:
@nurulhuda859 please consider opening an issue in the upstream project, mentioning also the working version.  Thank you

nurulhuda859, 2018-12-21:
I haired that reaver is no longer developed by anyone.

nurulhuda859, 2018-12-21:
So now I think we can close this issue.

closed 2018-12-22 by yousong