openwrt/packages #7712: [RFC] Compile but not run testing point release package updates

cshoredaniel, 2018-12-17:
@thess @jow- @dedeckeh @dibdot @hnyman @blogic @champtar @dangowrt @hauke @yousong 

I've noticed that a *lot* of recent point release package updates are being compile tested but not run tested.  I'm a little disturbed that some folks feel that being up-to-date with upstream is more important than actually testing what goes into the packages feed, and I was wondering if there was an official position by members of the OpenWrt team on this, or if there should be.

karlp, 2018-12-17:
There's a lot of PRs being made that do this, I'm not aware of a lot of them just being blindyly merged.

diizzyy, 2018-12-18:
I also have the same impression that many are merged without any run testing, at least it's not mentioned. I think it should be done for any "update" unless it's cosmetic.

champtar, 2019-01-02:
It really depends on the software IMO, I know in my case when I was updating SQLite my run test was only doing a dumd select when I was doing one, but I know it's used everywhere, with a ton of tests, and really I never had an issue

diizzyy, 2019-01-02:
@champtar 
There's still a difference between doing anything and nothing at all

karlp, 2019-01-02:
I'm very much against "compile only" testing for lua packages, because it doesn't test anything.  Other packages are harder to tell, dot releases from packages where upstream uses (some semblance of) semantic versioning are normally just fine, but others are very hit and miss.  I _generally_ feel it's noise, and things that maintainers should take care of themselves, though they can certainly be prodded into updating more often sometimes.

hnyman, 2019-01-02:
I see the basic problem to be that half of the packages are semi-abandoned by the original maintainers. 

Many feed repo commit access holders are maintaining/updating just 1-2 packages by themselves, and most packages have been imported years ago into the repo by somebody, but many of the one-time maintainers have lost interest for the packages. Also some core Openwrt committers have abandoned packages in the feed repo in practice.

The alternatives are pretty much as
* leaving packages to rot 
* accept minor release version updates without much testing

I see normal minor x.X upstream release bumps without other changes as run-of-the-mill stuff, not needing extensive testing. (Same goes also for stuff like "parallel build" definition in Makefile, as it should only have impact on the build process.). Most of the packages are not even dependencies of other packages, so the possible fallout is limited.

More extensive changes like library changes likely require more testing.

Recently there has also been a bit special effort of making packages compatible with Openssl 1.1.x. That also helped to get several long-term broken packages to at least compile.

The large number of semi-abandoned packages is also the reason, why I am reluctant to accept PRs for new packages.

cshoredaniel, 2019-01-02:
@hnyman I largely agree except that, as I've said, I think there needs to be pruning of the feed instead of the situation as it is.  It won't happen for 19.0x release but I still have a COPR/PPA-style thing for OpenWrt in my sights.  I think  medium to long term it's a way to reduce the pressure to accept / keep packages that aren't maintained, and to concentrate efforts on the subset that are really being used.

diizzyy, 2019-01-02:
What we're concerned about is **no testing at all** and the removal process of abandoned/very outdated packages is next to impossible so it's not like things are going to improve without a change.

I still personally think that packages should move towards a "community review system" and not have a hard requirement of a resident maintainer unless someone wants to "claim" a package. Only a handful of packages actually do have active maintainers while the rest are more or less handled by community reviews anyway.

While it's a best effort community project I honestly don't think "attracting" more especially new contributors by having PRs open for months without any constructive feedback is the way to go. If XXX is too niché etc just mention it and possibly add your concerns about keeping it up to date. If a few contributors have the same conclusion I don't think it's going to much of an issue. Possibly add something in the CONTRIBUTING.md about it so people know that their package might be not accepted.

cshoredaniel, 2019-01-02:
I have to say I think that at least smoke testing should happen -- and IMO if a package has been so broken for ages that smoke testing is an issue, why are we keeping it in the tree?

cshoredaniel, 2019-01-02:
Could we perhaps add a tag PKG_MAINTAINER:="OpenWrt Packages Community <packages@lists.openwrt.org>" for packages that say at least three (number to be discussed) agree that they'll take care PR's of when members of the wider community submit them, and will step in to take care of CVE's and major bugfixes for those packages?  I also think that point release updates and such should be taken care of mostly by those actually interested in the packages, and not just a constant stream of them, especially if they are **not run tested**.  I think that trying to do too many packages in too short a period of time has a huge negative impact on the quality of result.

neheb, 2019-01-02:
@cshoredaniel I must admit that my approach is to throw a package on a wall and see if it sticks. Mistakes can always be fixed. And yes, that does mean limited or in some cases, no testing.

As @hnyman alluded to, a lot of the maintainers here don't maintain.

cshoredaniel, 2019-01-02:
@neheb I come from a very different philosophy (I think I've come around more to the way @nbd168 and @blogic approach the question of what should be in the codebase.  There was a time when I got impatient with that but I think I view that as a mistake to not repeat).  I guess the question is what do most of those who have decision making power feel about this question.

That's not personal, as I said I come from a very different approach to the question of testing, but I think if it were only that, I'd probably get used it.  The thing that really gets me frustrated is things like misleading PR / commit titles that make it see (to me) like you're deliberately not being straight up about your approach and what you are doing.  I don't like subterfuge especially when it comes to PR's and such.  @karlp seems to have noticed the same thing in a PR on a package he maintains -- I don't know if he'd agree with me in the general case, but it certainly isn't helpful to a positive vibe IMO.

karlp, 2019-01-03:
@neheb some of the maintainers might jsut have a different idea of "maintains" tahn you do :)  Latest versions of pacakges aren't always flawless improvements.  Sometimes newer versions don't even affect openwrt, or remove functionality even.  While it's "nice" to always have the latest, it's not something _I_ consider nice _enough_ that it should be done _at all costs_ the minute newer versions are available.

luizluca, 2019-01-04:
Runtime tests is not a panacea to get rid of all upgrade problems. They are normally quite simple. The upgrade might simply have broken some function not tested. I imagine what kind of test would be enough to do a good test for a python update on multiple targets... If the bump is about a bugfix/security release, breaks are really an upstream bug. OpenWrt is just unlucky to be affected.

Everytime you add a new rule to accept a PR, you'll increase the cost of that PR. This will eventually drive off new PR, new contributors and potentially new maintainers.

For major updates, I think a runtime test is mandatory, even a basic one. For a security/bugfix release, I think it's up to the maintainer to decide when a runtime test is needed and what kind of test is enough. In the end, the question is: is it better to apply an update that fixes (published security) bugs or to pass all packages to a simple runtime test that will scarcely catch a rare upgrade problem?

It would be nice if a package could have a Package/xxx/runtime-test section that generates a simple, potentially destructible, runtime test script. CircleCI (or something else) could run that in a OpenWrt machine. For VMs and snapshots, this is not very complex. I don't if there are already images that run in an emulated environment (qemu?) for main archs. GSoC 2019?

cshoredaniel, 2019-01-04:
@luizluca My only caveat is that depending on the upstream upgrading a point release __just because upstream has one__ doesn't necessarily make sense.  I think more care needs to go into evaluating whether a point release (other than security fix) is worth the risk on a distro (OpenWrt) that caters to a much wider range of devices than a typical developer has access to.  If it's a 'minor' fix does it bring enough benefit to actually warrant the challenges it could create?

cshoredaniel, 2019-01-05:
I wonder if we could distinguish between not maintained packages and those with a maintainer who does participate, even if not as quickly as the impatient would like.  I think official policy should be more than 'a couple/few days and push if not NAKed' IFF there is a participating in the past couple months maintainer (and that if there is a maintainer who is not participating and hasn't indicated a temporary leave situation that it's okay to remove the maintainer flag...in the case of temporary leave situation I'd say leaving maintainer but the quicker push policy for that period makes sense).  @thess @hnyman @dibdot @jow- @yousong, @karlp and others -- thoughts?

cshoredaniel, 2019-04-17:
A combination of factors has interfered with working on much of anything non-local for 2-3 months now and in any event I am considering dropping OpenWrt from my roster of 'Things To Do' even when the difficulties are resolved, as I need to consider where too allocate my limited time.

closed 2019-04-17 by cshoredaniel