openwrt/packages #7728: <ip> and <tc> packages no longer available

mcaptur, 2018-12-19:
since a week ago ip and tc packages are no longer available and i cannot install banip and sqm scripts since these depend on tc and ip

nobk, 2018-12-19:
@jow- you patch  cause compile error /home/user/git/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib/libelf.so: undefined reference to 'libintl_dgettext'.

```
    CC       tc_core.o
    CC       tc_red.o
    CC       tc_cbq.o
    CC       tc_estimator.o
    CC       tc_stab.o
    AR       libtc.a
files="e_bpf.c em_canid.c em_cmp.c em_ipset.c em_ipt.c em_meta.c em_nbyte.c em_u32.c f_basic.c f_bpf.c f_cgroup.c f_flow.c f_flower.c f_fw.c f_matchall.c f_route.c f_rsvp.c f_tcindex.c f_u32.c m_action.c m_bpf.c m_connmark.c m_csum.c m_ematch.c m_estimator.c m_gact.c m_ife.c m_ipt.c m_mirred.c m_nat.c m_pedit.c m_police.c m_sample.c m_simple.c m_skbedit.c m_skbmod.c m_tunnel_key.c m_vlan.c m_xt.c m_xt_old.c p_eth.c p_icmp.c p_ip.c p_ip6.c p_tcp.c p_udp.c q_atm.c q_cake.c q_cbq.c q_cbs.c q_choke.c q_clsact.c q_codel.c q_drr.c q_dsmark.c q_etf.c q_fifo.c q_fq.c q_fq_codel.c q_gred.c q_hfsc.c q_hhf.c q_htb.c q_ingress.c q_mqprio.c q_multiq.c q_netem.c q_pie.c q_prio.c q_qfq.c q_red.c q_rr.c q_sfb.c q_sfq.c q_skbprio.c q_tbf.c static-syms.c tc.c tc_cbq.c tc_class.c tc_core.c tc_estimator.c tc_exec.c tc_filter.c tc_monitor.c tc_qdisc.c tc_red.c tc_stab.c tc_util.c" ; \
        for s in `grep -B 3 '\<dlsym' $files | sed -n '/snprintf/{s:.*"\([^"]*\)".*:\1:;s:%s::;p}'` ; do \
                sed -n '/'$s'[^ ]* =/{s:.* \([^ ]*'$s'[^ ]*\) .*:extern char \1[] __attribute__((weak)); if (!strcmp(sym, "\1")) return \1;:;p}' $files ; \
        done > static-syms.h
    CC       static-syms.o
    LINK     tc
/home/user/git/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/lib/gcc/mipsel-openwrt-linux-musl/7.4.0/../../../../mipsel-openwrt-linux-musl/bin/ld: warning: libintl.so.8, needed by /home/user/git/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib/libelf.so, not found (try using -rpath or -rpath-link)
/home/user/git/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/lib/gcc/mipsel-openwrt-linux-musl/7.4.0/../../../../mipsel-openwrt-linux-musl/bin/ld: /home/user/git/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib/libelf.so: undefined reference to `libintl_dgettext'
collect2: error: ld returned 1 exit status
make[5]: *** [tc] Error 1
rm emp_ematch.lex.c
make[4]: *** [all] Error 2
make[4]: Leaving directory `/home/user/git/openwrt/build_dir/target-mipsel_24kc_musl/iproute2-full/iproute2-4.19.0'
make[3]: *** [/home/user/git/openwrt/build_dir/target-mipsel_24kc_musl/iproute2-full/iproute2-4.19.0/.built] Error 2
make[3]: Leaving directory `/home/user/git/openwrt/package/network/utils/iproute2'
time: package/network/utils/iproute2/full/compile#16.84#0.89#19.41
make[2]: *** [package/network/utils/iproute2/compile] Error 2
make[2]: Leaving directory `/home/user/git/openwrt'
make[1]: *** [/home/user/git/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/user/git/openwrt'
make: *** [world] Error 2
 
```

dibdot, 2018-12-20:
@jow with latest master I receive the following error in my local build environment - is this related?

```
make[4]: Entering directory '/home/dirk/openwrt/GL-AR750S/source/build_dir/target-mips_24kc_musl/elfutils-0.174'
make --no-print-directory all-recursive
Making all in config
make[6]: Nothing to be done for 'all'.
Making all in m4
make[6]: Nothing to be done for 'all'.
Making all in lib
  CC       xmalloc.o
In file included from xmalloc.c:37:
xmalloc.c: In function 'xmalloc':
system.h:39:2: error: macro expands to multiple statements [-Werror=multistatement-macros]
  fflush(stdout); \
  ^~~~~~
xmalloc.c:52:5: note: in expansion of macro 'error'
     error (EXIT_FAILURE, 0, _("memory exhausted"));
     ^~~~~
xmalloc.c:51:3: note: some parts of macro expansion are not guarded by this 'if' clause
   if (p == NULL)
   ^~
In file included from xmalloc.c:37:
xmalloc.c: In function 'xcalloc':
system.h:39:2: error: macro expands to multiple statements [-Werror=multistatement-macros]
  fflush(stdout); \
  ^~~~~~
xmalloc.c:65:5: note: in expansion of macro 'error'
     error (EXIT_FAILURE, 0, _("memory exhausted"));
     ^~~~~
xmalloc.c:64:3: note: some parts of macro expansion are not guarded by this 'if' clause
   if (p == NULL)
   ^~
In file included from xmalloc.c:37:
xmalloc.c: In function 'xrealloc':
system.h:39:2: error: macro expands to multiple statements [-Werror=multistatement-macros]
  fflush(stdout); \
  ^~~~~~
xmalloc.c:77:5: note: in expansion of macro 'error'
     error (EXIT_FAILURE, 0, _("memory exhausted"));
     ^~~~~
xmalloc.c:76:3: note: some parts of macro expansion are not guarded by this 'if' clause
   if (p == NULL)
   ^~
cc1: all warnings being treated as errors
Makefile:415: recipe for target 'xmalloc.o' failed
make[6]: *** [xmalloc.o] Error 1
Makefile:483: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
Makefile:399: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/dirk/openwrt/GL-AR750S/source/build_dir/target-mips_24kc_musl/elfutils-0.174'
Makefile:95: recipe for target '/home/dirk/openwrt/GL-AR750S/source/build_dir/target-mips_24kc_musl/elfutils-0.174/.built' failed
make[3]: *** [/home/dirk/openwrt/GL-AR750S/source/build_dir/target-mips_24kc_musl/elfutils-0.174/.built] Error 2
make[3]: Leaving directory '/home/dirk/openwrt/GL-AR750S/source/package/libs/elfutils'
time: package/libs/elfutils/compile#0.03#0.02#0.13
package/Makefile:107: recipe for target 'package/libs/elfutils/compile' failed
make[2]: *** [package/libs/elfutils/compile] Error 2
make[2]: Leaving directory '/home/dirk/openwrt/GL-AR750S/source'
package/Makefile:103: recipe for target '/home/dirk/openwrt/GL-AR750S/source/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/dirk/openwrt/GL-AR750S/source/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/dirk/openwrt/GL-AR750S/source'
/home/dirk/openwrt/GL-AR750S/source/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
```

jow-, 2018-12-20:
I don't know, I cannot reproduce it. In any case this seems to be a different issue.

dibdot, 2018-12-20:
Thanks! Seems to be compiler related, with gcc7 the build runs flawlessly.

diizzyy, 2019-03-28:
None of these are in the package repo, please use http://bugs.openwrt.org/ instead.

closed 2019-03-28 by diizzyy