openwrt/packages #7735: Using our CircleCI Docker images for integration testing

thess, 2018-12-19:
I have put together a couple of bash scripts to setup and use the Docker OpenWrt integration test environment along the lines of our current CircleCI 'buildpr' check. 

If you don't have Docker installed, you must first do that available for your distro. (I have only tested this on Ubuntu 18.04/bionic with Docker.io)

There are 2 scripts (attached ci-testing.zip) - You can customize the shell variables defined at the top of each for your specific environment. Ex: location of build-root, package-tree, SDK location and branch selection, etc.. Normally, these scripts are located in your OpenWrt buildroot.

The first script `docker-startup` launches the Docker image from our public repository and maps your package source tree into the container. It then copies the 2nd script into /home/build directory and then enters the running container.

The second script `build-package` will download the SDK, un-pack it and setup a single package build using a nearly identical procedure as the CircleCI integration check. Upon entering the Docker container, you can:
```
cd /home/build
./build-package <package-name>
```

[ci-testing.zip](https://github.com/openwrt/packages/files/2696485/ci-testing.zip)

thess, 2018-12-19:
@neheb , @cotequeiroz - Here's something I hope you might find useful for testing your builds locally in an SDK environment similar to the packages buildbot production and CircleCI checks.

cshoredaniel, 2018-12-19:
@thess Awesome!  Thank you.  I'm interested in this too.

cshoredaniel, 2019-01-07:
Er, why unpinned?  Or at least could there be pinned a way to have builds configured and commands used to be close to CircleCI (e.g. for me USERID conflicts in packages don't error out locally and prevent feeds installation, but with the config (or commands) used with CircleCI it does)).

thess, 2019-01-07:
@heil - A few words of explanation would have been nice before you unilaterally unpinned my post. Was it something I said? How would you do it differently?

Andy2244, 2019-01-11:
ah nice, i use something similar to get a clean, reproduce-able package build environment.
https://github.com/Andy2244/openwrt-package-builder