openwrt/packages #7742: [RFC] [18.06] default to run as root or backport hotplug feature?

cshoredaniel, 2018-12-21:
For #7638 (backport nut fixes to 18.06) there were enhancements I didn't backport (since stable tree shouldn't usually get new stuff that isn't a bug fix) which includes https://github.com/openwrt/packages/pull/6897/commits/1cd4dcfa99f1c619874ee75b978d59118a43934b and related fixes that set the user and permissions so that nut running as user nut can access USB UPS devices without having to doing anything 'manually' on the router.  I realized recently that the changes I backport default to running as user nut, which could lead to permissions problems on 18.06 branch (master is working well).
My question is whether I should default 18.06 to run as root (old behavior) or backport the hotplug script that sets the permissions correctly?

@malakingpusa, @ivanich, @tofurky, @mantinan, @hnyman, @dibdot, @jow-

jow-, 2018-12-21:
Took a brief look at the referenced commits and couldn't easily spot what and how permissions are set in the hotplug script. My first thought was whether it would be feasible to introduce an "usb" group in OpenWrt base instead.

cshoredaniel, 2018-12-22:
@jow- I guess my question is what is best for 18.06 -- from my point of view master is in good shape (at least for setting permissions etc).  I'm not sure catchall usb group is a good idea depending on how many usb devices folks want to add, and what the impact would be on a combination of storage and other devices from a security PoV.

For master:
The hotplug script in tree comes in two parts.  The head sets the perms and user here: https://github.com/openwrt/packages/blob/cb6fc9d7a994e1b67f72992c8a3004e5cb00f624/net/nut/files/30-libhid-ups.head#L14, but ONLY if the productid and vendorid match the ones that NUT has identified as being a UPS (there is a file generated in the NUT build that has that information, which we mangle to something we can use).

An example of the hotplug script actually on a device is:

```
#!/bin/sh

. /lib/functions.sh

nut_driver_config() {
	local cfg="$1"
	local nomatch="$2"

	config_get runas "$cfg" runas "nut"
	config_get vendorid "$cfg" vendorid
	config_get productid "$cfg" productid

	[ "$ACTION" = "add" ] &&[ -n "$DEVNAME" ] && {
		chmod 0660 /dev/"$DEVNAME"
		chown ${runas:-root}:$(id -gn "${runas:-root}") /dev/"$DEVNAME"
	}

	if [ "$nomatch" = "1" ]; then
		[ "$ACTION" = "add" ] && {
			/etc/init.d/nut-server start "$cfg"
		}
		[ "$ACTION" = "remove" ] && {
			/etc/init.d/nut-server stop "$cfg"
		}
	elif [ "$(printf "%04x" 0x"$pvendid")" = "$vendorid" ] && \
		[ "$(printf "%04x" 0x"$pprodid")" = "$productid" ]; then
			[ "$ACTION" = "add" ] && {
				/etc/init.d/nut-server start "$cfg"
				/etc/init.d/nut-server reload upsd
			}
			[ "$ACTION" = "remove" ] && {
				/etc/init.d/nut-server stop "$cfg"
			}
			found=1
	fi
}

perform_libhid_action() {
	local vendorid productid runas
	local pvendid pprodid found

	pvendid=${PRODUCT%/*}
	pvendid=${pvendid%/*}
	pprodid=${PRODUCT%/*}
	pprodid=${pprodid##*/}

	config_load nut_server
	config_foreach nut_driver_config driver 0
	[ "$found" != "1" ] && config_foreach nut_driver_config driver 1
}

[ -n "$PRODUCT" ] && case "$PRODUCT" in
3f0/1/* | \
3f0/1f01/* | \
3f0/1f02/* | \
3f0/1f06/* | \
3f0/1f08/* | \
3f0/1f09/* | \
3f0/1f0a/* | \
3f0/1fe0/* | \
3f0/1fe1/* | \
3f0/1fe2/* | \
3f0/1fe3/* | \
3f0/1fe5/* | \
3f0/1fe6/* | \
3f0/1fe7/* | \
3f0/1fe8/* | \
463/1/* | \
463/ffff/* | \
47c/ffff/* | \
483/35/* | \
4b4/5500/* | \
4d8/d004/* | \
4d8/d005/* | \
50d/375/* | \
50d/551/* | \
50d/750/* | \
50d/751/* | \
50d/900/* | \
50d/910/* | \
50d/912/* | \
50d/980/* | \
50d/f51/* | \
50d/1100/* | \
51d/0/* | \
51d/2/* | \
51d/3/* | \
592/2/* | \
592/4/* | \
5b8/0/* | \
665/5161/* | \
6da/2/* | \
6da/3/* | \
6da/4/* | \
6da/5/* | \
6da/201/* | \
6da/601/* | \
6da/ffff/* | \
75d/300/* | \
764/5/* | \
764/501/* | \
764/601/* | \
925/1234/* | \
9ae/1/* | \
9ae/1003/* | \
9ae/1007/* | \
9ae/1008/* | \
9ae/1009/* | \
9ae/1010/* | \
9ae/2005/* | \
9ae/2007/* | \
9ae/2008/* | \
9ae/2009/* | \
9ae/2010/* | \
9ae/2011/* | \
9ae/2012/* | \
9ae/2013/* | \
9ae/2014/* | \
9ae/3008/* | \
9ae/3009/* | \
9ae/3010/* | \
9ae/3011/* | \
9ae/3012/* | \
9ae/3013/* | \
9ae/3014/* | \
9ae/3015/* | \
9ae/3016/* | \
9ae/4001/* | \
9ae/4002/* | \
9ae/4003/* | \
9ae/4004/* | \
9ae/4005/* | \
9ae/4006/* | \
9ae/4007/* | \
9ae/4008/* | \
d9f/1/* | \
d9f/4/* | \
d9f/a2/* | \
d9f/a3/* | \
d9f/a4/* | \
d9f/a5/* | \
d9f/a6/* | \
f03/1/* | \
10af/1/* | \
10af/4/* | \
10af/8/* | \
14f0/c9/* | \
2b2d/ffff/* | \
ffff/0/* | \
"")
	[ ! -f /var/run/nut/disable-hotplug ] && \
		/etc/init.d/nut-server enabled && perform_libhid_action
	;;
esac
```

cshoredaniel, 2019-01-05:
Merged

closed 2019-01-05 by cshoredaniel

malakingpusa, 2019-01-11:
Thank you very much! Initial testing worked.

Regards,
Jag

<http://xeptocomputing.com/>

PR #7799 (2019-01-02): [18.06] nut: Default to run as root but fix alt runas
Since the new hotplug script in master was not backport (new feature),
for 18.06 branch revert the old behavior of running NUT daemons and
drivers as root by default to avoid permisions problems, but backport
fix the support for running as another user for those who can set the
appropriate permissions on the USB (or other) device.

Closes: #7742

Maintainer: me / @cshoredaniel 
Compile tested: TBD (will update when done) 
Run tested: TBD (will update when done)
**EDIT:**  (compile** and run tested now, see comment below)

Description:

This [... cut]
commit 8f41fa7f (2019-01-02): nut: Default to run as root but fix alt runas

Since the new hotplug script in master was not backport (new feature),
for 18.06 branch revert the old behavior of running NUT daemons and
drivers as root by default to avoid permisions problems, but backport
fix the support for running as another user for those who can set the
appropriate permissions on the USB (or other) device.

Closes: #7742