openwrt/packages #7756: bluez 5.49: can't connect to BLE device

AleksejsC, 2018-12-22:
My device is TP-link WDR3600:
DISTRIB_RELEASE='18.06.1'
DISTRIB_REVISION='r7258-5eb055306f'
DISTRIB_TARGET='ar71xx/generic'
DISTRIB_ARCH='mips_24kc'

Bluetooth is adapter found:
```
# hciconfig
hci0:   Type: Primary  Bus: USB
        BD Address: 00:1A:7D:DA:71:13  ACL MTU: 310:10  SCO MTU: 64:8
        UP RUNNING
        RX bytes:1722 acl:0 sco:0 events:90 errors:0
        TX bytes:1104 acl:0 sco:0 commands:90 errors:0
```
hcitool lescan works fine, but when I try to connect with gatttool:
```
# gatttool -b 4C:65:A8:DB:96:8A -I
[4C:65:A8:DB:96:8A][LE]> connect
Attempting to connect to 4C:65:A8:DB:96:8A
Error: connect: Connection refused (146)
```
hcidump while I'm running gatttool returns nothing:
```
# hcidump
HCI sniffer - Bluetooth packet analyzer ver 5.49
device: hci0 snap_len: 1500 filter: 0xffffffff

^C
```
Logread - nothing.

 With same adapter plugged to my laptop (Ubuntu 18.04 bluez 5.48) evrything works just fine.

I'm trying to run this https://github.com/sullin/hyg it does not work either (works on laptop with the same BT adapter).

Thank you!

diizzyy, 2018-12-22:
Can you give 5.50 a try?

AleksejsC, 2018-12-22:
If you could give me a brief guide on how to do this, I will give a try.

AleksejsC, 2018-12-23:
Today I've tested the same adapter on OpenWrt Chaos Calmer 15.05.1 ramips/rt305x (A5-V11), bluez 5.30, everything is working fine.

diizzyy, 2018-12-26:
You need to compile your own firmware or use snapshots
https://oldwiki.archive.openwrt.org/doc/howto/buildroot.exigence

diizzyy, 2019-01-31:
@thess @hnyman 
Close, submitter timeout

closed 2019-01-31 by hnyman