openwrt/packages #7764: grep: external routing not working

wapebira, 2018-12-24:
Maintainer: @\neheb
Environment: mvebu/cortexa9; OPenwrt 18.06.1 stable and snapshot;  also with Lede SNAPSHOT, r8810-09004e6e13

Description:
Not sure if grep is the issue or other components, but here is what I did:
- flashed the image and installed grep using "opkg install grep"
- reboot the router
- after router comes back, the routing to internet is broken
- tested with external hostname and ip address (eg www.google.com and 172.217.1.4) so that is not the DNS the issue
- this does not apply to local network private ip. they continue to work
- it looks like after 8-9sec the connection drops, coming back after 29-30sec

[image]
[image]

diizzyy, 2018-12-24:
I wouldn't be surprised if some script relies on busybox grep specific behaviour and by replacing it with gnu grep stuff breaks. This is why I don't think we shouldn't replace distro applications, perhaps a better solution is to place it in a directory where it doesn't replace busybox grep.

wapebira, 2018-12-25:
It looks like something is trying to use the busybox grep version. But don't know how to take the necessary logs to investigate any further. I tested with grep 2.26 just to make sure is not greo 3.1 issue and the problem persists.

wapebira, 2018-12-29:
Sorry for mentioning you as maintainer. I correct the initial post.

neheb, 2020-02-09:
Does this still happen with 19.07?

neheb, 2020-03-09:
Closing due to timeout. If still an issue, please reopen.

closed 2020-03-09 by neheb

natumbri, 2020-05-26:
Still happens in trunk...

neheb, 2020-06-12:
Can you debug?

natumbri, 2020-06-13:
Hi @neheb - I can't at the moment, but on a miwifi mini, installing the grep package broke a working system and restoring the link to busybox (without removing grep) fixed it.  I might be able to look more closely in a few weeks. Really difficult to spot when it happens - who would have thought installing grep could break anything?!

neheb, 2020-06-27:
My guess is that this is either a problem with base-files or with netifd.

The way to test this is to replace all instances of grep with busybox grep and see if it works.