openwrt/packages #7785: jq: running immediately causes segmentation fault

MikePetullo, 2018-12-26:
Maintainer: @ratkaj 
Environment: x86_64, OpenWrt master

Description:
Running "jq" immediately generates a segmentation fault.

I tried running jq in GDB. It seems that the line "jq_set_attr(jq, jv_string("JQ_ORIGIN"), jv_string(dirname(origin)));" in src/main.c is causing the segmentation fault. I see this just before executing this line:

```
(gdb) print origin
$1 = 0xfffffffff7f85c20 <error: Cannot access memory at address 0xfffffffff7f85c20>
```

Indeed, running "set origin = argv[0]" in GDB allows execution to continue, at least to the point the program prints its help message.

It appears strdup is failing for some reason.

MikePetullo, 2019-02-11:
See also the following reports:

https://github.com/stedolan/jq/issues/1598
https://github.com/stedolan/jq/issues/1676

closed 2019-03-10 by hnyman

ldvc, 2019-08-04:
Hi! Just installed `jq` on my new Turris Mox and get the `Segmentation fault` error.
Is the binary in the opkg repo still not refreshed?

More info:
```
root@turris:~# cat /etc/openwrt_version
d4ba1f0
root@turris:~# cat /etc/openwrt_release
DISTRIB_ID='TurrisOS'
DISTRIB_RELEASE='4.0-beta7'
DISTRIB_REVISION='d4ba1f0'
DISTRIB_TARGET='mvebu/cortexa53'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='TurrisOS 4.0-beta7 d4ba1f0'
DISTRIB_TAINTS='busybox'
root@turris:~# uname -ar
Linux turris 4.14.132 #0 SMP Mon Jul 22 23:09:55 2019 aarch64 GNU/Linux
root@turris:~# opkg list-installed | grep jq
jq - 1.5-2.0
```

neheb, 2019-08-04:
@ldvc I just pushed the fix to the openwrt-18.06 branch. I don't know how TurrisOS packaging works.

BKPepe, 2019-08-13:
I've just found it. @ldvc the cherry-picked commit made it into Turris OS 4.0-beta8 release, which was released 4 days ago. May I ask you to update your OS and let us know if it works?

ldvc, 2019-08-13:
@BKPepe thanks! `jq` is working properly on 4.0-beta9.

PR #8176 (2019-03-10): jq: compile with _GNU_SOURCE (fixes #7785)

Maintainer: @ratkaj
Compile tested: x86_64
Run tested: x86_64

Description:
Compile with _GNU_SOURCE (fixes #7785)
commit 4bb18b04 (2019-03-10): jq: compile with _GNU_SOURCE (fixes #7785)
commit 8bc5b810 (2019-03-10): Merge pull request #8176 from MikePetullo/jq

jq: compile with _GNU_SOURCE (fixes #7785)
commit 70a1ffdf (2019-08-04): jq: compile with _GNU_SOURCE (fixes #7785)

(cherry-picked from 4bb18b04b6750af9a62b68792415d527f0a54f5e)