openwrt/packages #7794: Package libmosquittopp is missing dependencies for the following libraries: libmosquitto.so.1

stintel, 2018-12-27:
Maintainer: @karlp 
Environment: OpenWrt master

Description:
Trying to build an image with libmosquittopp included and keep running into this issue:
```
Package libmosquittopp is missing dependencies for the following libraries:
libmosquitto.so.1
```
```
$ grep -i mosquitto .config
# CONFIG_PACKAGE_libmosquitto-nossl is not set
CONFIG_PACKAGE_libmosquitto-ssl=y
CONFIG_PACKAGE_libmosquittopp=y
# CONFIG_PACKAGE_mosquitto-client-nossl is not set
# CONFIG_PACKAGE_mosquitto-client-ssl is not set
CONFIG_PACKAGE_mosquitto-nossl=y
# CONFIG_PACKAGE_mosquitto-ssl is not set
```

neheb, 2018-12-31:
@stintel try https://github.com/openwrt/packages/pull/7772

Maybe the libmosquitto depends needs a +.

Or it's not being installed properly.

karlp, 2019-01-01:
It can't be a + depends, because you can't autoselect the virtual package.  I'll be back to work tomorrow and will be looking at this.

karlp, 2019-01-02:
@stintel do you reallllly have mosquitto _without_ ssl, but libmosquitto _with_ ssl?  I mean, I guess that should work, but was that the goal?

karlp, 2019-01-02:
I can confirm the issue if you have mosquitto-ssl + libmosquitto-nossl, but it works if you have mosquitto-ssl+libmosquitto-ssl _or_ mosquitto-nossl + libmosquitto-nossl.

I can only presume this is a build system problem with dependency tracking.  I'm not sure what I can do about it really, and while it _should_ be a legal build combination, it's not really a very _useful_ build combination IMO?

stintel, 2019-01-08:
Using libmosquitto-ssl for vallumd, which connects to a public MQTT broker with HTTPS, and running mosquitto-nossl internally, used by Domoticz and some sensor devices. I don't need SSL for that, and I usually select only what I need.

I can do mosquitto-ssl+libmosquitto-ssl as workaround for now, but would be nice if this can be fixed eventually.

karlp, 2019-01-08:
Well, it's a build issue, not a mosquitto issue, so there's nothing else I can do about it right now.  We could try making -ssl and -nossl variants for libmosquittopp too maybe? might help?