openwrt/packages #7800: firewall: Checking "Enabling logging for this zone" without effect in luci firewall settings for fw3

sigma74, 2018-12-29:
Please make sure that the issue subject starts with `<package-name>: `

This repo here is only for packages maintained in this repo.  For base packages residing in the same repo as the build system and maintained by core devs, please consider opening tickets there for more timely responses

 - OpenWrt base system: https://bugs.openwrt.org
 - Most LuCI packages: https://github.com/openwrt/luci/issues

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

```
Formating code blocks by wrapping them with pairs of ```
```

jow-, 2018-12-29:
Dupe of https://github.com/openwrt/luci/issues/2407

jow-, 2018-12-29:
Make sure your `media` network resolves to actual network devices, otherwise no logging rules will be added.

closed 2018-12-29 by jow-

sigma74, 2018-12-29:
Hmmm,
when ticking the box for the wan zone the log rules appear:
# iptables-save | grep LOG
-A zone_wan_dest_REJECT -o eth0.2 -m limit --limit 10/sec -m comment --comment "!fw3" -j LOG --log-prefix "REJECT wan out: "
-A zone_wan_src_REJECT -i eth0.2 -m limit --limit 10/sec -m comment --comment "!fw3" -j LOG --log-prefix "REJECT wan in: "

So what does your comment mean that I'd have to make sure my "media network resolves to actual network devices"? There are actual network devices using that "media" network (via wireless interface) which is configured to use the "media" firewall zone...

jow-, 2018-12-29:
Check `ifstatus media` and see if `l3_device` resolves to a network device. Also check `fw3 reload` for error messages.

sigma74, 2018-12-29:
Yep, it does:
root@omega:~# ifstatus media
{
	"up": true,
	"pending": false,
	"available": true,
	"autostart": true,
	"dynamic": false,
	"uptime": 4181,
	"l3_device": "wlan0-1",
	"proto": "static",
	"device": "wlan0-1",
	"updated": [
		"addresses"
	],
	"metric": 0,
	"dns_metric": 0,
	"delegation": true,
	"ipv4-address": [
		{
			"address": "192.168.193.254",
			"mask": 24
		}
	],
...

root@omega:~# fw3 reload
Warning: Unable to locate ipset utility, disabling ipset support
Warning: Option @zone[2].conntrack is unknown
Warning: Option @zone[3].conntrack is unknown
 * Clearing IPv4 filter table
 * Clearing IPv4 nat table
 * Clearing IPv4 mangle table
 * Populating IPv4 filter table
   * Rule 'Allow-DHCP-Renew'
   * Rule 'Allow-Ping'
   * Rule 'Allow-IGMP'
   * Rule 'Allow-IPSec-ESP'
   * Rule 'Allow-ISAKMP'
   * Redirect 'SIP'
   * Redirect 'RTP'
   * Redirect 'uShare'
   * Redirect 'DLNA'
   * Redirect 'ssh'
   * Redirect 'NFS'
   * Redirect 'RPC'
   * Forward 'lan' -> 'guest'
   * Forward 'guest' -> 'wan'
   * Forward 'lan' -> 'wan'
   * Forward 'media' -> 'wan'
   * Forward 'lan' -> 'media'
   * Zone 'lan'
   * Zone 'wan'
   * Zone 'guest'
   * Zone 'media'
...

sigma74, 2018-12-29:
Got rid of the two conntrack warnings - to me it seems that fw3 only supports adding logging rules for the default firewall zones "lan" and "wan" then?

There you go:
config defaults
	option syn_flood '1'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'REJECT'

config zone
	option name 'lan'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'
	option network 'lan'

config zone
	option name 'wan'
	option input 'REJECT'
	option output 'ACCEPT'
	option forward 'REJECT'
	option masq '1'
	option mtu_fix '1'
	option network 'wan wan6'

config rule
	option name 'Allow-DHCP-Renew'
	option src 'wan'
	option proto 'udp'
	option dest_port '68'
	option target 'ACCEPT'
	option family 'ipv4'

config rule
	option name 'Allow-Ping'
	option src 'wan'
	option proto 'icmp'
	option icmp_type 'echo-request'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-IGMP'
	option src 'wan'
	option proto 'igmp'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-DHCPv6'
	option src 'wan'
	option proto 'udp'
	option src_ip 'fc00::/6'
	option dest_ip 'fc00::/6'
	option dest_port '546'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-MLD'
	option src 'wan'
	option proto 'icmp'
	option src_ip 'fe80::/10'
	list icmp_type '130/0'
	list icmp_type '131/0'
	list icmp_type '132/0'
	list icmp_type '143/0'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Input'
	option src 'wan'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	list icmp_type 'router-solicitation'
	list icmp_type 'neighbour-solicitation'
	list icmp_type 'router-advertisement'
	list icmp_type 'neighbour-advertisement'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Forward'
	option src 'wan'
	option dest '*'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-IPSec-ESP'
	option src 'wan'
	option dest 'lan'
	option proto 'esp'
	option target 'ACCEPT'

config rule
	option name 'Allow-ISAKMP'
	option src 'wan'
	option dest 'lan'
	option dest_port '500'
	option proto 'udp'
	option target 'ACCEPT'

config include
	option path '/etc/firewall.user'

config redirect
	option target 'DNAT'
	option src 'wan'
	option dest 'lan'
	option proto 'tcp udp'
	option dest_ip '192.168.192.20'
	option name 'SIP'
	list src_mac '38:10:D5:E0:C1:F2'
	option src_ip '192.168.178.1'
	option src_dport '5060-5065'
	option dest_port '5060-5065'

config redirect
	option target 'DNAT'
	option src 'wan'
	option dest 'lan'
	option dest_ip '192.168.192.20'
	option dest_port '7078-7109'
	option name 'RTP'
	list src_mac '38:10:D5:E0:C1:F2'
	option src_ip '192.168.178.1'
	option src_dport '7078-7109'
	option proto 'tcp udp'

config zone
	option name 'guest'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option network 'guest'
	option forward 'ACCEPT'
	option log '1'
	option log_limit '10/minute'

config zone
	option name 'media'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option network 'media'
	option forward 'ACCEPT'
	option log '1'
	option log_limit '10/minute'

config redirect
	option target 'DNAT'
	option src 'media'
	option dest 'lan'
	option dest_ip '192.168.192.10'
	option src_dport '49200'
	option name 'uShare'
	option proto 'tcp'

config redirect
	option target 'DNAT'
	option src 'media'
	option dest 'lan'
	option proto 'udp'
	option dest_ip '192.168.192.10'
	option name 'DLNA'
	option src_dport '1900-1901'
	option dest_port '1900-1901'

config redirect
	option target 'DNAT'
	option src 'media'
	option dest 'lan'
	option dest_ip '192.168.192.10'
	option name 'ssh'
	option proto 'tcp'
	option src_dport '22'
	option dest_port '22'

config redirect
	option target 'DNAT'
	option src 'media'
	option dest 'lan'
	option src_dport '2049'
	option dest_ip '192.168.192.10'
	option dest_port '2049'
	option name 'NFS'
	option proto 'tcp udp'

config redirect
	option target 'DNAT'
	option src 'media'
	option dest 'lan'
	option src_dport '111'
	option dest_ip '192.168.192.10'
	option dest_port '111'
	option name 'RPC'
	option proto 'tcp udp'

config forwarding
	option dest 'wan'
	option src 'guest'

config forwarding
	option dest 'wan'
	option src 'media'

config forwarding
	option dest 'wan'
	option src 'lan'

jow-, 2018-12-29:
Your media zone has all input, output and forward policies set to ACCEPT while logging rules are only bound to DROP or REJECT events, this is why no log rules are instantiated for your media zone.

sigma74, 2018-12-29:
Hmmm,
you have a point there: I modelled the media and guest zones after the lan zone, which in turn wouldn't seem to be covered by logging either (as that uses input, output and forward policies set top ACCEPT as well)...
Thanks for that clarification!