openwrt/packages #7822: ibrdtn failing with uclibc++

neheb, 2018-12-31:
Maintainer: @morgenroth 
Environment: trunk

Description:

Switching to uclibc++ causes compilation failure of ibrdtn. This either requires modification of ibrdtn or more likely uclibc++ to fix compilation.

I've already done so for gptfdisk where support for long long in istream was missing but I can't figure it out for ibrdtn. Maybe a missing iterator? Not a C++ expert.

Here is the build failure: 

```
  CXX      Bundle.lo
In file included from ../../ibrdtn/data/SDNV.h:32:0,
                 from ../../ibrdtn/data/Number.h:25,
                 from ../../ibrdtn/data/Bundle.h:25,
                 from Bundle.cpp:22:
/home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/limits:24:2: warning: #warning limits header is nowhere complete or accurate [-Wcpp]
 #warning limits header is nowhere complete or accurate
  ^~~~~~~
In file included from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/vector:25:0,
                 from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/string:25,
                 from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/stdexcept:22,
                 from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/ibrcommon-1.0/ibrcommon/Exceptions.h:25,
                 from ../../ibrdtn/data/SDNV.h:23,
                 from ../../ibrdtn/data/Number.h:25,
                 from ../../ibrdtn/data/Bundle.h:25,
                 from Bundle.cpp:22:
/home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/algorithm: In instantiation of 'InputIterator std::find(InputIterator, InputIterator, const T&) [with InputIterator = std::list<dtn::data::Bundle::block_elem>::iter_list; T = unsigned char]':
Bundle.cpp:304:46:   required from here
/home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/algorithm:49:32: error: no match for 'operator!=' (operand types are 'dtn::data::Bundle::block_elem' and 'const unsigned char')
   while(first !=last && *first != value){
                         ~~~~~~~^~~~~~~~
In file included from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/memory:23:0,
                 from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/char_traits:22,
                 from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/string:21,
                 from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/stdexcept:22,
                 from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/ibrcommon-1.0/ibrcommon/Exceptions.h:25,
                 from ../../ibrdtn/data/SDNV.h:23,
                 from ../../ibrdtn/data/Number.h:25,
                 from ../../ibrdtn/data/Bundle.h:25,
                 from Bundle.cpp:22:
/home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/iterator_base:191:3: note: candidate: template<class Iterator> bool std::operator!=(const std::reverse_iterator<Iterator>&, const std::reverse_iterator<Iterator>&)
   operator!=(const reverse_iterator<Iterator>& x, const reverse_iterator<Iterator>& y)
   ^~~~~~~~
/home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/iterator_base:191:3: note:   template argument deduction/substitution failed:
In file included from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/vector:25:0,
                 from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/string:25,
                 from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/stdexcept:22,
                 from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/ibrcommon-1.0/ibrcommon/Exceptions.h:25,
                 from ../../ibrdtn/data/SDNV.h:23,
                 from ../../ibrdtn/data/Number.h:25,
                 from ../../ibrdtn/data/Bundle.h:25,
                 from Bundle.cpp:22:
/home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/algorithm:49:32: note:   'dtn::data::Bundle::block_elem' is not derived from 'const std::reverse_iterator<Iterator>'
   while(first !=last && *first != value){
                         ~~~~~~~^~~~~~~~
In file included from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/memory:24:0,
                 from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/char_traits:22,
                 from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/string:21,
                 from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/stdexcept:22,
                 from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/ibrcommon-1.0/ibrcommon/Exceptions.h:25,
                 from ../../ibrdtn/data/SDNV.h:23,
                 from ../../ibrdtn/data/Number.h:25,
                 from ../../ibrdtn/data/Bundle.h:25,
                 from Bundle.cpp:22:
/home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/utility:67:37: note: candidate: template<class T1, class T2> bool std::operator!=(const std::pair<T1, T2>&, const std::pair<T1, T2>&)
  template <class T1, class T2> bool operator!=(const pair<T1,T2>& x, const pair<T1,T2>& y){
                                     ^~~~~~~~
/home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/utility:67:37: note:   template argument deduction/substitution failed:
[... 210 lines cut ...]
                 from ../../ibrdtn/data/Number.h:25,
                 from ../../ibrdtn/data/Bundle.h:25,
                 from Bundle.cpp:22:
/home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/algorithm:49:32: note:   'dtn::data::Bundle::block_elem' is not derived from 'const std::deque<T, Allocator>'
   while(first !=last && *first != value){
                         ~~~~~~~^~~~~~~~
In file included from ../../ibrdtn/data/Bundle.h:37:0,
                 from Bundle.cpp:22:
/home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/set:312:72: note: candidate: template<class Key, class Compare, class Allocator> bool std::operator!=(const std::multiset<Key, Compare, Allocator>&, const std::multiset<Key, Compare, Allocator>&)
 emplate <class Key, class Compare, class Allocator> _UCXXEXPORT bool operator!=
                                                                      ^~~~~~~~
/home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/set:312:72: note:   template argument deduction/substitution failed:
In file included from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/vector:25:0,
                 from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/string:25,
                 from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/stdexcept:22,
                 from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/ibrcommon-1.0/ibrcommon/Exceptions.h:25,
                 from ../../ibrdtn/data/SDNV.h:23,
                 from ../../ibrdtn/data/Number.h:25,
                 from ../../ibrdtn/data/Bundle.h:25,
                 from Bundle.cpp:22:
/home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/algorithm:49:32: note:   'dtn::data::Bundle::block_elem' is not derived from 'const std::multiset<Key, Compare, Allocator>'
   while(first !=last && *first != value){
                         ~~~~~~~^~~~~~~~

```

morgenroth, 2019-01-01:
```
/home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/algorithm:49:32: error: no match for 'operator!=' (operand types are 'dtn::data::Bundle::block_elem' and 'const unsigned char')
```

I do not have the environment to test my hypothesis, but the first error looks like a missing operator within `ibrdtn`. uclibc++ seems to insist on "operator!=" while libstdcpp don't.

Please try this patch:
https://github.com/morgenroth/ibrdtn/commit/199c86591edc7e82b92903efecadc4f69ea63370

neheb, 2019-01-01:
That seemed to fix that error. Next one: 

```
BundleMerger.cpp: In function 'dtn::data::BundleMerger::Container& dtn::data::operator<<(dtn::data::BundleMerger::Container&, const dtn::data::Bundle&)':
BundleMerger.cpp:122:60: error: no matching function for call to 'std::basic_iostream<char>::seekp(std::fpos<char>)'
    (*stream).seekp(obj.fragmentoffset.get<std::streampos>());
                                                            ^
In file included from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/istream:24:0,
                 from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/iostream:27,
                 from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/ibrcommon-1.0/ibrcommon/data/File.h:27,
                 from /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/ibrcommon-1.0/ibrcommon/data/BLOB.h:27,
                 from ../../ibrdtn/data/BundleMerger.h:25,
                 from BundleMerger.cpp:23:
/home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/ostream:119:44: note: candidate: std::basic_ostream<charT, traits>& std::basic_ostream<charT, traits>::seekp(std::basic_ostream<charT, traits>::pos_type) [with charT = char; traits = std::char_traits<char>; std::basic_ostream<charT, traits>::pos_type = int]
   _UCXXEXPORT basic_ostream<charT,traits>& seekp(pos_type pos){
                                            ^~~~~
/home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/ostream:119:44: note:   no known conversion for argument 1 from 'std::fpos<char>' to 'std::basic_ostream<char>::pos_type {aka int}'
/home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/uClibc++/ostream:125:44: note: candidate: std::basic_ostream<charT, traits>& std::basic_ostream<charT, traits>::seekp(std::basic_ostream<charT, traits>::off_type, std::ios_base::seekdir) [with charT = char; traits = std::char_traits<char>; std::basic_ostream<charT, traits>::off_type = int; std::ios_base::seekdir = unsigned char]
   _UCXXEXPORT basic_ostream<charT,traits>& seekp(off_type off, ios_base::seekdir dir){
```

neheb, 2019-01-03:
Replacing streampos with streamoff seemed to do the trick.

Next failure seems to be beyond my knowledge. Maybe another missing operator:

```
MemoryBundleSet.cpp: In member function 'virtual bool dtn::data::MemoryBundleSet::has(const dtn::data::BundleID&) const':
MemoryBundleSet.cpp:137:46: error: conversion from 'std::__base_associative<dtn::data::MetaBundle, dtn::data::MetaBundle, std::less<dtn::data::MetaBundle>, std::allocator<dtn::data::MetaBundle> >::const_iterator {aka std::_associative_citer<dtn::data::MetaBundle, std::less<dtn::data::MetaBundle>, std::allocator<dtn::data::MetaBundle> >}' to non-scalar type 'std::set<dtn::data::MetaBundle>::iterator {aka std::_associative_iter<dtn::data::MetaBundle, std::less<dtn::data::MetaBundle>, std::allocator<dtn::data::MetaBundle> >}' requested
     bundle_set::iterator iter = _bundles.find(dtn::data::MetaBundle::create(bundle));
```

neheb, 2019-01-03:
Two errors this time

```
MetaStorage.cpp: In member function 'std::set<dtn::data::EID> dtn::storage::MetaStorage::getDistinctDestinations() const':
MetaStorage.cpp:69:59: error: conversion from 'dtn::storage::MetaStorage::const_iterator {aka std::_associative_citer<dtn::data::MetaBundle, dtn::storage::MetaStorage::CMP_BUNDLE_PRIORITY, std::allocator<dtn::data::MetaBundle> >}' to non-scalar type 'dtn::data::BundleList::const_iterator {aka std::_associative_citer<dtn::data::MetaBundle, std::less<dtn::data::MetaBundle>, std::allocator<dtn::data::MetaBundle> >}' requested
    for (dtn::data::BundleList::const_iterator iter = begin(); iter != end(); ++iter)
                                                      ~~~~~^~
MetaStorage.cpp:69:68: error: no match for 'operator!=' (operand types are 'dtn::data::BundleList::const_iterator {aka std::_associative_citer<dtn::data::MetaBundle, std::less<dtn::data::MetaBundle>, std::allocator<dtn::data::MetaBundle> >}' and 'dtn::storage::MetaStorage::const_iterator {aka std::_associative_citer<dtn::data::MetaBundle, dtn::storage::MetaStorage::CMP_BUNDLE_PRIORITY, std::allocator<dtn::data::MetaBundle> >}')
    for (dtn::data::BundleList::const_iterator iter = begin(); iter != end(); ++iter)
```

I can't seem to find where the missing operator would be.

diizzyy, 2019-01-08:
Is it really worth hacking just to get uclibc++ working? Most applications in tree seems to depend on libstdcpp which also is noticably faster than uclibc++ although larger but I don't think that a major issue if you want to run C++ applications...

neheb, 2019-01-08:
Don't know that hacking is the correct term here.

From what I've seen, many packages add libstdcpp as the dependency since that's what g++ uses (g++uc uses uclibcxx).

You also can't link between both.

Having said that, all of ibrcommon's dependencies with the exception of ibrdtnd can be compiled with uclibcxx after some work.

diizzyy, 2019-01-08:
I've also read several claims like this one https://github.com/dlitz/openwrt-packages/blob/master/net/nmap/Makefile#L9 which makes we wonder if its worth the hassle.

neheb, 2019-01-08:
For clarity these are the patches that I use:

https://gist.github.com/neheb/16b8f7787ebcd4b575ebb9e665982c8c
https://gist.github.com/neheb/5e953df730492f3f79267d6e1c2eb2d3

diizzyy, 2019-03-28:
@neheb @morgenroth 
Is this something you want to merge?

neheb, 2019-03-28:
https://github.com/neheb/packages/commit/7f9777e9e17a2adb88d71accf1b57676acaee21c

Current status:
ibrcommon - compiles
ibrdtn - compiles
ibrdtn-tools - compiles
ibrdtnd - does not compile

Current failure:
```
 CXX      MetaStorage.lo
In file included from /home/mangix/devstuff/openwrt/staging_dir/target-mips_24kc_musl/usr/include/ibrdtn-1.0/ibrdtn/data/SDNV.h:32:0,
                 from /home/mangix/devstuff/openwrt/staging_dir/target-mips_24kc_musl/usr/include/ibrdtn-1.0/ibrdtn/data/Number.h:25,
                 from /home/mangix/devstuff/openwrt/staging_dir/target-mips_24kc_musl/usr/include/ibrdtn-1.0/ibrdtn/data/MetaBundle.h:25,
                 from ../../src/storage/BundleResult.h:11,
                 from ../../src/storage/BundleSelector.h:11,
                 from ../../src/storage/MetaStorage.h:25,
                 from MetaStorage.cpp:22:
/home/mangix/devstuff/openwrt/staging_dir/target-mips_24kc_musl/usr/include/uClibc++/limits:24:2: warning: #warning limits header is nowhere complete or accurate [-Wcpp]
 #warning limits header is nowhere complete or accurate
  ^~~~~~~
MetaStorage.cpp: In member function 'std::set<dtn::data::EID> dtn::storage::MetaStorage::getDistinctDestinations() const':
MetaStorage.cpp:69:59: error: conversion from 'dtn::storage::MetaStorage::const_iterator {aka std::_associative_citer<dtn::data::MetaBundle, dtn::storage::MetaStorage::CMP_BUNDLE_PRIORITY, std::allocator<dtn::data::MetaBundle> >}' to non-scalar type 'dtn::data::BundleList::const_iterator {aka std::_associative_citer<dtn::data::MetaBundle, std::less<dtn::data::MetaBundle>, std::allocator<dtn::data::MetaBundle> >}' requested
    for (dtn::data::BundleList::const_iterator iter = begin(); iter != end(); ++iter)
                                                      ~~~~~^~
MetaStorage.cpp:69:68: error: no match for 'operator!=' (operand types are 'dtn::data::BundleList::const_iterator {aka std::_associative_citer<dtn::data::MetaBundle, std::less<dtn::data::MetaBundle>, std::allocator<dtn::data::MetaBundle> >}' and 'dtn::storage::MetaStorage::const_iterator {aka std::_associative_citer<dtn::data::MetaBundle, dtn::storage::MetaStorage::CMP_BUNDLE_PRIORITY, std::allocator<dtn::data::MetaBundle> >}')
    for (dtn::data::BundleList::const_iterator iter = begin(); iter != end(); ++iter)
```

This one I can't figure out.

neheb, 2019-10-20:
This has long been fixed.

closed 2019-10-20 by neheb