openwrt/packages #7831: p910nd: Process p9100d disappears after some time

triebelf, 2019-01-01:
Maintainer: @pkerling 
Environment: ARMv7 Processor rev 1 (v7l), Linksys WRT3200ACM, OpenWrt 18.06.1

Description:
After some time (don't know when), process `p9100d` just goes away. Logread or `ps | grep p9100d` shows nothing. The printer is power on very infrequently. After doing `/etc/init.d/p910nd restart`, everything works fine for a while. How can I diagnose and fix this?

I also see this, service is enabled:
```
> ls -l /etc/rc.d/[SK]??p910nd
lrwxrwxrwx    1 root     root            16 Nov 30 19:43 /etc/rc.d/S50p910nd -> ../init.d/p910nd
```

diizzyy, 2019-01-01:
Does logread say anything?

cshoredaniel, 2019-01-02:
Is the device p9100d is supposed to be sharing effectively unplugging and then reconnecting?  Will need to check the hotplug (or lack thereof) script as I'm looking as using this myself soon.

triebelf, 2019-01-02:
Logread output when powering printer on/off:
```
Wed Jan  2 20:08:02 2019 kern.info kernel: [627000.780875] usb 1-1: new full-speed USB device number 5 using orion-ehci
Wed Jan  2 20:08:02 2019 kern.info kernel: [627000.991528] usblp 1-1:1.0: usblp0: USB Bidirectional printer dev 5 if 0 alt 0 proto 2 vid 0x04E8 pid 0x326C
Wed Jan  2 20:08:11 2019 kern.info kernel: [627009.446621] usb 1-1: USB disconnect, device number 5
Wed Jan  2 20:08:11 2019 kern.info kernel: [627009.451809] usblp0: removed
```
Does this mean hotplug? Tell me more about a missing script.

cshoredaniel, 2019-01-02:
@triebelf yep, that means the device in /dev goes away (or at least the app can't use it), so what's needed is probably a a mod to /etc/hotplug.json that does the same thing as the tty subsystem.
The tty subsystem does this:
```
        [ "if",
                [ "and",
                        [ "eq", "SUBSYSTEM", "usb-serial" ],
                        [ "regex", "DEVNAME",
                                [ "^ttyUSB", "^ttyACM" ]
                        ]
                ],
                [ "exec", "/sbin/hotplug-call", "tty" ],
                [ "if",
                        [ "isdir", "/etc/hotplug.d/%SUBSYSTEM%" ],
                        [ "exec", "/sbin/hotplug-call", "%SUBSYSTEM%" ]
                ]
        ]
```
which results in scripts in /etc/hotplug.d/tty being executed with add and remove events when there is a match on usb-serial subsystem for a ttyUSB* or ttyACM* device.

I suppose a similar kind of thing is needed for the usb lp subsystem (although maybe doesn't require the regex bit).

Basically on add we want to run /etc/init.d/p9100d restart so that it comes back up when the device is hotplugged.

triebelf, 2019-01-06:
Hi, finally the weekend, found some time to look at this again.
I see now that the process isn't started when device boots, so I changed the start symlink to /etc/rc.d/S99p910nd
Now it's coming up during start, but I don't see log messages during boot.
Should this be changed in general?

cshoredaniel, 2019-01-06:
@triebelf Might need a proper procd trigger -- see https://github.com/openwrt/packages/pull/7863 for an example.

cshoredaniel, 2019-01-06:
Seems network applications can come up before there are any interfaces.   Either that START=xx is missing (haven't looked yet).

cshoredaniel, 2019-06-02:
So I discovered that if the network changes p910nd needs to be restarted -- I guess socket does something (or goes away) and p910nd doesn't like that and doesn't recover when it happens.

paucoma, 2020-11-11:
I too have had this "issue" appear after "some time" which in my case is related to a network change/or unavailability because I have the p910nd service attached to the **wan interface**.

It appears after a router reboot and wan port disconnected/unavailable to assign address

---
my solution was by adding the following to the end of my **/etc/init.d/p910nd** script:

```
service_triggers(){
        procd_add_interface_trigger "interface.*.up" wan "/etc/init.d/p910nd" restart
        logger -t p910nd "Restarting p910nd service due to up trigger of wan"
}
```

This now restarts the p910nd service when the **wan interface goes up**

---
### Investigation documented

- After the first time it happend, I setup some additional p910nd logging [as described here](https://openwrt.org/docs/guide-user/services/print_server/p910nd.server#p910nd_process_stdout_to_system_log). 

Here is the log output (filtered to relevant data): Network cable on WAN port unplugged and rebooted:

```
Wed Nov 11 11:09:41 2020 kern.notice kernel: [    0.000000] Linux version 4.14.180 (builder@buildhost) (gcc version 7.5.0 (OpenWrt GCC 7.5.0 r11063-85e04e9f46)) #0 Sat May 16 18:32:20 2020
Wed Nov 11 11:09:41 2020 kern.info kernel: [    0.000000] bootconsole [early0] enabled
Wed Nov 11 11:09:41 2020 kern.info kernel: [    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
Wed Nov 11 11:09:41 2020 kern.info kernel: [    0.000000] MIPS: machine is TP-Link TL-WR710N v1
Wed Nov 11 11:09:41 2020 kern.info kernel: [    0.000000] SoC: Atheros AR9330 rev 1
.
..
...
..
.
Wed Nov 11 11:09:41 2020 kern.notice kernel: [    2.561061] random: fast init done
Wed Nov 11 11:09:41 2020 user.info kernel: [    2.929541] init: Console is alive
Wed Nov 11 11:09:41 2020 user.info kernel: [    2.931906] init: - watchdog -
Wed Nov 11 11:09:41 2020 user.info kernel: [    4.156855] kmodloader: loading kernel modules from /etc/modules-boot.d/*
Wed Nov 11 11:09:41 2020 kern.info kernel: [    4.281172] usbcore: registered new interface driver usbfs
Wed Nov 11 11:09:41 2020 kern.info kernel: [    4.285315] usbcore: registered new interface driver hub
Wed Nov 11 11:09:41 2020 kern.info kernel: [    4.290631] usbcore: registered new device driver usb
Wed Nov 11 11:09:41 2020 kern.info kernel: [    4.306238] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
Wed Nov 11 11:09:41 2020 kern.info kernel: [    4.319334] ci_hdrc ci_hdrc.0: EHCI Host Controller
Wed Nov 11 11:09:41 2020 kern.info kernel: [    4.322888] ci_hdrc ci_hdrc.0: new USB bus registered, assigned bus number 1
Wed Nov 11 11:09:41 2020 kern.info kernel: [    4.351059] ci_hdrc ci_hdrc.0: USB 2.0 started, EHCI 1.00
Wed Nov 11 11:09:41 2020 kern.info kernel: [    4.356185] hub 1-0:1.0: USB hub found
Wed Nov 11 11:09:41 2020 kern.info kernel: [    4.359217] hub 1-0:1.0: 1 port detected
Wed Nov 11 11:09:41 2020 user.info kernel: [    4.365003] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
Wed Nov 11 11:09:41 2020 user.info kernel: [    4.373359] init: - preinit -
.
..
...
..
.
Wed Nov 11 11:09:41 2020 kern.info kernel: [    7.532210] eth0: link up (1000Mbps/Full duplex)
Wed Nov 11 11:09:41 2020 kern.info kernel: [   10.178493] eth0: link down
Wed Nov 11 11:09:41 2020 user.info kernel: [   11.244685] procd: - init -
Wed Nov 11 11:09:41 2020 user.info kernel: [   12.316393] kmodloader: loading kernel modules from /etc/modules.d/*
Wed Nov 11 11:09:41 2020 kern.info kernel: [   12.801398] usblp 1-1:1.0: usblp0: USB Bidirectional printer dev 2 if 0 alt 1 proto 2 vid 0x03F0 pid 0x0317
Wed Nov 11 11:09:41 2020 kern.info kernel: [   12.809981] usbcore: registered new interface driver usblp
Wed Nov 11 11:09:57 2020 kern.info kernel: [   32.901628] eth0: link up (1000Mbps/Full duplex)
Wed Nov 11 11:09:57 2020 kern.info kernel: [   32.911950] br-lan: port 1(eth0) entered blocking state
Wed Nov 11 11:09:57 2020 kern.info kernel: [   32.915731] br-lan: port 1(eth0) entered disabled state
Wed Nov 11 11:09:57 2020 kern.info kernel: [   32.921461] device eth0 entered promiscuous mode
Wed Nov 11 11:09:57 2020 kern.info kernel: [   32.963226] br-lan: port 1(eth0) entered blocking state
Wed Nov 11 11:09:57 2020 kern.info kernel: [   32.967008] br-lan: port 1(eth0) entered forwarding state
Wed Nov 11 11:09:57 2020 daemon.notice netifd: Interface 'lan' is enabled
Wed Nov 11 11:09:57 2020 daemon.notice netifd: Interface 'lan' is setting up now
Wed Nov 11 11:09:57 2020 daemon.notice netifd: Interface 'lan' is now up
Wed Nov 11 11:09:57 2020 kern.info kernel: [   33.072060] br-wan: port 1(eth1) entered blocking state
[... 239 lines cut ...]
                                        "if",
                                        [
                                                "eq",
                                                "interface",
                                                "wan"
                                        ],
                                        [
                                                "run_script",
                                                "/etc/init.d/p910nd",
                                                "restart"
                                        ]
                                ],
                                1000
                        ]
                ]
        }
}

```

you can monitor when the trigger is invoked with: 
```
root@OpenWrt:~# ubus monitor | grep '"method":"up","data":{"interface":"wan"}}'
<- 38326d9b #14a31b7d         invoke: {"objid":346233725,"method":"up","data":{"interface":"wan"}}
```

pprindeville, 2020-11-18:
Why would a printer inside the perimeter need `wan` to be up?  I don't get it.

paucoma, 2020-11-18:
> Why would a printer inside the perimeter need `wan` to be up? I don't get it.

In my case its a subnetwork, I don't want the LAN or the Wifi created and managed by the router to have access to whats connected to the WAN but I would like to print to the printer which is connected to the usb port of the router from sources connected to the WAN.

ptpt52, 2020-11-26:
the bind option in openwrt config is the ipaddr, which is determined during configuration
This may not be correct, because the interface may not have an ip address at that time.