openwrt/packages #7832: ipsec-tools: Removal from the packages repo

nmeyerhans, 2019-01-01:
I co-maintain ipsec-tools for both Debian and OpenWRT. After discussion with my co-maintainers in Debian, I opened [bug 917847](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=917847) to propose removing the package from future Debian releases. I'd like to propose the same thing for OpenWRT. My rationale is:

> The package is effectively orphaned upstream and has been for some time. Given the security-sensitive nature of the package, an active maintainer community is essential for safe usage. Racoon's lack of support for IKEv2, despite it being stable for a long time, and the availability of next-generation tunneling systems such as wireguard, also would seem to limit its future value. Setkey's functionality has been subsumed by 'ip xfrm'.
> 
> If you disagree that ipsec-tools should be removed from OpenWRT, please say so now. If there are still use cases for it that are not met by other IKE implmenentations that would be good to know. But more importantly, I think you'll need to convince us that ipsec-tools is actually safe to operate on today's Internet given its current state of development.

CC some people who have also contributed to ipsec-tools in OpenWRT: @aTanW @jow-

cshoredaniel, 2019-01-02:
Removal of a dead security sensitive package makes sense to me.

aTanW, 2019-01-02:
Is there any replacement? Can someone make IPSec connection from OpenWRT in any other way?

cshoredaniel, 2019-01-02:
there is at least strongswan that I know of

champtar, 2019-01-02:
@nmeyerhans could you try to cross post to the forum / mailing list ?
I'm in favor of removal, we can also mark it @broken

diizzyy, 2019-03-28:
@nmeyerhans 
Please do a pull request for removal, you can already consider it approved.

closed 2019-03-28 by diizzyy