openwrt/packages #7834: findutils-find: Cannot install package due to conflict with busybox

johnstonjs, 2019-01-01:
Maintainer: @ cshoredaniel
Environment: ARMv7, Netgear R7800, OpenWrt 18.06.1 r7258-5eb055306f

Description: I'm trying to build a package for [etckeeper](http://etckeeper.branchable.com) and in the process found that _findutils-find_ conflicts with the version installed by _busybox_.

```
root@OpenWRT:~# opkg install findutils-find
Installing findutils-find (4.6.0-1) to root...                                           
Downloading http://downloads.openwrt.org/releases/18.06.1/packages/arm_cortex-a15_neon-vfpv4/packages/findutils-find_4.6.0-1_arm_cortex-a15_neon-vfpv4.ipk                         
Collected errors:
 * check_data_file_clashes: Package findutils-find wants to install file /usr/bin/find   
        But that file is already provided by package  * busybox                          
 * opkg_install_cmd: Cannot install package findutils-find. 
```

In an emulated installation of OpenWRT (in VirtualBox) I was able to override this error by using the flag _--force-overwrite_, however I suspect this isn't the intended behavior.

Thanks for making such an excellent system for routers (and supporting such variety)!

cshoredaniel, 2019-01-02:
I'm working on an PR that uses the new ALTERNATIVES mechanism that should fix this.

cshoredaniel, 2019-01-03:
I've created #7862 to fix this

closed 2019-01-05 by hnyman