openwrt/packages #7858: grub-2.0.2 build/compile failure

ksp222, 2019-01-03:
Please make sure that the issue subject starts with `<package-name>: `

This repo here is only for packages maintained in this repo.  For base packages residing in the same repo as the build system and maintained by core devs, please consider opening tickets there for more timely responses

 - OpenWrt base system: https://bugs.openwrt.org
 - Most LuCI packages: https://github.com/openwrt/luci/issues

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

```
Formating code blocks by wrapping them with pairs of ```
```

ksp222, 2019-01-03:
Issue: grub 2.0.2 build fails
Environment: Fedora 29 Workstation

Openwrt Image configured for the following:
------------------------------------------------------
Target System (x86)  --->                                                                                                                                                                                                                                                                                 Subtarget (AMD Geode based systems)  --->                                                                                                                                             
Target Profile (Generic)  --->
-------------------------------------------------------

Build log:
make  all-am
make[7]: Entering directory '/opt/openwrt/build_dir/hostpkg/grub-2.02/grub-core'
i486-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I..  -Wall -W  -DGRUB_MACHINE_PCBIOS=1 -DGRUB_MACHINE=I386_PC -m32 -nostdinc -isystem /opt/openwrt/staging_dir/toolchain-i386_pentium_gcc-7.4.0_musl/lib/gcc/i486-openwrt-linux-musl/7.4.0/include -I../include -I../include -DGRUB_FILE=\"lib/i386/relocator64.S\" -I. -I. -I.. -I.. -I../include -I../include -I../grub-core/lib/libgcrypt-grub/src/    -I/opt/openwrt/staging_dir/host/include -I/opt/openwrt/staging_dir/hostpkg/include -I/opt/openwrt/staging_dir/target-i386_pentium_musl/host/include -D_FILE_OFFSET_BITS=64 -g  -m32 -msoft-float -DGRUB_FILE=\"lib/i386/relocator64.S\" -I. -I. -I.. -I.. -I../include -I../include -I../grub-core/lib/libgcrypt-grub/src/ -DASM_FILE=1    -O2 -I/opt/openwrt/staging_dir/host/include -I/opt/openwrt/staging_dir/hostpkg/include -I/opt/openwrt/staging_dir/target-i386_pentium_musl/host/include -MT lib/i386/relocator_module-relocator64.o -MD -MP -MF lib/i386/.deps-core/relocator_module-relocator64.Tpo -c -o lib/i386/relocator_module-relocator64.o `test -f 'lib/i386/relocator64.S' || echo './'`lib/i386/relocator64.S
lib/i386/relocator64.S: Assembler messages:
lib/i386/relocator64.S:66: Error: unknown pseudo-op: `.code64'
lib/i386/relocator64.S:74: Error: bad register name `%rax'
lib/i386/relocator64.S:98: Error: bad register name `%rax'
lib/i386/relocator64.S:132: Error: bad register name `%rip)'
make[7]: *** [Makefile:28991: lib/i386/relocator_module-relocator64.o] Error 1
make[7]: Leaving directory '/opt/openwrt/build_dir/hostpkg/grub-2.02/grub-core'
make[6]: *** [Makefile:23526: all] Error 2
make[6]: Leaving directory '/opt/openwrt/build_dir/hostpkg/grub-2.02/grub-core'
make[5]: *** [Makefile:10899: all-recursive] Error 1
make[5]: Leaving directory '/opt/openwrt/build_dir/hostpkg/grub-2.02'
make[4]: *** [Makefile:3125: all] Error 2
make[4]: Leaving directory '/opt/openwrt/build_dir/hostpkg/grub-2.02'
make[3]: *** [Makefile:92: /opt/openwrt/build_dir/hostpkg/grub-2.02/.built] Error 2
make[3]: Leaving directory '/opt/openwrt/package/boot/grub2'
time: package/boot/grub2/host-compile#1.26#0.29#1.61
make[2]: *** [package/Makefile:107: package/boot/grub2/host/compile] Error 2
make[2]: Leaving directory '/opt/openwrt'
make[1]: *** [package/Makefile:103: /opt/openwrt/staging_dir/target-i386_pentium_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/opt/openwrt'
make: *** [/opt/openwrt/include/toplevel.mk:218: world] Error 2

diizzyy, 2019-01-31:
@thess @hnyman 
Close as invalid

Wrong repo, it's in OpenWrt main repo

closed 2019-01-31 by hnyman