openwrt/packages #7873: DDNS private IP. No update sent

dailylama, 2019-01-04:
```
ddns-scripts: No or private or invalid IP
ddns-scripts: No update send to DDNS Provider
```

freedns.afraid.org is using update url with `random_token` of the form:

`https://sync.afraid.org/u/random_token`

Website (freedns.afraid.org) updates the hostname associated with that `random_token`. Points hostname to the IP from which request to the website was made.

```
config service 'afraid'
	option enabled		'1'
	option service_name	'freedns.afraid.org'
	option domain		'hostname'
	option update_url	'https://sync.afraid.org/u/random_token/'
	option interface	'wan'
	option ip_source	'network'
	option ip_network	'wan'
	option use_https	'1'
	option cacert		'/etc/ssl/certs'
```

cshoredaniel, 2019-01-07:
What do the ddns logs show?

dailylama, 2019-01-07:
`No update send to DDNS Provider`

cshoredaniel, 2019-01-07:
Can you give more complete / verbose logs?  DDNS is capable of showing a lot more than that.

dailylama, 2019-01-07:
If you can't find the issue, here is the line I've commented as a temporary fix in my local copy of the script

https://github.com/openwrt/packages/blob/b3629b9923eb09d7ddd086802fdee95cab57025f/net/ddns-scripts/files/dynamic_dns_functions.sh#L876

cshoredaniel, 2019-01-07:
I'm assuming you are using ipv4:

The code in question is below.

so first: do you **want** a private ip?  If so then you've failed to enable that option in your UCI.
second: if not, either the IP is not being detected on the wan network or it doesn't match the regexp.

To know that we need more of the logs (there is a verbose setting that will allow to actually see what is going on.  Please enable it and report the results.  Don't expect us to do the work for you -- no one is getting paid to look at this, and unless you do the necessary work and give the needed information probably no one is going to have enough time to spend on it.

```
if [ $upd_privateip -eq 0 ]; then
		# verify given IP / no private IPv4's / no IPv6 addr starting with fxxx of with ":"
		[ $use_ipv6 -eq 0 ] && __IP=$(echo $1 | grep -v -E "(^0|^10\.|^100\.6[4-9]\.|^100\.[7-9][0-9]\.|^100\.1[0-1][0-9]\.|^100\.12[0-7]\.|^127|^169\.254|^172\.1[6-9]\.|^172\.2[0-9]\.|^172\.3[0-1]\.|^192\.168)")
		[ $use_ipv6 -eq 1 ] && __IP=$(echo $1 | grep "^[0-9a-eA-E]")
	else
		__IP=$(echo $1 | grep -m 1 -o "$IPV4_REGEX")		# valid IPv4 or
		[ -z "$__IP" ] && __IP=$(echo $1 | grep -m 1 -o "$IPV6_REGEX")	# IPv6
	fi
	[ -z "$__IP" ] && {
		write_log 3 "No or private or invalid IP '$1' given! Please check your configuration"
		return 127
}
```

akostadinov, 2022-04-06:
There is a Luci DDNS global configuration option `Allow non-public IP's`. If you want your private IP to be sent, then enable it. It is off by default.