openwrt/packages #7881: adblock fails downloading SSL lists when using uclient-fetch.

schuft69, 2019-01-05:
Maintainer: @neheb  (hope your' the right one)
Environment: OpenWrt 18.06.1 r7258-5eb055306f  + Adblock Version 3.5.5-2

Description:

The standard download binary for fetching lists is uclient-fetch.
If the user chooses a list which needs SSL the download will fail (without any warning) because libustream-ssl is not delivered with the standard Openwrt image. That can only be found with debugging enabled.
```
Sat Jan  5 10:18:21 2019 user.debug adblock-3.5.5-2[1840]: f_main ::: name: reg_de, url: https://easylist-downloads.adblockplus.org/easylistgermany+easylist.txt, rc: 1, log: /bin/uclient-fetch: SSL support not available, please install on
e of the libustream-.*[ssl|tls] packages as well as the ca-bundle and ca-certificates packages.
```

Installing libustream-ssl solves this issue.

adblock should either install libustream-ssl by default (if uclient-fetch is used) OR warn the user about that dependency (without having debugging installed.

Regards,
Schuft

neheb, 2019-01-05:
@dibdot is the maintainer.

A TLS library is not installed by default for space reasons.

FireCulex, 2020-02-09:
It looks like all the lists use https. Not sure why it's optional since we're forced to install it anyway. Perhaps make SSL an option like libustream-mbedtls in the buildroot ?