openwrt/packages #7884: fwknop: Don't globally source network.sh

cshoredaniel, 2019-01-06:
See [How to fix in comment on PR#5005](https://github.com/openwrt/packages/pull/5005/files#r245496105)

Basically global sourcing results in error message during image build as the script is executed on the **host** in order to set the START/STOP symlinks.

yousong, 2019-01-07:
Right now it should be just a cosmetic issue.  If it matters, I prefer it to be a treewide change, not doing it bit by bit with inconsistency in the process.

```
➜  ~/git-repo/openwrt/packages git:(master) ag -l  '^\. [^$]'  | grep -vE '/sbin/|hotplug'
libs/avahi/files/netifd-autoip.sh
utils/lxc/files/lxc-auto.init
utils/rtty/files/rtty.init
utils/bmx7-dnsupdate/files/usr/bin/bmx7-dnsupdate
net/linknx/files/linknx.init
net/unbound/files/root.key
net/unbound/files/stopping.sh
net/unbound/files/unbound.ntpd
net/unbound/files/odhcpd.sh
net/unbound/files/unbound.sh
net/fwknop/files/fwknopd.init
net/vpnc/files/vpnc.sh
net/ipsec-tools/files/p1client-up
net/ipsec-tools/files/functions.sh
net/ipsec-tools/files/p1client-down
net/ipsec-tools/files/racoon.init
net/acme/files/run.sh
net/ddns-scripts/files/dynamic_dns_lucihelper.sh
net/openconnect/files/openconnect.sh
net/ddns-scripts/files/dynamic_dns_functions.sh
net/vpnc-scripts/files/vpnc-script
net/apcupsd/files/onbattery
net/apcupsd/files/commfailure
net/apcupsd/files/offbattery
net/apcupsd/files/changeme
net/apcupsd/files/commok
net/safe-search/files/safe-search-update
net/pppossh/files/pppossh.sh
net/nut/files/30-libhid-ups.head
net/nut/files/nutshutdown
net/nut/files/nut-sendmail-notify.default
net/bind/files/bind/db.root
net/miniupnpd/files/firewall.include
net/nft-qos/files/lib/priority.sh
net/nft-qos/files/lib/core.sh
net/nft-qos/files/lib/monitor.sh
net/nft-qos/files/lib/dynamic.sh
net/nft-qos/files/nft-qos.init
net/nft-qos/files/lib/static.sh
net/mwan3/files/usr/libexec/rpcd/mwan3
net/bcp38/files/run.sh
net/bonding/files/lib/netifd/proto/bonding.sh
```

yousong, 2019-01-07:
By the way, we can use `$IPKG_INSTROOT` for sourcing those files.

cshoredaniel, 2019-01-07:
Hrrmm...okay I'll poke at this after fork...

cshoredaniel, 2019-04-17:
A combination of factors has interfered with working on much of anything non-local for 2-3 months now and in any event I am considering dropping OpenWrt from my roster of 'Things To Do' even when the difficulties are resolved, as I need to consider where too allocate my limited time.

closed 2019-04-17 by cshoredaniel