openwrt/packages #7907: Cannot download packages in 18.06.1

petroul, 2019-01-07:
I had to upgrade to OpenWrt 18.06.1 and can no longer get packages.
I reported this initially in https://github.com/openwrt/luci/issues/2440 and followed up by changing file /etc/opkg/distfeeds.conf and changed **https** into **http**
So now it is:
*src/gz openwrt_core http://downloads.openwrt.org/releases/18.06.1/targets/ar71xx/generic/packages
src/gz openwrt_base http://downloads.openwrt.org/releases/18.06.1/packages/mips_24kc/base
src/gz openwrt_luci http://downloads.openwrt.org/releases/18.06.1/packages/mips_24kc/luci
src/gz openwrt_packages http://downloads.openwrt.org/releases/18.06.1/packages/mips_24kc/packages
src/gz openwrt_routing http://downloads.openwrt.org/releases/18.06.1/packages/mips_24kc/routing
src/gz openwrt_telephony http://downloads.openwrt.org/releases/18.06.1/packages/mips_24kc/telephony*

Still I am not able to update package list. It gives errors like:
Downloading http://downloads.openwrt.org/releases/18.06.1/targets/ar71xx/generic/packages/Packages.gz
*** Failed to download the package list from http://downloads.openwrt.org/releases/18.06.1/targets/ar71xx/generic/packages/Packages.gz_

I can download the package on my PC and scp it to the router. However this is tedious job because I don't know the relations between the packages.

Any input is welcome. Thanks in advance.

ImAnnoying2, 2019-01-08:
Try changing all the http:// to https://

cshoredaniel, 2019-01-09:
Where did you get the image?  It's not from an 'official' build it would seem, so you might need to contact the makers of the image.

petroul, 2019-01-09:
I used the file: *openwrt-18.06.1-ar71xx-generic-tl-wdr4300-v1-squashfs-sysupgrade.bin*
from location: *https://downloads.openwrt.org/releases/18.06.1/targets/ar71xx/generic/*

I guess this is the official site.

Any input is welcome!

cshoredaniel, 2019-01-10:
Are you sure you didn't use openwrt.com which someone is (apparently) using to distribute malware?  AFAIK official builds using http:// in their distfeeds.conf and shouldn't need any configuration like you've done.

yousong, 2019-01-11:
I checked the code.  It seems to me that for every possible path to cause `*** Failed to download the package list from xxx` there also exist at least one call to print error message explaining why, but this is not the case from the info you provided.

There are a few thing worth trying

 - Enable verbose output by running `opkg update -V 5`
 - Run opkg under strace `strace -s 512 opkg update`
 - If it's a network issue, capture and analyze the traffic with tcpdump

cshoredaniel, 2019-01-11:
@petroul what did you upgrade from? and did you preserve configuration.  If so, can you factory reset and reconfigure your device?

petroul, 2019-01-11:
> @petroul what did you upgrade from? and did you preserve configuration. If so, can you factory reset and reconfigure your device?

I upgraded from version 15.05.1 and have a configuration file containing the */etc* with following contents:
[image]

What do you recommend?

jow-, 2019-01-11:
```
cp /rom/etc/opkg.conf /etc/opkg.conf
cp /rom/etc/opkg/customfeeds.conf /etc/opkg/customfeeds.conf
cp /rom/etc/opkg/kernelfeed.conf /etc/opkg/kernelfeed.conf
cp /rom/etc/opkg/distfeeds.conf /etc/opkg/distfeeds.conf
```

petroul, 2019-01-13:
All, thanks for your input. So far little success. Finally I decided to do a full Installation using *openwrt-18.06.1-ar71xx-generic-tl-wdr4300-v1-squashfs-factory.bin* renamed to *wdr4300v1_en_3_14_3_up_boot(150518).bin*. The result is not (yet) as expected:
- Router is known in modem.
- PC connected to router has internet access.
- Ping from PC (*192.168.1.105*) to router result is: 
[image]
- Access to 192.168.1.101 results to:
[image]
after a few seconds browser reports *ERR_CONNECTION_TIMED_OUT*
- ssh using putty to *192.168.1.101* port *22* results in *Network error: Connection timed out*

Any input is welcome!

petroul, 2019-01-13:
> I checked the code. It seems to me that for every possible path to cause `*** Failed to download the package list from xxx` there also exist at least one call to print error message explaining why, [...]

Thanks. Verbose gave same errors; strace package is not on the router; tcpdump gave same information.

petroul, 2019-01-13:
> @petroul what did you upgrade from? and did you preserve configuration. If so, can you factory reset and reconfigure your device?

I did (upgraded from 15.05.1) no result. Next I did the full installation, see results above.

cshoredaniel, 2019-01-14:
FYI by default OpenWrt is not open on the WAN side of the device
(otherwise it would open to the internet in the usual case (which is not
double-NAT, but having the modem in bridge mode and having the router as
the edge (internet-facing) device)).  You need to connect your computer
the LAN side of the router and either use DHCP (to get a 192.168.1.x
address) or statically assign a 192.168.1.x address to you computer. 
You can then SSH into root@192.168.1.1 (default is no password and will
warn you to set a password ASAP, this was done to avoid the false sense
of security of a default (known) password).  If you have an image with
LuCI you can also log into the web UI at http://192.168.1.1.

HTH

petroul, 2019-01-14:
Unfortunately not.
See the first results I reported 6 hours before your post, starting with *All, thanks for your input. So far little success.*
All you advice has been checked with no results.

Any input is welcome!

yousong, 2019-01-15:
Seems the router's lan side ip is 192.168.1.101 and you are now locked out of it.  Or no one is actually at 192.168.1.101 as most of the time host unreachable from ping is caused by lack of arp response.

cshoredaniel, 2019-01-15:
Ok, could you describe your setup a little more.  Is your modem giving
out addresses in the 192.168.1.0/24 range?  If the that is the case and
the router is connected to the modem (which is in non-bridge mode), that
will conflict with the default LAN side of the router and lead to issues.

Can you try rebooting with no WAN connection and try connecting to the
LAN of the router? (That is, if the modem is indeed not in bridge mode
or your ISP does something weird and gives out 192.168.1.0/24 addresses;
I've heard stuff like that does happen for what is overseas for me).

petroul, 2019-01-16:
> Ok, could you describe your setup a little more.  Is your modem giving out addresses in the 192.168.1.0/24 range?  If the that is the case and the router is connected to the modem (which is in [...]

**Daniel, THX a lot!**
You were right I got access after removing link to the modem/router combination!
Now how to move on? Herewith some info on my configuration:
-Provider does not allow me to put modem/router in bridge mode;
-modem/router has default DHCP on.
So how to get the OpenWRT router operational with a link to the modem/router?

cshoredaniel, 2019-01-17:
@petroul while disconnected from modem, change you LAN IP range (you might want to check https://forums.openwrt.org for some advice on that as it can be tricky and/or https://openwrt.org (the wiki).

petroul, 2019-01-17:
I am back on the original status.
OpenWRT completely reinstalled using *openwrt-18.06.1-ar71xx-generic-tl-wdr4300-v1-squashfs-factory.bin*
Router has internet access.
Hardly any configuration possible. Results:
[image]
Also applicable for http configuration:
[image]
Supporting packages not availably by default:
[image]
Default OPKG configuration is:
[image]

Any support is welcome!

petroul, 2019-01-17:
On the OpenWrt router I have a PC and a Linux device connected.
PC and Linux device are connecting to the internet, eg *ping openwrt.org* is succesfull.
On the OpenWrt router itself I can ping the PC (192.168.1.105) however I cannot reach openwrt.org.
[image]
Somehow modem/router is blocking traffic from OpenWrt router and accepting traffic of any device connected to the OpenWrt router.

Any support is welcome!

cshoredaniel, 2019-01-18:
@petroul This is no longer a bug report but a support request.  Please use the forums (https://forums.openwrt.org) or self-help (wiki) at https://openwrt.org at this point.

@hnyman @jow- @thess @yousong I think this can now be closed.

amalgamy, 2020-02-20:
Package Bootstrapping Issue: 

Perhaps a documentation update is needed to add the steps for manually enabling TLS wget support [in the packages user guide](https://openwrt.org/packages/start)?

The nginx frontend for downloads.openwrt.org sends clients a redirect for HTTP --> HTTP upgrade, which breaks the default wget in OpenWRT images, since TLS support is not bundled in the default images. This server behavior protects against MitM HTTP attacks, but also blocks package downloads on a default OpenWRT installation. 

This prevents my default (official image) installation from downloading packages - neither by command line 'opkg update' nor by luci interface (since both use the default wget package). 

Attempts to download package lists will fail with the a mystery error with the default distfeeds.conf , but the underlying errors are visible by changing the URL scheme from http to https (which duplicates the effect of the HTTP Location redirect when my router connects) 

>   wget: SSL support not available, please install one of the libustream-.*[ssl|tls] packages as well as the ca-bundle and ca-certificates packages.

Problem verified with an official OpenWRT 18.06.7 image downloaded today:
OpenWrt 18.06.7 r7976-ca47026b7d / LuCI openwrt-18.06 branch (git-20.029.49294-41e2258)

Testing with curl shows the 301 redirect for base package URL download - same forced redirect happens for wget on my router:

== Info: Connected to downloads.openwrt.org (176.9.48.73) port 80 (#0)
=> Send header, 182 bytes (0xb6)
0000: GET /releases/18.06.7/packages/mips_24kc/base HTTP/1.1
0038: Host: downloads.openwrt.org
0055: User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:69.0) Gecko/20100
0095: 101 Firefox/69.0
00a7: Accept: */*
<= Recv header, 32 bytes (0x20)
0000: HTTP/1.1 301 Moved Permanently
<= Recv header, 22 bytes (0x16)
0000: Server: nginx/1.10.3
 ...
<= Recv header, 82 bytes (0x52)
0000: Location: http://downloads.openwrt.org/releases/18.06.7/packages
0040: /mips_24kc/base/

Note: I'm aware of these previous bug reports:

* https://bugs.openwrt.org/index.php?do=details&task_id=68&string=wget+SSL+support+not+available&advancedsearch=on&search_in_comments=1&search_in_details=1&search_for_all=1&type%5B0%5D=1&sev%5B0%5D=&pri%5B0%5D=&due%5B0%5D=&reported%5B0%5D=&cat%5B0%5D=&status%5B0%5D=&percent%5B0%5D=&opened=&dev=&closed=&duedatefrom=&duedateto=&changedfrom=&changedto=&openedfrom=&openedto=&closedfrom=&closedto= 
* https://github.com/openwrt/luci/issues/2440 
* https://github.com/openwrt/luci/issues/2453 
* https://github.com/openwrt/packages/issues/7907

amalgamy, 2020-02-20:
DNS error masquerading as an HTTPS problem
Issue resolved by updating the default /etc/resolv.conf

Mateando, 2020-10-27:
hahaha, I had the same problem, I needed to configure the DNS, the system simply did not respond to the names

twelvemogoddamming, 2021-04-09:
> DNS error masquerading as an HTTPS problem
> Issue resolved by updating the default /etc/resolv.conf

Hi. What did you do exactly? I am experiencing the same thing. Thank yoooou!

amalgamy, 2021-04-10:
I edited the upstream (WAN) DNS servers in LuCI and added Google and Cloudflare public DNS servers to the  "Use custom DNS servers" fields.

Assuming your router has the default IPv4 configuration, try navigating to this URL for the 
"Interfaces >> WAN" section:
  http://192.168.1.1/cgi-bin/luci/admin/network/network

Add public DNS servers (or your IPS's DNS servers), then restart the WAN interface.