openwrt/packages #7915: CircleCI / python{,3}: Doesn't fail on missing dependencies

cshoredaniel, 2019-01-08:
So I found out the hard way that CircleCI will pass python packages which have dependencies on other python packages, even if the other packages are not actually available.  This probably should be considered a fatal error, as otherwise things can look fine that will have missing dependencies when merged.

thess, 2019-01-09:
Can I have another hint? Example?

cshoredaniel, 2019-01-09:
Also https://github.com/openwrt/packages/pull/7871 should have failed as it depends on vobject which isn't in the tree yet.

cshoredaniel, 2019-01-09:
And vobject https://github.com/openwrt/packages/pull/7900 should have failed because it was pushed before the python3-dateutil update.

cshoredaniel, 2019-04-17:
A combination of factors has interfered with working on much of anything non-local for 2-3 months now and in any event I am considering dropping OpenWrt from my roster of 'Things To Do' even when the difficulties are resolved, as I need to consider where too allocate my limited time.

closed 2019-04-17 by cshoredaniel