openwrt/packages #7917: miniupnpd: /etc/init.d/miniupnpd not work

GongT, 2019-01-08:
# Issue template (remove lines from top till here)

Maintainer: @Kevin Darbyshire-Bryant <ldir@darbyshire-bryant.me.uk>
Environment: `OpenWrt 18.06.1, r7258-5eb055306f` `armv7l`

Description:

```text
start-stop-daemon: unrecognized option: f
BusyBox v1.28.3 () multi-call binary.

Usage: start-stop-daemon [OPTIONS] [-S|-K] ... [-- ARGS...]

Search for matching processes, and then
-K: stop all matching processes
-S: start a process unless a matching process is found

Process matching:
        -u USERNAME|UID Match only this user's processes
        -n NAME         Match processes with NAME
                        in comm field in /proc/PID/stat
        -x EXECUTABLE   Match processes with this command
                        command in /proc/PID/cmdline
        -p FILE         Match a process with PID from FILE
        All specified conditions must match
-S only:
        -x EXECUTABLE   Program to run
        -a NAME         Zeroth argument
        -b              Background
        -c USER[:[GRP]] Change user/group
        -m              Write PID to pidfile specified by -p
-K only:
        -s SIG          Signal to send
        -t              Match only, exit with 0 if found
Other:
        -q              Quiet
```

Add some log into it, The command will run is:
> SERVICE_DAEMONIZE= service_start /usr/sbin/miniupnpd -f /var/etc/miniupnpd.conf

neheb, 2020-02-07:
This crap needs procd badly.

neheb, 2020-02-07:
Is this problem still present?

neheb, 2020-02-07:
I reproduced this on malta. I'm closing this as WONTFIX.

The issue is that the init script does not populate the interface options. At a minimum, you need TWO interfaces for this to work properly, one external and one internal.

basically, network_find_wan fails and so does network_get_device . There's no good way to solve this unless bogus defaults are placed.

closed 2020-02-07 by neheb