openwrt/packages #7934: ulogd2 json timestamp 1969 without realtime clock

CanadianJeff, 2019-01-10:
this bug only affects devices that do not have a real time clock [RTC] installed in them

my APU2 is a full on x86 board and comes with a bios battery and [RTC] to keep date/time across reboots

my TP-LINK WDR4300 is just a simple store bought cheapo router which does the have a [RTC] installed so the timestamps in ulogd firewall logs all show up as 1969 even after the device has grabbed the updated date time from a valid NTP server

CanadianJeff, 2019-01-10:
not sure if this bug exists in the new version 2.0.7 listed on the official website
https://netfilter.org/projects/ulogd/downloads.html

CanadianJeff, 2019-01-10:
the file in question would be
https://git.netfilter.org/ulogd2/tree/output/ulogd_output_JSON.c
and
https://github.com/openwrt/packages/tree/master/net/ulogd

neheb, 2020-01-21:
My guess is that this line is the problem: https://git.netfilter.org/ulogd2/tree/output/ulogd_output_JSON.c#n299

No idea why they don't use use time(NULL)

neheb, 2020-01-28:
Maybe check /var/log/ulogd.log for any hints.

neheb, 2020-02-07:
Yeah I was right: https://git.netfilter.org/ulogd2/commit/?id=2b39df550fbad944b4aab77617d4272c5d62ba70

It seems this packet timestamp routine is failing.

CanadianJeff, 2020-12-19:
closing this issue as it might no longer be needed feel free to reopen

closed 2020-12-19 by CanadianJeff