openwrt/packages #7935: python3: ModuleNotFoundError: No module named '_sqlite3'

allaudet, 2019-01-10:
Maintainer: @commodo
Environment: **imx6**, **18.04 and master**

Description:

I was trying to compile a custom python package ([diskcache](https://pypi.org/project/diskcache)) that uses the sqlite3 python library.

When I compile it for Python2 there is no problem, but when I do it for **Python3** then this error shows up:
```
Traceback (most recent call last):
  File "./setup.py", line 6, in <module>
    import diskcache
  File "/home/openwrt/openwrt/build_dir/target-arm_cortex-a7+neon_musl_eabi/diskcache-3.1.1/diskcache/__init__.py", line 3, in <module>
    from .core import Cache, Disk, UnknownFileWarning, EmptyDirWarning, Timeout
  File "/home/openwrt/openwrt/build_dir/target-arm_cortex-a7+neon_musl_eabi/diskcache-3.1.1/diskcache/core.py", line 13, in <module>
    import sqlite3
  File "/home/openwrt/openwrt/staging_dir/target-arm_cortex-a7+neon_musl_eabi/usr/lib/python3.6/sqlite3/__init__.py", line 23, in <module>
    from sqlite3.dbapi2 import *
  File "/home/openwrt/openwrt/staging_dir/target-arm_cortex-a7+neon_musl_eabi/usr/lib/python3.6/sqlite3/dbapi2.py", line 27, in <module>
    from _sqlite3 import *
ModuleNotFoundError: No module named '_sqlite3'
```

Looks like he cannot reach the library, but when I search the library file I see this:
```
$ find . -name "_sqlite3*"
./staging_dir/target-arm_cortex-a7+neon_musl_eabi/root-imx6/usr/lib/python3.6/lib-dynload/_sqlite3.cpython-36.so
./staging_dir/target-arm_cortex-a7+neon_musl_eabi/root-imx6/usr/lib/python2.7/lib-dynload/_sqlite3.so
./build_dir/target-arm_cortex-a7+neon_musl_eabi/Python-2.7.15/PCbuild/_sqlite3.vcxproj
./build_dir/target-arm_cortex-a7+neon_musl_eabi/Python-2.7.15/PCbuild/_sqlite3.vcxproj.filters
./build_dir/target-arm_cortex-a7+neon_musl_eabi/Python-2.7.15/.pkgdir/python-sqlite3/usr/lib/python2.7/lib-dynload/_sqlite3.so
./build_dir/target-arm_cortex-a7+neon_musl_eabi/Python-2.7.15/PC/VS8.0/_sqlite3.vcproj
./build_dir/target-arm_cortex-a7+neon_musl_eabi/Python-2.7.15/PC/VS7.1/_sqlite3.vcproj
./build_dir/target-arm_cortex-a7+neon_musl_eabi/Python-2.7.15/PC/VS9.0/_sqlite3.vcproj
./build_dir/target-arm_cortex-a7+neon_musl_eabi/Python-2.7.15/PC/VC6/_sqlite3.dsp
./build_dir/target-arm_cortex-a7+neon_musl_eabi/Python-2.7.15/ipkg-install/usr/lib/python2.7/lib-dynload/_sqlite3.so
./build_dir/target-arm_cortex-a7+neon_musl_eabi/Python-2.7.15/build/lib.linux2-2.7/_sqlite3.so
./build_dir/target-arm_cortex-a7+neon_musl_eabi/Python-2.7.15/ipkg-arm_cortex-a7_neon/python-sqlite3/usr/lib/python2.7/lib-dynload/_sqlite3.so
./build_dir/target-arm_cortex-a7+neon_musl_eabi/Python-3.6.5/PCbuild/_sqlite3.vcxproj
./build_dir/target-arm_cortex-a7+neon_musl_eabi/Python-3.6.5/PCbuild/_sqlite3.vcxproj.filters
./build_dir/target-arm_cortex-a7+neon_musl_eabi/Python-3.6.5/.pkgdir/python3-sqlite3/usr/lib/python3.6/lib-dynload/_sqlite3.cpython-36.so
./build_dir/target-arm_cortex-a7+neon_musl_eabi/Python-3.6.5/ipkg-install/usr/lib/python3.6/lib-dynload/_sqlite3.cpython-36.so
./build_dir/target-arm_cortex-a7+neon_musl_eabi/Python-3.6.5/build/lib.linux2-3.6/_sqlite3.cpython-36.so
./build_dir/target-arm_cortex-a7+neon_musl_eabi/Python-3.6.5/ipkg-arm_cortex-a7_neon/python3-sqlite3/usr/lib/python3.6/lib-dynload/_sqlite3.cpython-36.so
./build_dir/hostpkg/Python-2.7.15/PCbuild/_sqlite3.vcxproj
./build_dir/hostpkg/Python-2.7.15/PCbuild/_sqlite3.vcxproj.filters
./build_dir/hostpkg/Python-2.7.15/PC/VS8.0/_sqlite3.vcproj
./build_dir/hostpkg/Python-2.7.15/PC/VS7.1/_sqlite3.vcproj
./build_dir/hostpkg/Python-2.7.15/PC/VS9.0/_sqlite3.vcproj
./build_dir/hostpkg/Python-2.7.15/PC/VC6/_sqlite3.dsp
```
The library is correctly generated but, as can be seen, the main difference is that
in Python-2.7.15 the library is named: **_sqlite3.so**
and in Python-3.6.5: **_sqlite3.cpython-36.so**.

Seems reasonable to think that this is the problem. What do you think?

---

_The Makefile code of the custom package that I'm using is the following one:_
```
#
# Copyright (C) 2014-2017 OpenWrt.org
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=python3-diskcache
PKG_VERSION:=3.1.1
PKG_RELEASE:=1

PKG_SOURCE:=diskcache-$(PKG_VERSION).tar.gz
PKG_SOURCE_URL:=https://files.pythonhosted.org/packages/61/51/6ea4d36a05314fe61c4107a45b28067b4ffe0e4e775b6edef5104c4a17d8/
PKG_HASH:=f5fa6a2274bc4fc2f1c884034f9ab1cc838054421ff493524732f52cc1174fbc

PKG_BUILD_DIR:=$(BUILD_DIR)/$(BUILD_VARIANT)-diskcache-$(PKG_VERSION)

include $(INCLUDE_DIR)/package.mk
include $(TOPDIR)/feeds/packages/lang/python/python-package.mk
include $(TOPDIR)/feeds/packages/lang/python/python3-package.mk

PKG_UNPACK=$(HOST_TAR) -C $(PKG_BUILD_DIR) --strip-components=1 -xzf $(DL_DIR)/$(PKG_SOURCE)

define Package/python-diskcache/Default
  SECTION:=lang
  CATEGORY:=Extra packages
  SUBMENU:=Python
  URL:=https://pypi.org/project/diskcache
endef

define Package/python-diskcache
  $(call Package/python-diskcache/Default)
  TITLE:=python-diskcache
  DEPENDS:=+PACKAGE_python-diskcache:python
  VARIANT:=python
endef

define Package/python3-diskcache
  $(call Package/python-diskcache/Default)
  TITLE:=python3-diskcache
  DEPENDS:=+PACKAGE_python3-diskcache:python3
  VARIANT:=python3
endef

define Package/python-diskcache/description
  Disk and file backed cache.

  DiskCache efficiently makes gigabytes of storage space available for caching.
  By leveraging rock-solid database libraries and memory-mapped files, cache
  performance can match and exceed industry-standard solutions. There’s no need
  for a C compiler or running another process. Performance is a feature and
  testing has 100% coverage with unit tests and hours of stress.
endef

define Package/python3-diskcache/description
$(call Package/python-diskcache/description)
.
(Variant for Python3)
endef

$(eval $(call PyPackage,python-diskcache))
$(eval $(call BuildPackage,python-diskcache))

$(eval $(call Py3Package,python3-diskcache))
$(eval $(call BuildPackage,python3-diskcache))
```

cshoredaniel, 2019-01-10:
You need a depends on python3-sqlite in Package/python3-diskcache .. (similarly on python-sqlite for python-diskcache) -- it happenstance that python-diskcache didn't try to build before python-sqlite was available and thus succeeded.

commodo, 2019-01-10:
Actually, that doesn't seem to be the case.
Your OpenWrt host Python seems to build `sqlite3` properly host-side for Python, but not for Python3.
[As mentioned here](https://github.com/openwrt/packages/tree/master/lang/python#build-considerations) OpenWrt builds Python & Python3 host-side ; those builds don't typically work all that well on all hosts [because some people have CentOS, some have Debian, some Arch Linux, etc]. But for the most part they do build reasonably well on all dists.

The issue seems to be with `diskcache` itself.
In the `setup.py` it does a `import diskcache`, which is not a good idea when cross-building.

My suggestion would be a patch like the one below, and this should work for both python-diskcache & python3-diskcache
```
diff --git a/diskcache/__init__.py b/diskcache/__init__.py
index 26532cc..d5dc592 100644
--- a/diskcache/__init__.py
+++ b/diskcache/__init__.py
@@ -1,9 +1,12 @@
 "DiskCache: disk and file backed cache."
 
-from .core import Cache, Disk, UnknownFileWarning, EmptyDirWarning, Timeout
-from .core import DEFAULT_SETTINGS, ENOVAL, EVICTION_POLICY, UNKNOWN
-from .fanout import FanoutCache
-from .persistent import Deque, Index
+import os
+
+if (not ("__SETUP_PY_INCLUDE__" in os.environ)):
+    from .core import Cache, Disk, UnknownFileWarning, EmptyDirWarning, Timeout
+    from .core import DEFAULT_SETTINGS, ENOVAL, EVICTION_POLICY, UNKNOWN
+    from .fanout import FanoutCache
+    from .persistent import Deque, Index
 
 __all__ = [
     'Cache',
diff --git a/setup.py b/setup.py
index 0f00554..751dc92 100644
--- a/setup.py
+++ b/setup.py
@@ -3,6 +3,8 @@ from setuptools import setup, find_packages
 from setuptools.command.test import test as TestCommand
 import sys
 
+import os
+os.environ['__SETUP_PY_INCLUDE__'] = '1'
 import diskcache
 
 class Tox(TestCommand):
```

It would also be a good idea to send this to the `diskcache` project [whether in this form, or a neater form].
Importing `diskcache` in `setup.py` [as they do] is not a good practice for cross-building; but generally building Python packages sometimes have quirks for cross-building anyway.

allaudet, 2019-01-10:
Thank you very much @cshoredaniel and @commodo.
Effectively the patch solves the problem and makes a lot of sense once you see it.
I will propose a PR to the `diskcache` project for this cross-building issue.

closed 2019-01-10 by allaudet