openwrt/packages #7937: nut: upsdrvctl command cannot find driver pid

realtmf, 2019-01-11:
Maintainer: @\cshoredaniel
Environment: VMware, Inc. VMware Virtual Platform, x86-64, OpenWrt 18.06.1 r7258-5eb055306f

Description: 

upsdrvctl commands appear to be looking for the driver's PID in /var/run/nut, while the PID appears to get written to /tmp/run/nut. The command therefore fails:

root@openwrt01:~# upsdrvctl stop
Network UPS Tools - UPS driver controller 2.7.4
kill: No such process
Stopping /var/run/nut/usbhid-ups-auto.pid failed: No such process
root@openwrt01:~# find / -name usbhid-ups-auto.pid
/tmp/run/nut/usbhid-ups-auto.pid

cshoredaniel, 2019-01-11:
@realtmf actually on OpenWrt /var is a symlink to /tmp hence writing to /tmp/run/nut/usbhid-ups-auto.pid.  Can you verify that the PID in /var/run/nut/usbhid-ups-auto.pid actually exists at the time you run upsdrvctl stop?  Also is a driver instance running at the time, or did you preceed with /etc/init.d/nut-server stop -- which would also stop the driver?

Can you show more of what you are doing and trying to do?

realtmf, 2019-01-11:
Thanks a lot for your quick response, @cshoredaniel .

And apologies - apparently that PID file was a leftover from testing (i.e. I created that manually).

I am just experimenting with nut at this point, and I was trying out the upsdrvctl commands. Upsdrvctl seems to fail because there is no PID file (which is why I created it manually).

Here's a sequence of steps - starting from fresh boot, checking that nut is running, then listing upsc output to confirm, finally trying upsdrvctl commands:

```

BusyBox v1.28.3 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 18.06.1, r7258-5eb055306f
 -----------------------------------------------------
root@openwrt01:~# ps | grep nut
 2201 root      9292 S    /lib/nut/usbhid-ups -D -a ups01
 2676 root      3084 R    grep nut
root@openwrt01:~# ll /var/run/nut/
drwxr-x---    2 root     root            60 Jan 11 17:18 ./
drwxr-xr-x    4 root     root           240 Jan 11 17:18 ../
srw-rw----    1 root     root             0 Jan 11 17:18 usbhid-ups-ups01=
root@openwrt01:~# upsc ups01
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 50
battery.date: 2001/09/25
battery.mfr.date: 2018/07/28
battery.runtime: 19800
battery.runtime.low: 90
battery.type: PbAc
battery.voltage: 27.1
battery.voltage.nominal: 24.0
device.mfr: American Power Conversion
device.model: Back-UPS XS 1400U
device.serial: xxxxxxxxxxxx
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: APC HID 0.96
driver.version.internal: 0.41
input.sensitivity: medium
input.transfer.high: 280
input.transfer.low: 155
input.voltage: 230.0
input.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.firmware: 926.T2 .I
ups.firmware.aux: T2
ups.load: 0
ups.mfr: American Power Conversion
ups.mfr.date: 2018/07/28
ups.model: Back-UPS XS 1400U
ups.productid: 0002
ups.realpower.nominal: 700
ups.serial: xxxxxxxxxxxx
ups.status: OL OFF
ups.test.result: No test initiated
ups.timer.reboot: 0
ups.timer.shutdown: -1
ups.vendorid: 051d
root@openwrt01:~# upsdrvctl stop ups01
Network UPS Tools - UPS driver controller 2.7.4
Can't open /var/run/nut/usbhid-ups-auto.pid: No such file or directory
```

cshoredaniel, 2019-01-11:
Ah, you shouldn't need to use upsdrvctl, if you configure /etc/config/nut according to https://openwrt.org/docs/guide-user/services/ups/software.nut you should just be able to /etc/init.d/nut-server start ups01 or /etc/init.d/nut-server stop ups01 (or to stop driver and sever, /etc/init.d/nut-server stop, likewise start).  According to upstream nut devs upsdrvctl isn't meant for end users (and openwrt doesn't actually use it, so I should probably remove it from the install).

realtmf, 2019-01-11:
Clear, thanks (and thank you for your hard work on the project - much appreciated!)

closed 2019-01-11 by realtmf

cshoredaniel, 2019-01-22:
Hey all pushed https://github.com/openwrt/packages/pull/8017!  This ought to be a major improvement and if all goes well I'm hoping to backport the current, much improved, master to 18.06.