openwrt/packages #7961: openssh: scp client vulnerabilities

ja-pa, 2019-01-14:
Openssh scp client suffers for multiple vulnerabilities related to malicious scp server. 
Full advisory https://sintonen.fi/advisories/scp-client-multiple-vulnerabilities.txt

@tripolar Do you think it's worth patching?

https://security-tracker.debian.org/tracker/CVE-2019-6110
https://security-tracker.debian.org/tracker/CVE-2019-6109
https://security-tracker.debian.org/tracker/CVE-2019-6111

BKPepe, 2019-03-03:
@tripolar Thank you for fixing CVE-2019-6111 by this commit https://github.com/openwrt/packages/commit/d92d34db5a382997e3d9230247644e55a1d84602!

BKPepe, 2019-07-22:
Just for reference. Commit https://github.com/openwrt/packages/commit/4729c3fddf05d6c8496ac2602a843e7dce5990e4#diff-7e78486312acc037feda42e1bea3dfc6 was applied applied to master (and it is already present in openwrt-19.07 branch). But the issue is still present in openwrt-18.06, there is a version of openssh 7.7p1, that's why I'm reopening it.

neheb, 2020-01-15:
@ja-pa I don't see why not.

BKPepe, 2020-02-22:
@ja-pa Would you please create pull request for OpenWrt 18.06?

cotequeiroz, 2021-03-18:
18.06 is EOL, so I'm closing this.

closed 2021-03-18 by cotequeiroz