openwrt/packages #7963: minidlna & ffmpeg invalid data when processing input

FireCulex, 2019-01-15:
Media Info Reports
`General
Complete name                            : \\192.168.1.1\file.mpg
Format                                   : MPEG-PS
File size                                : 209 MiB
Duration                                 : 11 min 32 s
Overall bit rate mode                    : Variable
Overall bit rate                         : 2 537 kb/s

Video
ID                                       : 224 (0xE0)
Format                                   : MPEG Video
Format version                           : Version 1
Format settings, BVOP                    : No
Format settings, Matrix                  : Default
Format settings, GOP                     : N=12
Duration                                 : 11 min 32 s
Bit rate mode                            : Variable
Bit rate                                 : 2 359 kb/s
Width                                    : 512 pixels
Height                                   : 384 pixels
Display aspect ratio                     : 4:3
Frame rate                               : 25.000 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Compression mode                         : Lossy
Bits/(Pixel*Frame)                       : 0.480
Time code of first frame                 : 00:00:00:00
Time code source                         : Group of pictures header
GOP, Open/Closed                         : Open
Stream size                              : 195 MiB (93%)

Audio
ID                                       : 192 (0xC0)
Format                                   : MPEG Audio
Format version                           : Version 1
Format profile                           : Layer 2
Duration                                 : 11 min 32 s
Bit rate mode                            : Constant
Bit rate                                 : 128 kb/s
Channel(s)                               : 2 channels
Sampling rate                            : 44.1 kHz
Frame rate                               : 38.281 FPS (1152 SPF)
Compression mode                         : Lossy
Stream size                              : 10.6 MiB (5%)`

diizzyy, 2019-01-15:
Platform? Settings?

diizzyy, 2019-02-09:
@hnyman @thess 
Close due to submitter timeout

closed 2019-02-09 by FireCulex