openwrt/packages #7973: avahi-daemon: crashes system when using enable-reflector=yes

spitfire, 2019-01-17:
Please make sure that the issue subject starts with `<package-name>: `

This repo here is only for packages maintained in this repo.  For base packages residing in the same repo as the build system and maintained by core devs, please consider opening tickets there for more timely responses

 - OpenWrt base system: https://bugs.openwrt.org
 - Most LuCI packages: https://github.com/openwrt/luci/issues

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

```
Formating code blocks by wrapping them with pairs of ```
```

neheb, 2019-01-17:
This could be an OOM problem. Try adding a swap device.

That version of OpenWrt IIRC had a kernel bug relating to IPv6. Might want to update.

spitfire, 2019-01-17:
> This could be an OOM problem. Try adding a swap device.
> 
> That version of OpenWrt IIRC had a kernel bug relating to IPv6. Might want to update.

This router has 128MB of RAM, and when I ran Avahi without reflector it was using ~30, so I doubt this is a memory issue.
Do you mean upgrade to 18.06.1?

neheb, 2019-01-17:
Yes.

AFAIK it should have the cache coherency issue fixed with that kernel. I remember adding the patch before 18.06 was released.

I used to run an mt7621 device with 512MB of RAM and would run out of it when transmission was active. Run free -a to see if you have any. Memory usage with this platform tends to be fairly weird.

spitfire, 2019-01-17:
I'm not really running anything else besides basics on it, BTW shouldn't OOM just kill the process, not the OS :)?
```root@r01:~# free -a
             total       used       free     shared    buffers     cached
Mem:        123936      49492      74444        564       1972      11236
-/+ buffers/cache:      36284      87652
Swap:            0          0          0```

neheb, 2019-01-17:
I don't think so. There's no OOM killer in OpenWrt as far as I know. When transmission used up massive RAM, it slowed the system down to a crawl (I mean serial console was delayed massively).

Could also be some kernel bug. I know samba with sendfile for example panics the kernel and reboots the device. Without sendfile it works great. I proposed removing to the OpenWrt mailing list, only to be met by opposition.

spitfire, 2019-01-17:
> I don't think so. There's no OOM killer in OpenWrt as far as I know. When transmission used up massive RAM, it slowed the system down to a crawl (I mean serial console was delayed massively).
>
> Could [...]

ok, thanks for the info - I'll try sysupgrade now and I'll know if it helped by tomorrow:)

spitfire, 2019-01-17:
Nope, that did not help - it ran for about 25 minutes and rebooted again. I was looking at htop with one eye, and avahi-daemon was using crazy amount of CPU power (2% when it was rather idle, then jumping between 20, 50 and 100%). I used to run it on some veeery old (from today's point of view) hardware and it's usage was never even noticeable, so there is something very wrong going on.

karlp, 2019-01-18:
I've seen the name incrementing issue when there were two devices on the network with the same IP.  Perhaps that's exploding something in avahi?

spitfire, 2019-01-18:
> I've seen the name incrementing issue when there were two devices on the network with the same IP. Perhaps that's exploding something in avahi?

No 2 devices on my network have the same IP address.

neheb, 2020-01-24:
Maybe an strace log would show something.

neheb, 2020-03-09:
Is this an issue with the latest version?