openwrt/packages #7975: domoticz: unable to change userdata path

lbdroid, 2019-01-17:
The default userdata (and therefore database) path for domoticz is set to /var/lib/domoticz/ , which is a ramdisk, and therefore its contents do not persist across reboots.

After changing the userdata path in /etc/config/domoticz to something more persistent, such as '/etc/domoticz/' (and setting the permissions on that path appropriately), domoticz is unable to work with the database file, with I/O errors listed on the system log, for example;

```Thu Jan 17 11:05:55 2019 daemon.err domoticz: SQL Query("create index if not exists ll_id_idx       on LightingLog(DeviceRowID);") : disk I/O error``

There is no actual disk error. The disk can read and write correctly under all circumstances and there are absolutely no errors from any process besides domoticz. This appears to be an issue with using the overlayfs.

Without the ability to store configurations, this package cannot be used at all.

poranje, 2019-02-03:
As a work-around, bind-mounting a persistent path on /var/lib/domoticz/ may help in this situation.
E.g. add to /etc/rc.local:

	mount -o bind /mnt/example /var/lib/domoticz/
	/etc/init.d/domoticz restart

But, you're right, the data path should be settable.

diizzyy, 2019-03-28:
@stintel

kofec, 2019-03-28:
Hi, I just test and it works, We can switch to forum:
https://forum.openwrt.org/t/domoticz-at-18-06-2/33866/10
but:
first try manually:
roginal command is:
`/usr/bin/domoticz -noupdates -approot /usr/share/domoticz/ -debuglevel -syslog daemon -userdata /var/lib/domoticz/ -sslwww 0`
so stop domoticz and change to e.g:
`/usr/bin/domoticz -noupdates -approot /usr/share/domoticz/ -debuglevel -syslog daemon -userdata /root/ -sslwww 0 -www 8084`
in my case:
/usr/bin/domoticz -noupdates -approot /usr/share/domoticz/ -debuglevel -syslog daemon -userdata /root/ -sslwww 0 -www 8084
```
2019-03-28 10:44:14.394  Status: Domoticz V4.10531 (c)2012-2019 GizMoCuz
2019-03-28 10:44:14.396  Status: Build Hash: bf19d8bb, Date: 2019-02-27 09:04:05
2019-03-28 10:44:14.398  Status: Startup Path: /usr/share/domoticz/
domoticz: Domoticz is starting up....
domoticz: Domoticz running...
domoticz: PluginSystem: Started, Python version '3.7.2'.
2019-03-28 10:44:16.355  Status: PluginSystem: Started, Python version '3.7.2'.
domoticz: Active notification Subsystems: gcm, http (2/13)
2019-03-28 10:44:16.405  Active notification Subsystems: gcm, http (2/13)
domoticz: WebServer(HTTP) started on address: :: with port 8084
2019-03-28 10:44:16.413  Status: WebServer(HTTP) started on address: :: with port 8084
domoticz: Proxymanager started.
2019-03-28 10:44:16.425  Status: Proxymanager started.
```
After that you have:
```
root@RpiTestowe:~# uci show | grep domo
domoticz.@domoticz[0]=domoticz
domoticz.@domoticz[0].disabled='1'
domoticz.@domoticz[0].loglevel='1'
domoticz.@domoticz[0].syslog='daemon'
domoticz.@domoticz[0].sslwww='0'
domoticz.@domoticz[0].userdata='/var/lib/domoticz/'
root@RpiTestowe:~# uci set domoticz.@domoticz[0].userdata='/root/'
root@RpiTestowe:~# uci commit
root@RpiTestowe:~#

```
If you have further questions please switch to forum. And I think that you can close this issue

lbdroid, 2019-03-28:
Is there a commit where this behavior was fixed?

kofec, 2019-03-28:
Have you tried to change it ?
e.g :
```
uci set domoticz.@domoticz[0].userdata='/root/'
uci commit
```

stintel, 2020-01-12:
The userdata path can be changed just fine. The problem is that sqlite does not work on JFFS2. This is not something that is easily fixable. There are workarounds, see https://www.sqlite.org/compile.html#temp_store and https://www.sqlite.org/pragma.html#pragma_temp_store.

Closing as it's not a bug.

closed 2020-01-12 by stintel