openwrt/packages #7980: net-snmp: default rules generated by fw3 open 161 udp on wan

gituriu1, 2019-01-18:
Environment:  
OpenWrt 18.06.1, r7258-5eb055306f

Description:
Default config opens port 161 udp on wan side

```
fw3 print | grep 161

iptables -t filter -A zone_wan_input -p udp -m udp --dport 161 -m comment --comment "!fw3: ubus:snmpd[instance1] rule 0" -j ACCEPT
iptables -t filter -A zone_lan_input -p udp -m udp --dport 161 -m comment --comment "!fw3: ubus:snmpd[instance1] rule 2" -j ACCEPT

```

Just noticed: This happens only when package mwan3 is installed. Is this the code that inserts the wrong rule in zone_wan_input?

```
/etc/init.d/snmpd

snmpd_setup_fw_rules() {                                              
        local net="$1"                                                
        local zone                                                    
                                                                         
        zone=$(fw3 -q network "$net" 2>/dev/null)                     
                                                                                  
        local handled_zone                                                        
        for handled_zone in $HANDLED_SNMP_ZONES; do                               
                [ "$handled_zone" = "$zone" ] && return                           
        done                                                                      
                                                                                  
        json_add_object ""                                                        
        json_add_string type rule                                                 
        json_add_string src "$zone"                                               
        json_add_string proto udp                                                 
        json_add_string dest_port 161                                             
        json_add_string target ACCEPT                                             
        json_close_object                                                      
                                                                               
        HANDLED_SNMP_ZONES="$HANDLED_SNMP_ZONES $zone"                         
} 
```
On a router without the mwan3 package installed no rules are inserted at all for udp 161

yousong, 2019-01-22:
Very likely commit 0bd19db0498780c3ac1e69ebc16c1334a609e285 already fixed this.  Can you confirm that?

stintel, 2019-02-16:
That PR was merged on 2/08/18, so it is included in 18.06.1.
@dedeckeh as you contributed the inbound firewall rule support, can you have a look at this?

dedeckeh, 2019-02-17:
@gituriu1 Can you attach the snmpd config.
I fail to understand the link with mwan3

feckert, 2020-11-04:
Since no feedback has been received from the creator and a solution has been described here. I assume that this issue was solved.

closed 2020-11-04 by feckert